esp8266/Arduino #4147: esp8266 connects to wifi, yet I can't access the webpage hosted by the esp8266

spencerkraisler, 2018-01-12:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			Wemos D1 mini lite with an esp8266
Core Version:      	2.4.0

### Description

I upload a basic sketch I got from online on turning the built-in LED on and off over the wifi. The device successfully connects to my school's wifi, yet I cannot access the webpage hosted by the esp8266. A few weeks ago, I did this and was successful at it. 

I also turned the Wemos D1 mini lite into a soft AP, and was successful in connecting to it. This shows the board isn't broken or anything, I think.

### Settings in IDE

Module:  		        Wemos D1 mini lite
Flash Size: 		        1 MB
CPU Frequency:		80Mhz
Flash Mode:			idk
Flash Frequency:	        40Mhz
Upload Using:		Microusb
Reset Method:		idk

### Sketch

#include <ESP8266WiFi.h>
const char* ssid = "WIFI";
const char* password = "oooo5834";
int led = D4;

WiFiServer server(80);

void setup() {
  Serial.begin(74880);
  delay(10);
  pinMode(led, OUTPUT);

  Serial.println();
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);
  while(WiFi.status() != WL_CONNECTED) {
    Serial.println(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  server.begin();
  Serial.println("Server started");

  Serial.print("Use this URL: ");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.println("/");
}

void loop() {
  WiFiClient client = server.available();
  if (client == 0) {
    return;
  }

  Serial.println("new client");
  while(client.available() == 0) {
    delay(1);
  }

  String request = client.readStringUntil('\r');
  Serial.println(request);
  client.flush();

  int value;
  if (request.indexOf("/LED=ON") > 0) {
    digitalWrite(led, LOW);
    value = 1;
  }
  if (request.indexOf("/LED=OFF") > 0) {
    digitalWrite(led, HIGH);
    value = 0;
  }
  Serial.println(value);
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: text/html");
  client.println(""); //  do not forget this one
  client.println("<!DOCTYPE HTML>");
  client.println("<html>");
 
  client.print("Led pin is now: ");
 
  if(value == 1) {
    client.print("On");  
  } else {
    client.print("Off");
  }
  client.println("<br><br>");
  client.println("Click <a href=\"/LED=ON\">here</a> turn the LED ON<br>");
  client.println("Click <a href=\"/LED=OFF\">here</a> turn the LED OFF<br>");
  client.println("</html>");
 
  delay(1);
  Serial.println("Client disconnected");
  Serial.println("");
 
}

### Debug Messages

```
messages here
```

5chufti, 2018-01-12:
does it still work with older core?
maybe port 80 was recently blocked on client (dynamic) IPs?

spencerkraisler, 2018-01-13:
@5chufti So I added "#include <WiFiClient.h>" and "#include <ESP8266WebServer.h>" to the sketch and it worked for some reason. I don't understand why though, since this sketch sued to work without those two lines.

5chufti, 2018-01-13:
hmmm, if this includes were missing, it should allready have complained during compile ...
so consider this issue as closed?

spencerkraisler, 2018-01-13:
@5chufti it’s literally doing it again, even though I had added the includes. I don’t understand, I literally had it working all proper a few hours ago, and I didn’t change anything.

spencerkraisler, 2018-01-13:
@5chufti Boy if you thought things were weird already, wait until you hear this. So I'm wondering if the d1 mini lite can even connect to a webpage. I uploaded a sketch that would extract weather data from a website and although it was successful in connecting to the website, it wouldn't extract any data. Then I go back to the sketch I posted in this thread, and it suddenly began working. It's almost as if the weather sketch woke up the board's "connect to webpage" abilities. I have no idea what's going on.

schlaegerz, 2018-01-22:
I also have problems with consistency at time.

Doing this when I setup

 ESP.eraseConfig();
    WiFi.disconnect();
    WiFi.setAutoConnect(false);
    WiFi.setAutoReconnect(false);
    WiFi.softAP(softAp_ssid.c_str());

seems to help

devyte, 2018-07-13:
There have been several fixes merged into the lwip glue since this was opened. Also, the original cause could have been the school networking.
Closing due to age.

closed 2018-07-13 by devyte

dolby360, 2019-08-16:
I can see the web page on the esp just prom the computer and not from phone.