esp8266/Arduino #4154: USB Host Shield Library fails to compile using 2.4.0

gdsports, 2018-01-13:
### Basic Infos

USB Host Shield Library 2.0 compiles using 2.3.0 but fails to 2.4.0.

#### Hardware
Hardware:			Any
Core Version:      	2.4.0 and git master

### Description

The [USB Host Shield Library 2.0](https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/board_qc) example board_qc.ino compiles using 2.3.0 but fails using 2.4.0. Other examples fail to compile in the same way.

The problem started with the same commit as in #4096. 

### Settings in IDE

Module:  			NodeMCU 1.0
default values

### Debug Messages

```
Arduino: 1.8.5 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Build options changed, rebuilding all
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp: In member function 'void PS4Parser::Parse(uint8_t, uint8_t*)':
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:87:105: error: no matching function for call to 'min(uint8_t, unsigned int)'
                         memcpy(&ps4Data, buf + 1, min((uint8_t)(len - 1), MFK_CASTUINT8T sizeof(ps4Data)));
                                                                                                         ^
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:87:105: note: candidates are:
In file included from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62:0,
                 from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:239,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/version_helper.h:24,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/settings.h:87,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/Usb.h:27,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.h:21,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:18:
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)
     min(initializer_list<_Tp> __l, _Compare __comp)
     ^
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4226:5: note:   template argument deduction/substitution failed:
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:87:105: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
                         memcpy(&ps4Data, buf + 1, min((uint8_t)(len - 1), MFK_CASTUINT8T sizeof(ps4Data)));
                                                                                                         ^
In file included from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62:0,
                 from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:239,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/version_helper.h:24,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/settings.h:87,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/Usb.h:27,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.h:21,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:18:
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)
     min(initializer_list<_Tp> __l)
     ^
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4221:5: note:   template argument deduction/substitution failed:
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:87:105: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
                         memcpy(&ps4Data, buf + 1, min((uint8_t)(len - 1), MFK_CASTUINT8T sizeof(ps4Data)));
                                                                                                         ^
In file included from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/algorithm:61:0,
                 from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:239,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/version_helper.h:24,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/settings.h:87,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/Usb.h:27,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.h:21,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:18:
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:239:5: note:   template argument deduction/substitution failed:
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:87:105: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'unsigned int')
[... 58 lines cut ...]
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:239:5: note:   template argument deduction/substitution failed:
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:96:105: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'unsigned int')
                         memcpy(&ps4Data, buf + 3, min((uint8_t)(len - 3), MFK_CASTUINT8T sizeof(ps4Data)));
                                                                                                         ^
In file included from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/algorithm:61:0,
                 from /home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:239,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/version_helper.h:24,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/settings.h:87,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/Usb.h:27,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.h:21,
                 from /home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:18:
/home/me/ardtmp/arduino-1.8.5-git/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)
     min(const _Tp& __a, const _Tp& __b)
[... cut ...]
/home/me/ardtmp/arduino-1.8.5-git/portable/sketchbook/libraries/USB_Host_Shield_Library_2.0/PS4Parser.cpp:96:105: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'unsigned int')
                         memcpy(&ps4Data, buf + 3, min((uint8_t)(len - 3), MFK_CASTUINT8T sizeof(ps4Data)));
                                                                                                         ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

devyte, 2018-01-13:
@gdsports it's right there send nd line down:
PS4Parser.cpp:87:105: error: no matching function for call to 'min(uint8_t, unsigned int)'
With 2.4.0 we migrated away from the #defined min() to std::min(), which is templated. Please read up on std::min, it requires both args to be of the same type. That code calls min with different types.
Closing due to exposed potential bug in 3rd party lib.

closed 2018-01-13 by devyte

gdsports, 2018-01-13:
@igrr @devyte
Have you considered finding another fix that does not break other libraries?

gdsports, 2018-01-13:
Clean compile with this patch.

````
diff --git a/cores/esp8266/Arduino.h b/cores/esp8266/Arduino.h
index b5621e4..8812025 100644
--- a/cores/esp8266/Arduino.h
+++ b/cores/esp8266/Arduino.h
@@ -248,13 +248,11 @@ void optimistic_yield(uint32_t interval_us);
 #include "Updater.h"
 #include "debug.h"
 
-using std::min;
-using std::max;
 using std::isinf;
 using std::isnan;
 
-#define _min(a,b) ((a)<(b)?(a):(b))
-#define _max(a,b) ((a)>(b)?(a):(b))
+#define min(a,b) ((a)<(b)?(a):(b))
+#define max(a,b) ((a)>(b)?(a):(b))
 
 uint16_t makeWord(uint16_t w);
 uint16_t makeWord(byte h, byte l);
diff --git a/cores/esp8266/spiffs_hal.cpp b/cores/esp8266/spiffs_hal.cpp
index 2d66bd5..68c3d34 100644
--- a/cores/esp8266/spiffs_hal.cpp
+++ b/cores/esp8266/spiffs_hal.cpp
@@ -24,6 +24,10 @@
 #include "spiffs/spiffs.h"
 #include "debug.h"
 
+#ifdef min
+#undef min
+#endif
+
 extern "C" {
 #include "c_types.h"
 #include "spi_flash.h"
````

devyte, 2018-01-13:
@gdsports I understand why you want that. However, use of #defined min/max is considered a potential bug in the C++ world. Even C compilers warn about some different type comparisons. 
My personal opinion is that it makes no sense to continue to support such code in favor of compatibility. It is better to force libs to fix their code. The fix is trivial enough. At the very least, it makes the authors briefly think about the types and how it makes sense to compare them. It also teaches them on usage for the future, to do it right next time.
If you have doubts, investigate and consider very carefully about why the C++ standard adopted this.

igrr, 2018-01-13:
FWIW, I think that there are a few cases in which std::min/max produce errors, but can actually be unambiguously resolved by widening the two types.

E.g. if Ta is uint8 and Tb is int32, then a can be widened to int32 instead of producing an error. Of course, in some cases it is not possible — e.g. when int32 and uint64 are being compared.
I still have such a "relaxed" implementation in one of my older projects, will dig it up...