esp8266/Arduino #4161: 2.4.0 will not automatically reconnect

pidloop, 2018-01-13:
The following sketch would automatically reconnect using 2.3.0 but in 2.4.0 it never does. To test:

1. edit ssid and pw to your AP creds
2. turn on your Access Point radio, start sketch, it will soon report WL_CONNECTED
3. turn off the AP radio and it will soon report WL_DISCONNECTED
4. turn on AP radio. Version 2.3.0 will soon report WL_CONNECTED but 2.4.0 stays WL_DISCONNECTED forever
5. Note that in 2.4.0 even though the status stays WL_DISCONNECTED after the AP restart pinging the ESP IP does succeed, so it seems the ESP really does automatically reconnect but does not report this as status.

Thank you.

`
#include <stdlib.h>
#include <string.h>
#include <math.h>

#include <SPI.h>
#include <ESP8266WiFi.h>
#include <WiFiServer.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>

char ssid[] = "xxx";
char pw[] = "yyy";

void setup()
{
    Serial.begin(115200);

    WiFi.enableSTA(true);

    WiFi.setAutoConnect (true);
    WiFi.setAutoReconnect (true);

    WiFi.begin (ssid, pw);

    Serial.println("start");
}

void loop()
{
    prWiFiStatus(WiFi.status());

    if (WiFi.isConnected())
        Serial.println (WiFi.localIP());

    delay(2000);
}

void prWiFiStatus (int s)
{
    #define     VALCASE(x)      case x: Serial.println(#x); break;

    switch (s) {
    VALCASE(WL_NO_SHIELD);
    VALCASE(WL_IDLE_STATUS);
    VALCASE(WL_NO_SSID_AVAIL);
    VALCASE(WL_SCAN_COMPLETED);
    VALCASE(WL_CONNECTED);
    VALCASE(WL_CONNECT_FAILED);
    VALCASE(WL_CONNECTION_LOST);
    VALCASE(WL_DISCONNECTED);
    default: Serial.println(s); break;
    }
}
`

tonton81, 2018-01-14:
i noticed it doesnt reconnect as well, have to call wifi begin again or reboot ESP.. didnt have this issue before 2.4

d-a-v, 2018-01-14:
Would you agree this is duplicate of #4101 ?
If so, try latest git master version of core.

pidloop, 2018-01-15:
Thanks for really fast response.

Yes, I agree this appears to be the same issue.

Unfortunately I do not understand how to use the git versions, I can only
get the stable versions to work via the Arduino IDE on my macOS system. I
found the directions at https://github.com/esp8266/Arduino#using-git-version
but it is unclear what is meant by "Go to Arduino directory". I tried
following the remaining steps using these *hardware* directories:

   - /Applications/Arduino.app/Contents/Java/hardware
   - ~/Library//Arduino15/packages/esp8266/hardware

but if I use either of these the IDE no longer lists any esp8266 boards
after restarting. Sorry.

d-a-v, 2018-01-15:
You can also try and use lwip v1.4 in Tools>Menu instead. Sorry I forgot to mention that as in #4166.
This will be fixed in the next release.

pidloop, 2018-01-15:
Solved. See 4166.

Many thanks.

closed 2018-01-15 by pidloop

R360GRADOS, 2018-04-03:
```
boolean WiFiReturns() {
  switch (WiFi.status()) {
    case WL_NO_SHIELD: return 0;
    case WL_IDLE_STATUS: return 0;
    case WL_NO_SSID_AVAIL: return 0;
    case WL_SCAN_COMPLETED: return 1;
    case WL_CONNECTED: return 1;
    case WL_CONNECT_FAILED: return 0;
    case WL_CONNECTION_LOST: return 0;
    case WL_DISCONNECTED: return 0;
    default: return 0;
  }
}
```
https://www.arduino.cc/en/Reference/WiFiStatus
    WL_CONNECTED: assigned when connected to a WiFi network;
    WL_NO_SHIELD: assigned when no WiFi shield is present;
    WL_IDLE_STATUS: it is a temporary status assigned when WiFi.begin() is called and remains active until the number of attempts expires (resulting in WL_CONNECT_FAILED) or a connection is established (resulting in WL_CONNECTED);
    WL_NO_SSID_AVAIL: assigned when no SSID are available;
    WL_SCAN_COMPLETED: assigned when the scan networks is completed;
    WL_CONNECT_FAILED: assigned when the connection fails for all the attempts;
    WL_CONNECTION_LOST: assigned when the connection is lost;
    WL_DISCONNECTED: assigned when disconnected from a network; 

```
void loop() {
  if (!WiFiReturns()) { 
    WiFi.reconnect(); //Connection OFF - conexión OFF
  } else { 
    //conexión ON - Connection ON
  }
}
```
Saludos