esp8266/Arduino #4163: Multiple definition of `strtof'

Loreno10, 2018-01-14:
### Basic Infos
I am trying to run basic IoT Hub examples. Everything worked fine until I downgraded Azure libraries to some older version.
Now on every compilation I get:
`Arduino:1.8.5 (Mac OS X), Board:"SparkFun ESP8266 Thing Dev, 80 MHz, 512K (no SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Project options changed, reload all
Archiving built core (caching) in: /var/folders/7n/0lxjg6bx47z8c8h0jhqwnl1w0000gn/T/arduino_cache_62019/core/core_esp8266_esp8266_thingdev_CpuFrequency_80,FlashSize_512K0,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_5148a477b5096f9137ee6e8061fe60fd.a
/Users/marcin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-strtod.o): In function `strtof':
/Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtod.c:1267: multiple definition of `strtof'
libraries/AzureIoTHub/esp8266/azcpgmspace.cpp.o:/Users/marcin/Documents/Arduino/libraries/AzureIoTHub/src/esp8266/azcpgmspace.cpp:24: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Compilation error for board SparkFun ESP8266 Thing Dev.
`

#### Hardware
Hardware:			SparkFun Thing Dev
Core Version:      	2.4.0

### Description

Problem description

### Settings in IDE

Module:  			SparkFun Thing Dev
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

Loreno10, 2018-01-14:
Solved by downgrading to version 2.3.0 of ESP8266 definitions. Seems to me that 2.4.0 should not be a recommended update, looking at problems with reconnections that also occur.

closed 2018-01-14 by Loreno10

igrr, 2018-01-14:
Why is AzureIoTHub/src/esp8266/azcpgmspace.cpps defining strtof though? Seems it needs to be marked "weak" in that library, so that the conflict does not occur is the core has this symbol defined.