esp8266/Arduino #4164: macOS vs Windows - program works well only when uploaded on Windows

Loreno10, 2018-01-14:
### Basic Infos
I have a program that connects my micro controller to Microsoft's Azure IoT Hub.
Same Arduino sketch works properly when uploaded from Windows 7 (Arduino IDE), but it throws an exception during its execution when uploaded from macOS 10.13.2 (FTDI driver installed, Arduino IDE).

#### Hardware
Hardware:			Sparkfun Thing DEV
Core Version:      	230

### Description
As stated in "Basic infos" section macOS does something wrong with the uploaded program. Here's the console logs:

```
connected with Redmi4X, channel 6
dhcp client start...
ip:192.168.43.239,mask:255.255.255.0,gw:192.168.43.1
Attempting to connect to SSID: Redmi4X.
f r0, You device with MAC address 5c:cf:7f:e3:56:9b connects to Redmi4X failed! Waiting 10 seconds to retry.
scandone
f 0, scandone
reconnect
state: 5 -> 0 (0)
rm 0
f -180, scandone
no Redmi4X found, reconnect after 1s
reconnect
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with Redmi4X, channel 6
dhcp client start...
ip:192.168.43.239,mask:255.255.255.0,gw:192.168.43.1
Connected to wifi Redmi4X.
please start sntp first !
Fetching NTP epoch time failed! Waiting 2 seconds to retry.
Sun Jan 14 13:17:13 2018

Fetched NTP epoch time is: 1515935834.
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4000bf64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4024b231, depc=0x00000000

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024b231 depc=0x00000000

ctx: cont 
sp: 3fff2050 end: 3fff2330 offset: 01a0

>>>stack>>>
3fff21f0:  00000001 3fff23b8 00000004 00000048  
3fff2200:  4010612b 00000000 00000000 3fff22a0  
3fff2210:  3fff22d0 40107524 3fff12d8 00000008  
3fff2220:  3fff12d8 00000027 3fff2277 402103dc  
3fff2230:  40106510 00000000 3ffe925d 00000028  
3fff2240:  3fff12d8 00000008 3fff2250 4020fa6c <
3fff2250:  63746546 20646568 2050544e 636f7065  
3fff2260:  69742068 6920656d 31203a73 39353135  
3fff2270:  33383533 0a0d2e34 00000000 3fff22b0  
3fff2280:  3fff22a0 40202399 3fff12fc 3ffe9306  
3fff2290:  3fffdad0 3fff22a0 00000008 3fff1304  
3fff22a0:  4024b231 3ffe96c1 60000000 0000007e  
3fff22b0:  6000001c 0000002b 3fff1310 3fff120c  
3fff22c0:  3ffe92b4 3fff1304 3fff22a0 3fff22d0  
3fff22d0:  4020c0ae 000007d0 00000000 3fff1304  
3fff22e0:  3fffdad0 00000000 3fff12d8 4020d30e  
3fff22f0:  3fffdad0 00000000 3fff12fc 3fff1304  
3fff2300:  3fffdad0 00000000 3fff12fc 4020d555  
3fff2310:  feefeffe feefeffe feefeffe 4020ff38  
3fff2320:  feefeffe feefeffe 3fff1310 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

First thing that is rather strange is the fact that in this console log it looks like my ESP8266 first connects to my hotspot (Redmi4X) and after that it attempts to connect to it! In the code the line
'Attempting to connect to SSID: Redmi4X.'
is printed before the actual connection takes place. On Windows 7 the order is correct.

Second, I get the exception, which decoded by ESP Exception Decoder translates into:
```
0x4010612b: os_printf_plus at ?? line ?
0x40107524: vsnprintf at /Users/marcin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/libc_replacements.c line 342
0x402103dc: Print::write(unsigned char const*, unsigned int) at /Users/marcin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.cpp line 84
0x40106510: sntp_dns_found at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/sntp.c line 910
0x4020fa6c: Print::printf(char const*, ...) at /Users/marcin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.cpp line 84
0x40202399: IoTHubClient_LL_CreateFromConnectionString at /Users/marcin/Documents/Arduino/libraries/AzureIoTHub/src/sdk/iothub_client_ll.c line 1776
0x4024b231: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4020c0ae: delay at /Users/marcin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring.c line 53
0x4020d30e: initTime() at /Users/marcin/Desktop/Sparkfun-led/app/credentials.ino line 46
0x4020d555: setup at /Users/marcin/Desktop/Sparkfun-led/app/credentials.ino line 46
0x4020ff38: loop_wrapper at /Users/marcin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at /Users/marcin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S line 109

```

In the below section I also have given information about libraries used and versions of them.

This is the type of errors I hate the most, because I'm totally lost what is the cause. I am just uploading a program onto micro controller. Simple sketch like 'Blink' works just fine.

### Settings in IDE

Module:  			Sparkfun Thing DEV
All settings are DEFAULT

Used libraries:
ArduinoJson (5.11.2)
AzureIoTHub (1.0.4.0)
AzureIoTProtocol_MQTT (1.0.4.1)
AzureIoTUtility (1.0.41)

### Sketch
The sketch is available here:
https://www.odrive.com/s/d8131129-0991-4170-a7cc-738e7efcc808-5a5b5916

(I didn't paste the code here, because it spans across multiple files)
It's actually very similar to IoT-Hub startup guide (maybe the Microsoft's example code would even behave the same as mine, since I didn't change much)

### Debug Messages

```
messages here
```

pidloop, 2018-01-15:
Hi. Not sure this helps, but I just had a somewhat similar experience that might shed a clue.

My sketch worked fine using ESP Board release 2.3.0 but had very weird connection issues using 2.4.0. So I'm just wondering if by some chance your Microsoft IDE system is using the 2.3.0 library and your Mac is configured to use 2.4.0. You can check by opening Tools -> Board -> Board Manager, then looking for the esp 8266 board library.

Anyway, good luck.

Loreno10, 2018-01-16:
@pidloop I saw your thread about that and I've been using 2.3.0 because of that.
Still, thanks for your suggestion control theory fan ;)

schlaegerz, 2018-01-18:
I do also get slight differences, one of them i noticed is it deals with flash strings differently. On my PC 

memberVar = String(F("String"));

worked fine but not on my mac, could be a related issue.

earlephilhower, 2019-01-16:
Closing as there's no MCVE and we're pretty confident that Mac, Windows, and Linux all compile the same code to the same bits (we're building the compiler at the same time for each OS, with the exact same source code).

closed 2019-01-16 by earlephilhower