esp8266/Arduino #4167: Issue with AP password length in AP_STA mode

hk-on-github, 2018-01-15:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			Wemos Mini D1
Core Version:      	2.4.0

### Description

When the password length for the AP in AP_STA mode is more than 8 characters I am not
getting any answers from NTP

### Settings in IDE

Module:  			Wemos D1 R2 mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

The sketch below is the standard NTP example sketch (adapted from Ethernet to WiFi).
I did not make any changes in the NTP part. My changes are only in the setup part
to adapt from Ethernet to WiFi. (original example is ethernet).

Step 0: Use IDE version **2.4.0** (Arduino 1.8.5)
Step 1: Look for the critical line in the code below (marked <=========)
Step 2: Try with a AP password that is less than 8 characters long
            e.g.  WiFi.softAP("TEST", "1234567");
Step 3: watch the  NTP answers coming in every 11 seconds
                    Seconds since Jan 1 1900 = 3724996794
                    Unix time = 1516007994
                    The UTC time is 9:19:54
                    ++++++++++++++++++++++++++++++++++++++++++
Step 4: No change the AP password to
             e.g.  WiFi.softAP("TEST", "12345678");
Step 5: watch that there are no NTP answers getting through
            you only get:
                    ++++++++++++++++++++++++++++++++++++++++++
                    ++++++++++++++++++++++++++++++++++++++++++
Step 6: change the IDE version to **2.4.0-rc2**
Step 7: watch the  NTP answers coming in every 11 seconds
                    Seconds since Jan 1 1900 = 3724996794
                    Unix time = 1516007994
                    The UTC time is 9:19:54
                    ++++++++++++++++++++++++++++++++++++++++++ 

So the workaround is:

a) either stay on 2.4.0-rc2
or
b) use an AP password length of less than 8 characters
### Sketch

/*

 Udp NTP Client

 Get the time from a Network Time Protocol (NTP) time server
 Demonstrates use of UDP sendPacket and ReceivePacket
 For more on NTP time servers and the messages needed to communicate with them,
 see http://en.wikipedia.org/wiki/Network_Time_Protocol

 created 4 Sep 2010
 by Michael Margolis
 modified 9 Apr 2012
 by Tom Igoe

 This code is in the public domain.

 */

#include <Arduino.h>

#if defined (ARDUINO_ARCH_ESP8266)
extern "C" {
	#include "user_interface.h"
}
#endif

#if defined (ARDUINO_ARCH_AVR)
	#include <EEPROM.h>
#endif

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

char my_ssid_0[]		=	"YOUR_SSID";
char my_password_0[]	=	"YOUR_STA_PASSWORD";
unsigned int localPort = 8888;       // local port to listen for UDP packets
char timeServer[] = "ptbtime1.ptb.de"; // time.nist.gov NTP server
const int NTP_PACKET_SIZE = 48; // NTP time stamp is in the first 48 bytes of the message
byte packetBuffer[NTP_PACKET_SIZE]; //buffer to hold incoming and outgoing packets
WiFiUDP UDPNTPClient;

void setup()
{
	// Open serial communications and wait for port to open:
	delay(1000);
	Serial.begin(115200);
	Serial.println(F("SETUP START"));

	// WiFi.mode(WIFI_STA);
	WiFi.mode(WIFI_AP_STA);
	WiFi.begin(my_ssid_0, my_password_0);
	while(WiFi.waitForConnectResult() != WL_CONNECTED)
	{
		Serial.print(".");
		delay(1000);
	}

	// WiFi.softAP("TEST", "TEST");
        WiFi.softAP("TEST", "12345678");   // <======== critical line

	UDPNTPClient.begin(localPort);
}

void loop()
{
  sendNTPpacket(timeServer); // send an NTP packet to a time server

  // wait to see if a reply is available
  delay(1000);
  if ( UDPNTPClient.parsePacket() ) {
    // We've received a packet, read the data from it
    UDPNTPClient.read(packetBuffer, NTP_PACKET_SIZE); // read the packet into the buffer

    //the timestamp starts at byte 40 of the received packet and is four bytes,
    // or two words, long. First, esxtract the two words:

    unsigned long highWord = word(packetBuffer[40], packetBuffer[41]);
    unsigned long lowWord = word(packetBuffer[42], packetBuffer[43]);
    // combine the four bytes (two words) into a long integer
    // this is NTP time (seconds since Jan 1 1900):
    unsigned long secsSince1900 = highWord << 16 | lowWord;
    Serial.print("Seconds since Jan 1 1900 = " );
    Serial.println(secsSince1900);

    // now convert NTP time into everyday time:
    Serial.print("Unix time = ");
    // Unix time starts on Jan 1 1970. In seconds, that's 2208988800:
    const unsigned long seventyYears = 2208988800UL;
    // subtract seventy years:
    unsigned long epoch = secsSince1900 - seventyYears;
    // print Unix time:
    Serial.println(epoch);

    // print the hour, minute and second:
    Serial.print("The UTC time is ");       // UTC is the time at Greenwich Meridian (GMT)
    Serial.print((epoch  % 86400L) / 3600); // print the hour (86400 equals secs per day)
    Serial.print(':');
    if ( ((epoch % 3600) / 60) < 10 ) {
      // In the first 10 minutes of each hour, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.print((epoch  % 3600) / 60); // print the minute (3600 equals secs per minute)
    Serial.print(':');
    if ( (epoch % 60) < 10 ) {
      // In the first 10 seconds of each minute, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.println(epoch % 60); // print the second
  }
  
  Serial.println(F("++++++++++++++++++++++++++++++++++++++++++"));
  
  // wait ten seconds before asking for the time again
  delay(10000);
}

// send an NTP request to the time server at the given address
unsigned long sendNTPpacket(char* address)
{
  // set all bytes in the buffer to 0
  memset(packetBuffer, 0, NTP_PACKET_SIZE);
  // Initialize values needed to form NTP request
  // (see URL above for details on the packets)
  packetBuffer[0] = 0b11100011;   // LI, Version, Mode
  packetBuffer[1] = 0;     // Stratum, or type of clock
  packetBuffer[2] = 6;     // Polling Interval
  packetBuffer[3] = 0xEC;  // Peer Clock Precision
  // 8 bytes of zero for Root Delay & Root Dispersion
  packetBuffer[12]  = 49;
  packetBuffer[13]  = 0x4E;
  packetBuffer[14]  = 49;
  packetBuffer[15]  = 52;

  // all NTP fields have been given values, now
  // you can send a packet requesting a timestamp:
  UDPNTPClient.beginPacket(address, 123); //NTP requests are to port 123
  UDPNTPClient.write(packetBuffer, NTP_PACKET_SIZE);
  UDPNTPClient.endPacket();
}

5chufti, 2018-01-15:
and did you check if your "Test" AP is indeed appearing and you can connect when only using 7 char password? AFAIK a password of at least 8char is needed for WPA, 
(I don't know if a concurrent use of AP and STA mode is possible).

hk-on-github, 2018-01-15:
Hi,

you are right I cannot connect.

I see the access point in my smart phone.
However when I try to connect it expects at least 8 characters as password.

The connect button therefore is greyed out and I cannot press connect.

When I use 8 characters, I get the error described.

Everything is fine if I go back to version 2.4.0-rc2

Regards,
Holger

lrmoreno007, 2018-01-24:
I can't reproduce the problem with 2.4.0 core (git version). And I can connect to "TEST" in my smart phone.
[image]

hk-on-github, 2018-01-26:
I used a new Windows10 installation, I installed Arduino IDE 1.8.5.
I use the Wemos D1 R2 &mini. 
I am able to connect to "TEST" with my smart phone, however I am not getting NTP answers.

This is what I see:

[image]

hk-on-github, 2018-01-26:
When I go to a password with only 7 characters, I get NTP answers:

[image]

DIRR70, 2018-03-15:
Hi guys,

I've encountered similar problems as hk.

I found that the length of the password is not important (of course it has to be at least 8 characters long according to the specs)!

**The problem is the order of starting AP and STA in combination with the lwIP variant!
- if you call WiFi.softAP before WiFi.begin the UDP packages will be received
- if you call WiFi.softAP after WiFi.begin the UDP packages will NOT be received, except you switch from lwIP variant 2.0 to variant 1.4**

I've tested with Version 2.4.1 ans 2.4.0

Here the test sketch what is the NTPCLient example extended by the softAP call (see ****)
``/*

 Udp NTP Client

 Get the time from a Network Time Protocol (NTP) time server
 Demonstrates use of UDP sendPacket and ReceivePacket
 For more on NTP time servers and the messages needed to communicate with them,
 see http://en.wikipedia.org/wiki/Network_Time_Protocol

 created 4 Sep 2010
 by Michael Margolis
 modified 9 Apr 2012
 by Tom Igoe
 updated for the ESP8266 12 Apr 2015 
 by Ivan Grokhotkov

 This code is in the public domain.

 */

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

char ssid[] = "*************";  //  your network SSID (name)
char pass[] = "********";       // your network password

unsigned int localPort = 2390;      // local port to listen for UDP packets

/* Don't hardwire the IP address or we won't get the benefits of the pool.
 *  Lookup the IP address for the host name instead */
//IPAddress timeServer(129, 6, 15, 28); // time.nist.gov NTP server
IPAddress timeServerIP; // time.nist.gov NTP server address
const char* ntpServerName = "time.nist.gov";

const int NTP_PACKET_SIZE = 48; // NTP time stamp is in the first 48 bytes of the message

byte packetBuffer[ NTP_PACKET_SIZE]; //buffer to hold incoming and outgoing packets

// A UDP instance to let us send and receive packets over UDP
WiFiUDP udp;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  // ************
  // If AP is started before STA, UDP will work well!
  //WiFi.softAP("TESTAP", "");
  // ************

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  //************
  // If AP is started after STA, UDP will only work with lwIP Variant 1.4 !!!
  WiFi.softAP("TESTAP", "");
  //************

  Serial.println("Starting UDP");
  udp.begin(localPort);
  Serial.print("Local port: ");
  Serial.println(udp.localPort());
}

void loop()
{
  //get a random server from the pool
  WiFi.hostByName(ntpServerName, timeServerIP); 

  sendNTPpacket(timeServerIP); // send an NTP packet to a time server
  // wait to see if a reply is available
  delay(1000);
  
  int cb = udp.parsePacket();
  if (!cb) {
    Serial.println("no packet yet");
  }
  else {
    Serial.print("packet received, length=");
    Serial.println(cb);
    // We've received a packet, read the data from it
    udp.read(packetBuffer, NTP_PACKET_SIZE); // read the packet into the buffer

    //the timestamp starts at byte 40 of the received packet and is four bytes,
    // or two words, long. First, esxtract the two words:

    unsigned long highWord = word(packetBuffer[40], packetBuffer[41]);
    unsigned long lowWord = word(packetBuffer[42], packetBuffer[43]);
    // combine the four bytes (two words) into a long integer
    // this is NTP time (seconds since Jan 1 1900):
    unsigned long secsSince1900 = highWord << 16 | lowWord;
    Serial.print("Seconds since Jan 1 1900 = " );
    Serial.println(secsSince1900);

    // now convert NTP time into everyday time:
    Serial.print("Unix time = ");
    // Unix time starts on Jan 1 1970. In seconds, that's 2208988800:
    const unsigned long seventyYears = 2208988800UL;
    // subtract seventy years:
    unsigned long epoch = secsSince1900 - seventyYears;
    // print Unix time:
    Serial.println(epoch);

    // print the hour, minute and second:
    Serial.print("The UTC time is ");       // UTC is the time at Greenwich Meridian (GMT)
    Serial.print((epoch  % 86400L) / 3600); // print the hour (86400 equals secs per day)
    Serial.print(':');
    if ( ((epoch % 3600) / 60) < 10 ) {
      // In the first 10 minutes of each hour, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.print((epoch  % 3600) / 60); // print the minute (3600 equals secs per minute)
    Serial.print(':');
    if ( (epoch % 60) < 10 ) {
      // In the first 10 seconds of each minute, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.println(epoch % 60); // print the second
  }
  // wait ten seconds before asking for the time again
  delay(10000);
}

// send an NTP request to the time server at the given address
void sendNTPpacket(IPAddress& address)
{
  Serial.println("sending NTP packet...");
  // set all bytes in the buffer to 0
  memset(packetBuffer, 0, NTP_PACKET_SIZE);
  // Initialize values needed to form NTP request
  // (see URL above for details on the packets)
  packetBuffer[0] = 0b11100011;   // LI, Version, Mode
  packetBuffer[1] = 0;     // Stratum, or type of clock
  packetBuffer[2] = 6;     // Polling Interval
  packetBuffer[3] = 0xEC;  // Peer Clock Precision
  // 8 bytes of zero for Root Delay & Root Dispersion
  packetBuffer[12]  = 49;
  packetBuffer[13]  = 0x4E;
  packetBuffer[14]  = 49;
  packetBuffer[15]  = 52;

  // all NTP fields have been given values, now
  // you can send a packet requesting a timestamp:
  udp.beginPacket(address, 123); //NTP requests are to port 123
  udp.write(packetBuffer, NTP_PACKET_SIZE);
  udp.endPacket();
}``