esp8266/Arduino #4168: Add LWIP2 support for platformio

Redferne, 2018-01-15:
Please add support for building with lwip2 in PlatformIO. Or please provide instructions on how to build with lwip2 without manually editing platformio-build.py file.

d-a-v, 2018-01-15:
What about adding a new script ```platformio-build-lwip2.py``` patched script for a better audience before further integration ?
Could you make a pull-request for that ?

Redferne, 2018-01-16:
Sure, can you please exlain how this new script will be executed by PlatformIO?
I guess it possible to manually edit `~/.platformio/platforms/espressif8266/builder/frameworks/arduino.py` but that kinds of defeats the purpose?

ivankravets, 2018-01-16:
Which changes do you need in build script?

Redferne, 2018-01-17:
To be able to build with lwip2, I made the following change, replace lwip with lwip2 and replace lwip_gcc with lwip2:
```
        join(FRAMEWORK_DIR, "tools", "sdk", "lwip2", "include"),

        "mesh", "wpa2", "smartconfig", "espnow", "pp", "main", "wpa", "lwip2",
```
But I think it should be possible to configure this for your target in platformio.ini instead, like a possiblity to override parts of the CPPPATH and LIBS.

ivankravets, 2018-01-17:
Have you tried http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html ?

Redferne, 2018-01-18:
Yes. Did not work.

ivankravets, 2018-04-13:
Duplicate of https://github.com/platformio/platform-espressif8266/issues/65

Indeed, you can change LWIP using `build_unflags` and `build_flags` options. However, we will add the better option.

closed 2018-04-13 by ivankravets

ivankravets, 2018-04-20:
See http://docs.platformio.org/en/latest/platforms/espressif8266.html#lwip-variant