esp8266/Arduino #4169: How to use gdbstub

weswitt, 2018-01-15:
I need some help getting gdbstub to work. This is what I'm seeing. I am running this on Windows.

I have compiled gdbstub and linked it into my sketch. I modified user_init() to call gdbstub_init(). I then flash the device and reset. I then start xtensa-lx106-elf-gdb with the gdb commands below. I have also listed the gdb log below too. 

**GDB COMMANDS:**
file d:/dev/src/homeauto/sonoff/app/bin/esp8266/sonoff.elf
set remotelogfile d:/dev/src/homeauto/sonoff/app/gdblog.txt
set serial baud 115200
set remote hardware-breakpoint-limit 1
set remote hardware-watchpoint-limit 1
target remote com4

**GDB LOG:**
w +$qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEve
nts+;no-resumed+#df
r <Timeout: 2 seconds>
w $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEven
ts+;no-resumed+#df
r <Timeout: 2 seconds>
w $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEven
ts+;no-resumed+#df
r <Timeout: 2 seconds>
w $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEven
ts+;no-resumed+#df
r <Timeout: 2 seconds><Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -+$Hg0#df
r <Timeout: 2 seconds>
w $Hg0#df
r <Timeout: 2 seconds>
w $Hg0#df
r <Timeout: 2 seconds>
w $Hg0#df
r <Timeout: 2 seconds>
End of log

freeck, 2018-01-15:
Did you "see" the gdbstub output on the serial port (visible after resetting the target)?
Did you include the compiler flags "-ggdb -Og" ?

weswitt, 2018-01-15:
I would expect that the device would stop at a breakpoint after reset since I've set GDBSTUB_BREAK_ON_INIT in gdbstub-cfg.h, but that is not happening. After resetting the device appears to boot normally as I see the debug output from my sketch. I do see some output immediately after restart that looks binary-ish and is not from my code. Yes I built with -ggdb and -Og.

**Output immediately after reset:**
slpglbn|dcoo

freeck, 2018-01-15:
I do not recognize the output, you should see something like $ST05#d9....

weswitt, 2018-01-15:
What baud rate does the GDB stub use? I could not find anything in the stub that established the rate.

freeck, 2018-01-15:
The baud rate is flexible. 
I included the following code in function setup() of an Arduino-app:
  uart_div_modify(0, UART_CLK_FREQ / 115200);
  Serial.begin(115200); 
  gdbstub_init();
You should also include a RAM-directive __attribute__((section(".entry.text"))) in the function headers you want to debug. And add GDB-command "thbreak loop" , so gdb knows where to break the first time.
It is quit a hassle to make it work. I have written a sort of tutorial (in Dutch) which describes the whole setup for Sloeber-IDE. So if you are interested I share this document.

weswitt, 2018-01-15:
Thanks for the tips. I have it working. My sketch stops at the initial breakpoint in the gdbstub just as it should. Beautiful. Now if I can find a way to get vscode to work with remote serial debugging I'll be happy.

devyte, 2018-01-24:
@weswitt I've seen this question several times. Given that you got it working, could you please do a writeup with instructions? You could either do a PR with a new doc that links from readthedocs, or just post the guide here and someone will transcribe it into a doc.

freeck, 2018-01-24:
First off all: what do you mean by "vscode" ?
As I said  I have written an instruction in Dutch. See: https://1drv.ms/f/s!AgjNziQJTfNOhoJzP0uwbECd3NppYA  It also contains a sample Sketch, and a youtube clip. The instruction is based on using Eclipse-sloeber IDE , but you could also use command line gdb-debugging. The instruction needs some updating as in the mean time v2.4.0 is released.

Another instruction can be found here: https://www.youtube.com/watch?v=S_QTMNhaDwM&t=742s

devyte, 2018-01-24:
@freeck I speak german, but not dutch, but in any case, the link is blocked for me due to policy beyond my control.
As for the youtube links, I'm looking to have a written guide included in our docs, and have it focus on general command line gdb for compatibility across all user development platforms. Could you maybe put something together for integration into our docs? or work with @weswitt on a PR?

freeck, 2018-01-24:
I am in the middle of the process of making an instruction manual for the esp32 using OpenOCD-JTAG.
I could put together a similar document for the esp8266 using gdbstub-serial debugging.  
Will it be published as part of an open source document?

devyte, 2018-01-24:
I said:

> have a written guide included in our docs

My intent is to have it in a doc here in the repo, to be visible from readthedocs, so yes. There is a [Troubleshooting doc ](https://github.com/esp8266/Arduino/tree/master/doc/Troubleshooting) already with a debugging section, it could be added as an addition there.

freeck, 2018-01-25:
OK. I will translate my doc into English, update it and send a link as soon as possible.

freeck, 2018-01-25:
@weswitt @devyte 
BUT are we talking about the same subject? 
The doc section @devyte  refers to seems to handle post-mortem/crash/panic debugging info, not?
I am refering to real source level debugging....which option was available with previous versions of gdbstub. As the Arduino-IDE does not support this option, I use Eclipse/Sloeber combined with an older version of the gdbstub sources..... See also this topic: https://github.com/esp8266/Arduino/issues/2663

devyte, 2018-01-25:
@weswitt @freeck we're talking about the same. Both using gdb and building with debug messages are debugging, they're just different aspects of it. There are other methods as well.
To be clear, I meant a new doc describing how to set up gdbstub for use with gdb running on a host computer, and link to that new doc from the current debugging section. Yes Arduino IDE doesn't support integrated gdb, but there are other gdb frontends out there, if command line is too hairy. And if command line gdb works, then the other frontends will work as well.

freeck, 2018-01-25:
I will complete an instruction document asap. What I still do not understand why the core-team changed gdbstub in such a way that it does not support source level debugging.... See discussion:
> As the Arduino-IDE does not support this option, I use Eclipse/Sloeber combined with an older version of the gdbstub sources..... See also this topic: #2663.

weswitt, 2018-01-25:
In looking at the diffs between the 2.4.0 gdbstub and the private copy from @freeck the material changes are in gdbstub-cfg.h for config settings and some minor changes in gdbstub.c. It seems that the changes, if necessary, in gdbstub.c should be pushed into the mast code base for everyone.

freeck, 2018-01-26:
@weswitt I you read topic #2663  it seems to be an intentional decision not to support source level debugging. I refer to my previous question to the core-team (@devyte) .
The changes seems to be subtle, but it interferes with the uart-interrupt handler. Anyway I didn't get it working. I am satisfied with the "old" gdbstub and don't use the panic-handler.

freeck, 2018-02-03:
I prepared a document , and will send a PM to you both for a first review.

freeck, 2018-02-03:
Stupid question: how do I send a PM?

freeck, 2018-02-03:
OK folks I finally found some inspiration to finalize an instruction...it is available at 
https://1drv.ms/f/s!AgjNziQJTfNOhoJzP0uwbECd3NppYA  (onedrive). 
Please feel free to comment on the instruction itself, whether it meets your expectations.

devyte, 2018-02-08:
@freeck I haven't delved into gdbstub and debugging with gdb myself (I'm old school: on embedded devices I use printf everywhere), I'm not aware of the reasons behind the decision either.
I can't access onedrive, blocked due to work policy both on phone and laptop. Could you make a pr please? That way it's also directly open for discussion,  corrections, etc.

freeck, 2018-02-08:
@devyte I reopened the link, do you have access now? (onedrive: https://1drv.ms/f/s!AgjNziQJTfNOhoJztuFNADmz1_F_1g ).
I am "very old school" on embedded systems , I have no idea how to make a PR....with some help I will perhaps get there.

devyte, 2018-02-08:
Nope, my devices are company issued (Free Stuff!) and cloud storage sites are blocked because IT is retarded.
To learn about PRs, I read this first:
https://www.digitalocean.com/community/tutorials/how-to-create-a-pull-request-on-github

freeck, 2018-02-08:
Quite a hassle....a totally other environment what I am used to.
Well I have created something on GitHub:
https://github.com/esp8266/Arduino/commit/445bfe98845733aa3a0d2133dc1452b7b2e14aea#diff-53fc8f98eb1abcfcb44f3d4c4dea6a97
are you able to access this, and the Word-file I attached?

freeck, 2018-02-09:
@devyte  I think I have created a pull request :)
I have placed my contribution to a new entry in the library section and is called 

"GDBStub-source-level-debug".

 No idea whether that is the right location. Please let me know whether you have acces to the files.

freeck, 2018-02-19:
I just read this post: https://github.com/esp8266/Arduino/pull/4386
Seems to be the ultimate serial gdbstub, is it not?
I have stopped working on the instruction on how to install and use gdbstb, as I doubt whether it still relevant.

earlephilhower, 2019-01-16:
Closing as this is better tracked in #5559 where there is a documented example in the docs/ subdir.

closed 2019-01-16 by earlephilhower