esp8266/Arduino #4170: WiFiClient.connected() = true but WiFiClient.write() return 0

Raphyyy, 2018-01-15:
Hi,

I'm very sorry to post this in a quick way, I could be more explicite when I get much time but I'm running out of it right now.

Since 2.4.0, I have issues to send MQTT message trough WiFiClient lib. I didn't make any change except passing from 2.3.0 to 2.4.0.

It runs well for few seconds (for ~20 calls of WiFiClient.write()), then at a random time I get **WiFiClient.write(byte*, uint8_t)** returning 0 for all call.

In my code, I have a condition **if (WiFiClient.connected())** before WiFiClient.write().

On my broker logs, I get all message where WiFiClient.write returning more than 0, but no error or socket error.

My MQTT message are QoS 0. But I get at least the TCP trame response and I'm counting on it to validate the good transmission of the trame.

According to the MQTT protocol, if there is any problem in the transmission, the TCP tunnel get broken. So WiFiClient.connected() should return false.

So here's my question : Is there any better way to check the TCP trame response other than the function write() ?

d-a-v, 2018-01-15:
There is currently a bug with WiFi.status() and lwip v2 in 2.4.0 release, which is fixed in the git version. 
You can try again with lwip v1.4 (choice is in Tools menu).
If it works with lwip v1.4, then could you please try the git version with lwip v2 ?
If it does not, we will welcome a simple sketch and steps needed to reproduce the error.

Raphyyy, 2018-01-16:
Thank you :)
I use PlatformIO as IDE.
I set my configuration file as followed here : http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version
This normally fetch the latest dev Arduino ESP8266 core.

The problem is that if the bug is with WiFi.status() like you said, it looks like the socket get broken some minutes after my code running, and this is systematic (after 1-2 minutes running). Or I haven't that bug with 2.3.0. 
This morning I will work on it, keep you in touch

Raphyyy, 2018-01-17:
Hi
I'm allowing me to up this because it became pretty urgent, I use this framework for small professionnal prototype project.
The problem remains the same, few minutes after the program runs, WiFiClient.write(const uint8_t *buf, size_t size) return 0, and I get nothing on my MQTT broker. I tried with another broker test MQTT (broker.hivemq.com), same.
I try to completely reinstance the WiFiClient object before each write(), same.
The strange things is that, each time I try to publish to the broker with the the function write and it return me 0, I reconnect to the broker with function WiFiClient.connect and it works (so the TCP tunnel is establish) but then, I'm unable to establish the MQTT Tunnel with function write() (return me 0).
As it is a professionnal project, it is difficult to me to send you some code, but I will try to build a sketch to reproduce the issue.

ch570512, 2018-01-17:
Hello!
I use WiFi.status(); to check if I'm connected. With lwip v2, after a few minutes/hours, I get status 6 but the connection is still active. If I use lwip v1.4 the problem is gone. Hope this helps a little.
Code is here: http://tmw-it.ch/qlockwork/
Best regards
Thorsten

d-a-v, 2018-01-18:
@Raphyyy @ch570512 can you enable debug output in menus and post the results showing the issue ?

Raphyyy, 2018-01-18:
My problem appear to be caused by something much more weird... See #4193

ch570512, 2018-01-19:
Hello

Here is the WiFi log from the last 6 hours:

`bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 200
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with CDVup55g, channel 6
dhcp client start...
wifi evt: 0
pm open,type:2 0

# From here on Wifi.status() returns 6

bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 200
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with CDVup55g, channel 6
dhcp client start...
wifi evt: 0
pm open,type:2 0
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
`

zmeglic, 2018-05-14:
I have the same issue with 2.4.1.

After a few minutes, all WiFiClient.write()s return 0 and nothing is sent anymore. The same code worked fine on 2.3.0.

devyte, 2018-07-13:
@Raphyyy is this issue still valid with latest git? several fixes have been merged into lwip glue code.

devyte, 2019-11-20:
No feedback in a long while. A lot has changed since the last comment, including many stability fixes that are relevant.
Closing.
If you think the problem is still valid, please open a new issue and follow the template instructions.

closed 2019-11-20 by devyte