esp8266/Arduino #4172: sketch using 2.4.0 is more than 30 kB larger than 2.3.0

pidloop, 2018-01-15:
While working on issue #(4166) I noticed the compiled size of the test program I was using grew by about 34 kB and RAM usage grew about 1.4 kB in version 2.4.0 compared with 2.3.0. While this is not a critical issue, the sudden jump is a concern. Note there was only a very small increase from 2.2.0 to 2.3.0. I wonder if there is a particular reason for this latest increase and whether some way might be found to reduce it.

My application ( [here](http://www.clearskyinstitute.com/ham/HamClock) ) already uses about 850 kB and I am hoping to add more features so the loss of 34 kB may eventually become significant.

Here are the sizes reported after setting the ESP Huzzah Board version, using the toy sketch below:

```
Version  Program  RAM
2.2.0    223660  31543
2.3.0    226353  31904
2.4.0    260583  33384

```

`
#include <stdlib.h>
#include <string.h>
#include <math.h>

#include <SPI.h>
#include <ESP8266WiFi.h>
#include <WiFiServer.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>

char ssid[] = "clearskyinstitute";
char pw[] = "from651plan";

void setup()
{
    Serial.begin(115200);

    WiFi.enableSTA(true);

    WiFi.setAutoConnect (true);
    WiFi.setAutoReconnect (true);

    WiFi.begin (ssid, pw);

    Serial.println("start");
}

void loop()
{
    prWiFiStatus(WiFi.status());

    if (WiFi.isConnected())
        Serial.println (WiFi.localIP());

    delay(2000);
}

void prWiFiStatus (int s)
{
    #define     VALCASE(x)      case x: Serial.println(#x); break;

    switch (s) {
    VALCASE(WL_NO_SHIELD);
    VALCASE(WL_IDLE_STATUS);
    VALCASE(WL_NO_SSID_AVAIL);
    VALCASE(WL_SCAN_COMPLETED);
    VALCASE(WL_CONNECTED);
    VALCASE(WL_CONNECT_FAILED);
    VALCASE(WL_CONNECTION_LOST);
    VALCASE(WL_DISCONNECTED);
    default: Serial.println(s); break;
    }
}

`

earlephilhower, 2018-01-15:
I just built it against GIT head and got your #s for LWIP v1.4 precompiled w/in epsilon:

````
LWIP v1.4 - precompiled

Build options changed, rebuilding all
Sketch uses 260579 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33332 bytes (40%) of dynamic memory, leaving 48588 bytes for local variables. Maximum is 81920 bytes.
````

The top ~100 symbols in size are:
````
earle@server:~/src/newlib-xtensa/newlib/libc/include$ readelf -a  /tmp/arduino_build_622381/pidloop.ino.elf | sort -r -k3 -n | head -100
   679: 3ffeea00  4136 OBJECT  GLOBAL DEFAULT    3 g_cont
   539: 40204304  3446 FUNC    GLOBAL DEFAULT    4 _strtod_r
  1504: 402053b8  3282 FUNC    GLOBAL DEFAULT    4 _dtoa_r
  2018: 4022631c  2687 FUNC    GLOBAL DEFAULT    4 tcp_input
    94: 402259ec  2217 FUNC    LOCAL  DEFAULT    4 tcp_receive
  1315: 3ffed30c  1712 OBJECT  GLOBAL DEFAULT    3 g_ic
   861: 4020e0a4  1428 FUNC    GLOBAL DEFAULT    4 ieee80211_send_mgmt
    75: 4021ff70  1302 FUNC    LOCAL  DEFAULT    4 dhcp_recv
  1200: 4020311c  1284 FUNC    WEAK   DEFAULT    4 _printf_float
   494: 4020c8f4  1270 FUNC    GLOBAL DEFAULT    4 ieee80211_parse_beacon
  2057: 40203adc  1212 FUNC    WEAK   DEFAULT    4 _scanf_float
   744: 4020a81c  1201 FUNC    GLOBAL DEFAULT    4 hostap_input
  1942: 402063e4  1192 FUNC    GLOBAL DEFAULT    4 __gethex
  1186: 40231590  1136 FUNC    GLOBAL DEFAULT    4 register_chipv6_phy
   267: 3ffefa94  1120 OBJECT  LOCAL  DEFAULT    3 dns_table
  1077: 40222ce4  1055 FUNC    GLOBAL DEFAULT    4 tcp_output
   833: 402302b8  1048 FUNC    GLOBAL DEFAULT    4 phy_bb_rx_cfg
  1176: 40209320  1046 FUNC    GLOBAL DEFAULT    4 rijndaelEncrypt
  1884: 40214a6c  1018 FUNC    GLOBAL DEFAULT    4 __wpa_send_eapol
   705: 402145f8   975 FUNC    GLOBAL DEFAULT    4 wpa_receive
  1931: 40203728   947 FUNC    GLOBAL DEFAULT    4 _printf_i
  1218: 40210ad0   904 FUNC    GLOBAL DEFAULT    4 sta_input
  1433: 40210740   902 FUNC    GLOBAL DEFAULT    4 ieee80211_sta_new_state
   890: 402013d4   887 FUNC    WEAK   DEFAULT    4 pinMode
   578: 402013d4   887 FUNC    GLOBAL DEFAULT    4 __pinMode
  2046: 40210300   885 FUNC    GLOBAL DEFAULT    4 scan_parse_beacon
  1298: 40227c70   864 FUNC    GLOBAL DEFAULT    4 ets_vsnprintf
  1618: 4022fc40   856 FUNC    GLOBAL DEFAULT    4 phy_dig_spur_set
  1370: 4022e190   847 FUNC    GLOBAL DEFAULT    4 set_rx_gain_cal_iq
  1302: 40101f8c   836 FUNC    GLOBAL DEFAULT    5 wDev_ProcessFiq
  1697: 4022e6bc   814 FUNC    GLOBAL DEFAULT    4 set_rx_gain_testchip_50
   837: 40208430   800 FUNC    GLOBAL DEFAULT    4 __sfvwrite_r
   853: 40222834   785 FUNC    GLOBAL DEFAULT    4 tcp_write
  1452: 402097e0   768 FUNC    GLOBAL DEFAULT    4 rijndaelKeySetupEnc
  1742: 40232e64   762 FUNC    GLOBAL DEFAULT    4 tx_pwctrl_cal
   848: 4022d3ec   747 FUNC    GLOBAL DEFAULT    4 ram_set_channel_freq
  1997: 402079c4   732 FUNC    GLOBAL DEFAULT    4 _vfprintf_r
  1841: 402079c4   732 FUNC    GLOBAL DEFAULT    4 _vfiprintf_r
  1258: 40218898   730 FUNC    GLOBAL DEFAULT    4 umm_info
    55: 4021f3ac   720 FUNC    LOCAL  DEFAULT    4 handle_dhcp
   477: 40228f34   686 FUNC    GLOBAL DEFAULT    4 get_flash_bin_len
   627: 4021da28   679 FUNC    GLOBAL DEFAULT    4 pm_onBcnRx
  1666: 40221e04   679 FUNC    GLOBAL DEFAULT    4 tcp_slowtmr
  1947: 40230d94   666 FUNC    GLOBAL DEFAULT    4 register_chipv6_phy_init_
  1293: 4021f118   655 FUNC    GLOBAL DEFAULT    4 wifi_softap_dhcps_client_
   746: 40207610   637 FUNC    GLOBAL DEFAULT    4 _svfprintf_r
  1683: 40207610   637 FUNC    GLOBAL DEFAULT    4 _svfiprintf_r
   693: 4020d330   625 FUNC    GLOBAL DEFAULT    4 ieee80211_output_pbuf
  1445: 4020dddc   620 FUNC    GLOBAL DEFAULT    4 ieee80211_send_probereq
  1720: 402328d8   618 FUNC    GLOBAL DEFAULT    4 ram_rfcal_txiq
  1118: 4020e648   614 FUNC    GLOBAL DEFAULT    4 ieee80211_alloc_proberesp
  2031: 40224ed0   608 FUNC    GLOBAL DEFAULT    4 ip_input
  1394: 4022da40   605 FUNC    GLOBAL DEFAULT    4 chip_v6_rf_init
   917: 402348f8   602 FUNC    GLOBAL DEFAULT    4 pm_wakeup_init
  1865: 4022f030   602 FUNC    GLOBAL DEFAULT    4 tx_atten_set_interp
  1461: 402130ac   593 FUNC    GLOBAL DEFAULT    4 cnx_node_join
  1453: 40105a80   582 FUNC    GLOBAL DEFAULT    5 phy_adc_read_fast
[... 20 lines cut ...]
  1895: 40233998   458 FUNC    GLOBAL DEFAULT    4 ram_rxiq_get_mis
  1171: 4022d6f0   452 FUNC    GLOBAL DEFAULT    4 chip_60_set_channel
  1272: 4022cd8c   446 FUNC    GLOBAL DEFAULT    4 readvdd33
  1074: 40233598   446 FUNC    GLOBAL DEFAULT    4 ram_rfcal_txcap
  1377: 40228180   445 FUNC    GLOBAL DEFAULT    4 eagle_lwip_if_alloc
  2033: 4021b498   441 FUNC    GLOBAL DEFAULT    4 rcUpdatePhyMode
   432: 40232b44   438 FUNC    GLOBAL DEFAULT    4 rc_cal
  1203: 4023239c   433 FUNC    GLOBAL DEFAULT    4 tx_pwr_backoff
  1135: 4023103c   430 FUNC    GLOBAL DEFAULT    4 change_bbpll160_sleep
    67: 4021f778   427 FUNC    LOCAL  DEFAULT    4 dhcp_create_msg
   637: 4020ff54   427 FUNC    GLOBAL DEFAULT    4 scan_profile_check
  1674: 40228b18   426 FUNC    GLOBAL DEFAULT    4 system_restart_enhance
  1147: 40233b64   420 FUNC    GLOBAL DEFAULT    4 ram_rxiq_cover_mg_mp
   785: 4021a588   419 FUNC    GLOBAL DEFAULT    4 freq_change_check_scan_do
  1364: 40236b0c   419 FUNC    GLOBAL DEFAULT    4 espconn_udp_sendto
  1886: 4021e4ac   417 FUNC    GLOBAL DEFAULT    4 pm_send_nullfunc
    70: 4021fa8c   412 FUNC    LOCAL  DEFAULT    4 dhcp_select
  1166: 4020baf4   412 FUNC    GLOBAL DEFAULT    4 wifi_softap_stop
  1325: 4022ee90   406 FUNC    GLOBAL DEFAULT    4 tx_pwctrl_init_cal
  1908: 402177fc   405 FUNC    GLOBAL DEFAULT    4 wpa_parse_wpa_ie_rsn
  1741: 40236cec   399 FUNC    GLOBAL DEFAULT    4 espconn_udp_sent
  1382: 40217994   398 FUNC    GLOBAL DEFAULT    4 wpa_parse_wpa_ie_wpa
  1709: 402022b0   392 FUNC    GLOBAL DEFAULT    4 _ZN19ESP8266WiFiSTAClass5
earle@server:~/src/newlib-xtensa/newlib/libc/include$ 
````

earlephilhower, 2018-01-15:
In 2.3.0 I again get your #s:
````
Sketch uses 226329 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31856 bytes (38%) of dynamic memory, leaving 50064 bytes for local variables. Maximum is 81920 bytes.
````

Looking at the top 100 symbols shows:
````
earle@server:~/Arduino/hardware/esp8266com/esp8266$ readelf -a  /tmp/arduino_build_467376/pidloop.ino.elf | sort -r -k3 -n | head -100
   538: 3ffee430  4136 OBJECT  GLOBAL DEFAULT    3 g_cont
  1724: 40225c10  2432 FUNC    GLOBAL DEFAULT    4 tcp_input
   100: 402252e4  2217 FUNC    LOCAL  DEFAULT    4 tcp_receive
  1101: 3ffecefc  1660 OBJECT  GLOBAL DEFAULT    3 g_ic
   705: 4020cab4  1416 FUNC    GLOBAL DEFAULT    4 ieee80211_send_mgmt
    67: 4021f7c4  1302 FUNC    LOCAL  DEFAULT    4 dhcp_recv
   995: 4022b480  1189 FUNC    GLOBAL DEFAULT    4 register_chipv6_phy
   371: 4020b2f0  1127 FUNC    GLOBAL DEFAULT    4 ieee80211_parse_beacon
   181: 3ffef4e0  1120 OBJECT  LOCAL  DEFAULT    3 dns_table
   600: 40209408  1103 FUNC    GLOBAL DEFAULT    4 hostap_input
   896: 40222584  1059 FUNC    GLOBAL DEFAULT    4 tcp_output
   677: 40229fdc  1048 FUNC    GLOBAL DEFAULT    4 phy_bb_rx_cfg
   986: 402170ac  1046 FUNC    GLOBAL DEFAULT    4 rijndaelEncrypt
  1610: 4021324c  1015 FUNC    GLOBAL DEFAULT    4 __wpa_send_eapol
   347: 40106444  1003 FUNC    GLOBAL HIDDEN     5 __divdi3
   562: 40212de8   961 FUNC    GLOBAL DEFAULT    4 wpa_receive
  1207: 4020ee20   903 FUNC    GLOBAL DEFAULT    4 ieee80211_sta_new_state
  1019: 4020f1b0   898 FUNC    GLOBAL DEFAULT    4 sta_input
  1751: 4020e9e4   881 FUNC    GLOBAL DEFAULT    4 scan_parse_beacon
   758: 40202a6c   870 FUNC    GLOBAL HIDDEN     4 __umoddi3
  1088: 40218398   864 FUNC    GLOBAL DEFAULT    4 ets_vsnprintf
   734: 40201120   862 FUNC    WEAK   DEFAULT    4 pinMode
   443: 40201120   862 FUNC    GLOBAL DEFAULT    4 __pinMode
  1373: 40229960   856 FUNC    GLOBAL DEFAULT    4 phy_dig_spur_set
  1153: 40227ef4   847 FUNC    GLOBAL DEFAULT    4 set_rx_gain_cal_iq
  1092: 4010268c   836 FUNC    GLOBAL DEFAULT    5 wDev_ProcessFiq
   470: 4022a4e4   827 FUNC    GLOBAL DEFAULT    4 phy_gpio_cfg
  1438: 40228420   814 FUNC    GLOBAL DEFAULT    4 set_rx_gain_testchip_50
   697: 402220c4   790 FUNC    GLOBAL DEFAULT    4 tcp_write
  1224: 40217564   768 FUNC    GLOBAL DEFAULT    4 rijndaelKeySetupEnc
  1479: 4022cd7c   762 FUNC    GLOBAL DEFAULT    4 tx_pwctrl_cal
   692: 40227188   747 FUNC    GLOBAL DEFAULT    4 ram_set_channel_freq
    48: 4021cab8   732 FUNC    LOCAL  DEFAULT    4 handle_dhcp
  1411: 40221668   682 FUNC    GLOBAL DEFAULT    4 tcp_slowtmr
  1085: 4021c814   673 FUNC    GLOBAL DEFAULT    4 wifi_softap_dhcps_client_
  1661: 4022ac84   666 FUNC    GLOBAL DEFAULT    4 register_chipv6_phy_init_
   489: 402078b8   662 FUNC    GLOBAL DEFAULT    4 pm_onBcnRx
   552: 4020be24   622 FUNC    GLOBAL DEFAULT    4 ieee80211_output_pbuf
  1460: 4022c7f0   618 FUNC    GLOBAL DEFAULT    4 ram_rfcal_txiq
  1737: 402247b0   608 FUNC    GLOBAL DEFAULT    4 ip_input
  1176: 402277a4   605 FUNC    GLOBAL DEFAULT    4 chip_v6_rf_init
  1593: 40228d34   602 FUNC    GLOBAL DEFAULT    4 tx_atten_set_interp
  1232: 402115b4   597 FUNC    GLOBAL DEFAULT    4 cnx_node_join
   755: 4022e8d0   575 FUNC    GLOBAL DEFAULT    4 pm_wakeup_init
   676: 40229cb8   565 FUNC    GLOBAL DEFAULT    4 phy_dig_spur_prot
  1650: 4022d674   544 FUNC    GLOBAL DEFAULT    4 ram_rfcal_pwrctrl
  1217: 4020c840   531 FUNC    GLOBAL DEFAULT    4 ieee80211_send_probereq
   182: 40220354   527 FUNC    LOCAL  DEFAULT    4 dns_recv
  1310: 4022c5d8   526 FUNC    GLOBAL DEFAULT    4 txiq_cover
   969: 4022dc0c   523 FUNC    GLOBAL DEFAULT    4 ram_rfcal_rxiq
  1055: 40202e08   511 FUNC    GLOBAL DEFAULT    4 umm_info
  1683: 40224a14   510 FUNC    GLOBAL DEFAULT    4 ip_output_if_opt
   558: 4022946c   505 FUNC    GLOBAL DEFAULT    4 noise_init
   495: 40223cf8   503 FUNC    GLOBAL DEFAULT    4 ethernet_input
   321: 402289a4   490 FUNC    GLOBAL DEFAULT    4 tx_cap_init
  1512: 402230b4   490 FUNC    GLOBAL DEFAULT    4 udp_input
   749: 40206504   488 FUNC    GLOBAL DEFAULT    4 esf_buf_alloc
   940: 40224d14   487 FUNC    GLOBAL DEFAULT    4 ipaddr_aton
  1036: 4020d4fc   486 FUNC    GLOBAL DEFAULT    4 ieee80211_freedom_output
    87: 4021e2b0   477 FUNC    LOCAL  DEFAULT    4 espconn_Task
   932: 4020d05c   471 FUNC    GLOBAL DEFAULT    4 ieee80211_alloc_proberesp
[... 15 lines cut ...]
  1450: 40201fa8   420 FUNC    GLOBAL DEFAULT    4 _ZN19ESP8266WiFiSTAClass5
  1478: 4021ec9c   419 FUNC    GLOBAL DEFAULT    4 espconn_udp_sent
   983: 40227484   412 FUNC    GLOBAL DEFAULT    4 chip_60_set_channel
    62: 4021f2e0   412 FUNC    LOCAL  DEFAULT    4 dhcp_select
   978: 4020a4c8   408 FUNC    GLOBAL DEFAULT    4 wifi_softap_stop
   927: 4020d8ec   406 FUNC    GLOBAL DEFAULT    4 ieee80211_beacon_alloc
  1320: 40223a64   406 FUNC    GLOBAL DEFAULT    4 etharp_query
  1632: 402160d4   404 FUNC    GLOBAL DEFAULT    4 wpa_parse_wpa_ie_rsn
   834: 40219484   401 FUNC    GLOBAL DEFAULT    4 system_deep_sleep_instant
   707: 401056ac   397 FUNC    GLOBAL DEFAULT    5 spi_flash_issi_enable_QIO
   647: 40211afc   392 FUNC    GLOBAL DEFAULT    4 ieee80211_send_action_ven
  1219: 40229740   392 FUNC    GLOBAL DEFAULT    4 chip_v6_set_chan_rx_cmp
  1007: 4022c2dc   390 FUNC    GLOBAL DEFAULT    4 tx_pwr_backoff
  1611: 40208234   389 FUNC    GLOBAL DEFAULT    4 pm_send_nullfunc
[... cut ...]
   746: 4021c2cc   381 FUNC    GLOBAL DEFAULT    4 dhcps_start
   858: 40216d0c   377 FUNC    GLOBAL DEFAULT    4 eagle_auth_done
   643: 40223ef0   376 FUNC    GLOBAL DEFAULT    4 icmp_input
  1148: 40227a58   374 FUNC    GLOBAL DEFAULT    4 test_tout
  1766: 40204204   373 FUNC    GLOBAL DEFAULT    4 ppTxPkt
  1455: 40215a44   371 FUNC    GLOBAL DEFAULT    4 wpa_sm_rx_eapol
  1111: 40228bbc   365 FUNC    GLOBAL DEFAULT    4 tx_pwctrl_init_cal
   381: 40105148   364 FUNC    GLOBAL DEFAULT    5 ets_timer_arm_new
````

earlephilhower, 2018-01-15:
New library routines were included in 2.4:
````
   539: 40204304  3446 FUNC    GLOBAL DEFAULT    4 _strtod_r
  1504: 402053b8  3282 FUNC    GLOBAL DEFAULT    4 _dtoa_r
  1200: 4020311c  1284 FUNC    WEAK   DEFAULT    4 _printf_float
  2057: 40203adc  1212 FUNC    WEAK   DEFAULT    4 _scanf_float
````
...and others...

Looks like two smoking guns:  strtod/dtoa, and *floating point* functions were included in 2.4 in the main core libraries!  Not sure there's a good reason to have floating point in the core proper...

earlephilhower, 2018-01-15:
Actually, now that I think about it, I do know why.  Moving from the SDK libs to newlib.c added floating point format support to printf, which was a big sticking point for lots of folks.

Therefore, any scanf()/printf() now supports %f by, you guessed it, importing the floating point math library and the floating point string parser...

pidloop, 2018-01-15:
Here are the numbers for my real application:

`
2.3.0 802057  44684
2.4.0 830271  45820
`

I certainly use tons of FP and do hand-stands to avoid sprintf %f. Are saying I can now start using %f and not expect the program to grow much?

earlephilhower, 2018-01-15:
I think so.  It's not going to import things like powf() or logf() unless you use them, so there may be some increase, but it seems like you're already getting a good chunk of code no matter what. your own code uses.

earlephilhower, 2018-01-15:
You may also want to run the readelf -a commands I listed above on your real app to see if there's something else not accounted for...

pidloop, 2018-01-16:
changing all dtostrf to %f reduced 2.4.0 size by 1k: 830271 -> 829719.

But it's still 30k bigger than 2.3.0: 800313. Since they both included or pulled-in %f I don't think that's the reason.

I'm almost more worried by the 1.3k increase in precious RAM used in 2.4.0: 44556 -> 45824

5chufti, 2018-01-16:
@devyte seems to be similar to #4089 , couldn't it be handled in one issue?

pidloop, 2018-01-16:
Duplicate of 4089. Closing.

jonhp, 2018-01-16:
Isn't there an argument to be made that everyone shouldn't suffer the flash and memory increases for including floating point support by default. (Does the ESP8266EX even have floating point hardware?) There are possible designs where floating point is avoided within the ESP8266 altogether (because it is a performance hit). I have seen many code sequences where floating point is avoided by using fixed-point or simply refactored the floating point computations to another node (like web browser) which is more likely to have floating point support.

pidloop, 2018-01-16:
I think FP is a red herring. The code is larger for some other reason.

5chufti, 2018-01-16:
yes, usually linker only include actually called/used routines into final binary.

emontnemery, 2018-01-21:
@pidloop Is it really a red herring? If printf / scanf supports %f, how could  the linker ever remove float formatting/parsing support?

pidloop, 2018-01-21:
On Sun, Jan 21, 2018 at 11:36 AM, emontnemery <notifications@github.com>
wrote:

> @pidloop <https://github.com/pidloop> Is it really a red herring? If
> printf / scanf supports %f, how could the linker ever remove float
> formatting/parsing support?
>
>
>
Because my sketch is about 30 kB larger whether I use dtoa/strtod or
printf/scanf with %f.

devyte, 2018-01-25:
Something that popped out at me:
2.4:   1885: 40216f44  4058 FUNC    GLOBAL DEFAULT    4 __ieee754_pow
vs. 
 2.3:  1358: 4021ca68  2898 FUNC    GLOBAL DEFAULT    4 __ieee754_pow

Anyone know why a function like this could grow over 1KB? There are others that grew, but usually only a few bytes.

earlephilhower, 2018-01-25:
The compiler doesn't know if you use %f in a string, so you have to use the linker options, @pidloop , to remove the printf_float/scanf_float bits.

@devyte I think you have the versions backwards.  2.4 is 2898 bytes, 2.3 is 4058.  We saved 1KB here. :)

````
V2.4 since it had strtod_r
...
  2149: 40216444  3446 FUNC    GLOBAL DEFAULT    4 _strtod_r
...
  1358: 4021ca68  2898 FUNC    GLOBAL DEFAULT    4 __ieee754_pow

V2.3
  1885: 40216f44  4058 FUNC    GLOBAL DEFAULT    4 __ieee754_pow
````

devyte, 2018-01-26:
@earlephilhower oops you're right, the windows editor I used opens the panes in reverse order to what I'm used to

Hiddenvision, 2018-01-29:
My two bobs worth.
I found 2.4 compiles to be larger that 2.3.!

But can anyone explain why there is so much space used for Raw "un-displayed" text.
Function names, Full Paths, other non critical info littered within the Bin.
We should be blessed that comments are not also included.

It makes me want to use single letter function names and keep all the libs in a short folder name off root.

Can any smartie tell me how to remove all the "junk" from a compiled file.?
Guessing it is something in the linker or compiler that causes this.

romansavrulin, 2018-03-09:
@Hiddenvision compile with NDEBUG, -Os options, strip resulted elf. and you will get rid of it

Hiddenvision, 2018-03-09:
@romansavrulin, Cheers been using that for a while and it removes a little more..

pidloop, 2018-03-09:
All of these suggestions are completely missing the point. The exact same program compiles 30k larger on 2.4 than 2.3. The question is whether this is intentional (and if so why) or whether there is something wrong and needs to be corrected. Work arounds are NOT the answer here.

earlephilhower, 2018-03-09:
Not sure where you're going with this, @pidloop, but the code size change, in certain sketches, is the result of intentional decisions to add requested support in the standard libraries.  There is nothing wrong that needs to be corrected.

However, there are obviously some sketches in which this causes them to cross some line making them hard/impossible to OTA update.  In those cases, the workarounds above will get you as close as possible to old behavior.

If you really are in a tight bind, then you can always stick with 2.3.0.

Hiddenvision, 2018-03-09:
@earlephilhower 
the 500k limit is not a problem now and I think the size increase is as you say extra functions.

pidloop, 2018-03-09:
@earlephilhower

My point is if the increases in 2.4 are well thought out and _intentional_ then that's fine. I just wanted to encourage the maintainers to beware of unintentional code creep. Since all the discussions I have seen simply try to _explain_ or _mitigate_ the increase, rather than admitting it might need more thought, I can only assume the maintainers believe this increase in default size is justified and I will (must) accept their judgement.

Regarding OTA, after switching to 4M(1M SPIFFS) this is not an issue for me any longer.

Thanks.

earlephilhower, 2018-03-09:
Glad you're OK now!  I believe this tracker can be closed now.

The fact that code grew in some cases was shown, the reasons behind it explained, and several mitigation options were listed including @d-a-v's script (which is now in the repo) to remove %f format support.

closed 2018-03-09 by earlephilhower