esp8266/Arduino #4175: RS-485 (Modbus Rtu) half duplex communication/Arduino-uno

MohdShanawaz, 2018-01-16:
I am a beginner. I am trying to communicate with a device and the modbus address is ranging from 40200 to 40380. I am using max 485 ic to convert rtu to ttl. But on the serial monitor it is showing some garbage value. Is something missing in the code. I have seen this logic as per this video  https://www.youtube.com/watch?v=tBw15SfmuwI&t=83s. Note. Half duplex communication.

----------------------------------------------------------------------------------------------

#include <ModbusMaster.h>

/*!
  We're using a MAX485-compatible RS485 Transceiver.
  Rx/Tx is hooked up to the hardware serial port at 'Serial'.
  The Data Enable and Receiver Enable pins are hooked up as follows:
*/
#define MAX485_DE      3
#define MAX485_RE_NEG  2

// instantiate ModbusMaster object
ModbusMaster node;

void preTransmission()
{
  digitalWrite(MAX485_RE_NEG, 1);
  digitalWrite(MAX485_DE, 1);
}

void postTransmission()
{
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
}

void setup()
{
  pinMode(MAX485_RE_NEG, OUTPUT);
  pinMode(MAX485_DE, OUTPUT);
  // Init in receive mode
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);

  // Modbus communication runs at 9600 baud
  Serial.begin(9600);

  // Modbus slave ID 2
  node.begin(1, Serial);
  // Callbacks allow us to configure the RS485 transceiver correctly
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);
}

void loop() 
{
  uint8_t resultMain;

  resultMain = node.readInputRegisters(0X40200,56);
  if (resultMain == node.ku8MBSuccess)
  {
    Serial.println ("-----------------------------------------------------------");
//    Serial.print ("1. VR: ");
//    Serial.println(node.getResponseBuffer(0x00)/100.0f);
//    Serial.print ("2. VY: ");
//    Serial.println(node.getResponseBuffer(0x02)/100.0f);
//    Serial.print ("3. VB: ");
//    Serial.println(node.getResponseBuffer(0x04)/100.0f);
//    Serial.print ("4. IR: ");
//    Serial.println(node.getResponseBuffer(0x16)/100.0f);
//    Serial.print ("5. IY ");
//    Serial.println(node.getResponseBuffer(0x18)/100.0f);
//    Serial.print ("6. IB ");
//    Serial.println(node.getResponseBuffer(0x20)/100.0f);
//    Serial.print ("7. KWr ");
//    Serial.println(node.getResponseBuffer(0x46)/100.0f);
//    Serial.print ("8. KWy ");
//    Serial.println(node.getResponseBuffer(0x48)/100.0f);
//    Serial.print ("9. KWb ");
//    Serial.println(node.getResponseBuffer(0x50)/100.0f);
//    Serial.print ("10. KVAr ");
//    Serial.println(node.getResponseBuffer(0x52)/100.0f);
//    Serial.print ("11. KVAy: ");
//    Serial.println(node.getResponseBuffer(0x54)/100.0f);
//    Serial.print ("12. KVAb: ");
//    Serial.println(node.getResponseBuffer(0x56)/100.0f);
//    Serial.print ("13. SysKW: ");
//    Serial.println(node.getResponseBuffer(0x40)/100.0f);
//    Serial.print ("14. SysKVA: ");
//    Serial.println(node.getResponseBuffer(0x42)/100.0f);
//    Serial.print ("15. Freq.: ");
//    Serial.println(node.getResponseBuffer(0x30)/100.0f);
  }
  delay (1000);
}

MohdShanawaz, 2018-01-16:
Also note that the format given for this addresses is U-Long and Long.

tonton81, 2018-01-16:
you should post on forum rather than ask for troubleshooting in a bug report issue..

devyte, 2018-01-16:
Not the right place to request help with your particular project.
Closing per issue POLICY doc.

closed 2018-01-16 by devyte

icandura, 2018-01-27:
I think the 0x40200 is not a real address in MODBUS.
The 0x40200 address in PLC, means that real modbus address is 0xC7.