esp8266/Arduino #4177: development around the AT firmware: ESP8266AT library

aster94, 2018-01-17:
I know that this is not the right place where to post this, but here it will get more audience and possibly people who are interested (if espressif is not)

I think that the making of a library which follows the WiFi Arduino API which uses the AT command will be very useful. I think also than an example could explain this better then my poor english:

with the normal ESP8266WiFi library we do:
`client.write("request");`
in the ESP8266AT library it could became:
```
AT+CIPSEND=x
"request"
```

or again 

`WiFi.connect(ssid,password);`
could be a wrapper for:
`AT+CWJAP="my_ssid","my_password"`

do you agree?

devyte, 2018-01-17:
Either you don't understand how both firmwares work, or your explanation is confusing and I didn't understand it.
The AT firmware receives AT cmds over serial port, which are mapped to SDK api calls to manipulate things like the wifi.
Our firmware works by wrapping SDK api calls behind Arduino-compatible api.
If your idea is to implement an AT cmd firmware based on this repo, sure, go ahead. Maybe we could even link to your repo.
In any case, like you said, this is not the right place to discuss this. 
Closing per issue POLICY doc

closed 2018-01-17 by devyte

aster94, 2018-01-17:
Just immagine a "Serial.print" before the AT commands, i didn t wrote it because i thought it was clear but as you said i should have explained it unclearly
The second example becames:
Serial.print("AT+CWJAP="my_ssid","my_password"\r\n");

devyte, 2018-01-17:
I think you have it backwards. Serial.print() sends commands out the serial port to another device, which would have to have an AT cmd interpreter. Why would you want to send AT cmds from one ESP with our firmware to another ESP with the AT firmware?
Unless you mean some different device with a different AT cmd interpreter?

lrmoreno007, 2018-01-17:
I think this is what you are looking for, but please do not ask more about this here.

https://github.com/sparkfun/SparkFun_ESP8266_AT_Arduino_Library

aster94, 2018-01-17:
Yep that was exactly what i was looking for, thank you