esp8266/Arduino #4180: ESP8285 wifi direct

stern0m1, 2018-01-17:
Is there specific support for the ESP8285?
The datasheet for ESP8285 mentions it implements Wifi Direct. Did the arduino library implement it?
Where can I find more information about implementing wifi direct with the ESP8285?
Thanks

igrr, 2018-01-17:
No, there's not support for WiFi direct in Arduino. It's not exposed in the underlying ESP8266 Non-OS SDK either, as far as i can tell.

closed 2018-01-17 by igrr

stern0m1, 2018-01-17:
[image]
What does the datasheet mean by wifi direct? 
Is the underlying sdk the same for ESP8285 as esp8266?

igrr, 2018-01-17:
1) no idea about that, sorry. Please ask on bbs.espressif.com.

2) yes, the SDK is the same

Androbin, 2019-08-25:
Someone seems to have done it:
https://github.com/youmebangbang/esp8266-wifi-direct-examples