esp8266/Arduino #4181: Tone.h is it missing or not needed?

Fettkeewl, 2018-01-17:
### Basic Infos

#### Hardware
Hardware:			ESP-12f
Core Version:      	2.4.0

### Description

Trying to use Tone.h which is written for the esp8266 but I can't include it, I see that there is only a tone.cpp but not a tone.h. How is this then supposed to be used?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <Tone.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
20:18: fatal error: Tone.h: No such file or directory

 #include <Tone.h>

```

MarkyAD, 2018-01-20:
No includes are needed to use tone functions, they're part of the core.
```javascript
#define TONE_PIN D5

void setup() {
 Serial.begin(115200);
}

void loop() {
 tone(TONE_PIN, 5000, 250); // pin, frequency, duration (msec)
 delay(1000);
}

devyte, 2018-01-25:
Previous comment is correct. The tone functions are declared in Arduino.h, and Tone.h doesn't exist. Per arduino.cc reference, this looks correct.
Closing.

closed 2018-01-25 by devyte