esp8266/Arduino #4182: feature request: WiFi.mode(WIFI_AP_STA) / server.send()

Marcianikus, 2018-01-17:
When the server.send() function is used, the same message is sent to all connected clients. 

It could be interesting the possibility of sending independent messages to the AP or the ST connected
exmple: server.**send_AP** ()  /   server.**send_ST** ()

A more generic possibility: server.send (200,"text/html", "message", **IPAddress ip**)

jandrassy, 2018-01-19:
isn't server.send a response to the current http request from one client?

devyte, 2019-11-09:
I can't think of a use case for this. Also, there has been no interest by others since this was opened.
Closing. 
If you still think the request has merit, please open a new issue, follow the template instructions, and explain the use case in detail.

closed 2019-11-09 by devyte