esp8266/Arduino #4184: Can't connect to some routers with lwip2

schlaegerz, 2018-01-17:
### Basic Infos
My partner has a router that he cannot connect the esp8266 if i use the lwip2 library, it works fine with lwip1.4
#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			Node MCU 12E
Flash Size: 		4MB
CPU Frequency:		160
Upload Using:		 SERIAL

### Sketch

```cpp

#include <Arduino.h>
setup()
{
    WiFi.begin("SSID", "password");
    byte i = 0;
    bool failed = false;
    auto start = millis();
   unsigned int timeout_sec = 20
    while (!failed && WiFi.status() != WL_CONNECTED && (millis() - start) < (timeout_sec * 1000)) {

        switch (WiFi.status()) {
        case WL_NO_SSID_AVAIL:
            break;

        case WL_CONNECT_FAILED:
            break;

        default:
            break;
        }

        delay(100);
    }

    Serial.println(WiFi.status());
}
```

This doesnt repro for all routers, and I dont have access to this particular router, so I'm not sure how much we will be able to do, but just want to show  that there is an issue with the lwip2 here.

schlaegerz, 2018-01-22:
I was told this was the router he was using. 

https://www.cnet.com/products/apple-airport-extreme-base-station-fall-2009/review/

Can't quite figure out why it is not able to connect with LWIP2 but is with LWIP1

d-a-v, 2018-03-08:
@schlaegerz can you try release 2.4.1 and give more details ?
Your "failed" variable is a constant set to "false" in the above sketch.

devyte, 2018-07-13:
Closing due to lack of feedback in several months.

closed 2018-07-13 by devyte