esp8266/Arduino #4185: Change request : SPIFFS autoformat on mount failure

vdeconinck, 2018-01-17:
This issue to discuss the opportunity to change the default behaviour of SPIFFS mount() and begin().

In the current implementation, if the mounting filesystem fails, the API silently formats the filesystem and mounts it - empty.

Pro: after mount, the programmer is guaranteed to have a mounted filesystem

Con: if the chip is defective, or the API has a bug, or the FS has been formated with incompatible configuration, data gets lost without a warning. This is also contrary to the default behaviour chosen for ESP32

I would suggest to adopt the same convention as the ESP32: SPIFFS.begin() should accept an optional "formatOnFail" parameter, defaulting to false.

This would be breaking change (albeit in the direction of safety), so I guess it would have to be targeted to the next major version,

References: 
https://github.com/esp8266/Arduino/issues/2581#issuecomment-358132613
https://github.com/esp8266/Arduino/issues/4111#issuecomment-355855993
https://github.com/espressif/arduino-esp32/issues/411#issuecomment-329017296

devyte, 2018-01-24:
@igrr do we want to change this to match the esp32 behavior? If yes, then this should likely be targetd for 3.0.0 release.

earlephilhower, 2019-01-16:
#5525 will fix this, adding to tracking here.

earlephilhower, 2019-01-22:
Removing @igrr since this should be in the next revision (with the changes already in a PR).

earlephilhower, 2019-04-15:
Fixed in #5525 , there are now config helpers to `setAutoFormat(bool)` for filesystems.

closed 2019-04-15 by earlephilhower