esp8266/Arduino #4188: WifiClientSecure - Certificates not being sent upon reconnection when using PubSubClient

ztittle, 2018-01-18:
Due to some recent changes in the WifiClientSecure library, I am seeing that when using the PubSubClient library that it fails to reconnect. Upon inspection via WireShark, certificates are no longer being resent.

After some research, I have found the breaking change is occurring in the stop function of WifiClientSecure:

https://github.com/esp8266/Arduino/blob/b08ff1026990c18906c6751a7e107431f218b1d9/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp#L656

It invokes unref on the SSLContext, which is freeing the underlying ssl context. From a consumer's point of view, there is no good way to know when to invoke the loadCertificate function after the initial connect. 

In the PubSubClient, it invokes stop() in the main program loop. 

https://github.com/knolleary/pubsubclient/blob/dddfffbe0c497073d960f3b9f83c8400dc8cad6d/src/PubSubClient.cpp#L288

Since loadCertificate() was already invoked on the WifiClientSecure instance, I would not expect it to be invoked again for the lifetime of the instance. I was able to reconnect by calling loadCertificate before connect, but this eats up memory and occasionally prints "Error: maximum number of certs added (1) - change of compile-time configuration required" as the consumer does not know if certs should be reloaded.

I am thinking this change to stop() should be reverted or there needs to be a better way to signal that certs should be re-loaded because the underlying ssl context has changed. If the change is reverted, there are still potential issues because unref is also invoked on write and connect.

Example program https://github.com/copercini/esp8266-aws_iot/blob/master/examples/mqtt_x509_DER/mqtt_x509_DER.ino showing loadcertificate only invoked once during setup.

earlephilhower, 2018-01-18:
@ztittle That's my fault.  I'm overloading the ClientSecure for use as a ServerSecure, and in that use case when a server.stop() happens, we need to free the SSL context as it'll be rebuilt and renegotiated on the next connection.

I'll throw together and test a PR on the server side and update here.  If you could simply comment those 2 lines out in your own sources and confirm it fixes your issue in parallel that would streamline things.

ztittle, 2018-01-18:
Commenting out the two lines does indeed resolve this specific reconnect issue for client side mutual SSL. However, I do see a slight possibility of this issue resurfacing if there is a handshake failure or server sends a close notify on write.

For now, I am explicitly invoking stop() and loading certs before invoking connect(). Not ideal, but it would be best if loadCertificate() could be called once during setup. WifiClientSecure should handle re-loading the certs if necessary if the underlying context is in an unrecoverable state.

earlephilhower, 2018-01-18:
Yes, after I posted this I went back and looked over my code (it was sitting as a PR for ~a year so not fresh in my mind) and I think that' the only sane thing to do.

The underlying problem is there's only enough memory for a single SSL context.  I manage a HTTPS server and a SSL MQTT connection in my app by only allowing one at a time...when I get a 443 connection I have to drop the MQTT connection, and then when I'm done servicing I restart it.  That's what these lines are doing.

You'd have the same issue, I think, talking as a client to two outside SSL servers as well.

Could it be something conceptually as simple as deleting/unref'ing as-is, but on any connect force a reload during SSL object creation?

ztittle, 2018-01-18:
Possibly. I'm not very familiar with the axtls library. but if the underlying ssl context was trashed, would it be possible to copy the loaded certs over to a new ssl context? Perhaps @igrr can chime in.

I looked over the PubSubClient library again and it does have a setClient method. So instead of passing a singleton WifiClientSecure to it in the constructor, which is not reliable across reconnects, I am going to re-create it each time I need to (re)connect. It should avoid any weird state issues and be reliable for my needs. The only downside being additional memory fragmentation overtime since I believe there are some dynamic memory allocations going on with axtls.

earlephilhower, 2018-03-09:
@ztittle Can you verify things work with the 2.4.1 that was just released? The changes to the SecureServer went in a while back, and included a separate context for both a single server and a single client, so this should not be a problem anymore.

devyte, 2018-07-13:
Closing due to lack of feedback in several months.

closed 2018-07-13 by devyte