esp8266/Arduino #4189: SoftWDT immediately after ESP.deepSleep() ? (w/workaround)

NdK73, 2018-01-18:
Hi.
In one of my sketches I had the soft WDT kick in when going to sleep:
```
Soft WDT reset

ctx: cont 
sp: 3fff07e0 end: 3fff09f0 offset: 01b0

Decoding 7 results
0x4022cc78: fpm_open at ?? line ?
0x402091ce: ESP8266WiFiGenericClass::forceSleepBegin(unsigned int) at /home/ndk/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 1954
0x402091b9: ESP8266WiFiGenericClass::forceSleepBegin(unsigned int) at /home/ndk/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 1954
0x40208f72: loop at /home/ndk/Dropbox/Applicazioni/Arduino/tensore/tensore.ino line 180
0x40208fbb: loop at /home/ndk/Dropbox/Applicazioni/Arduino/tensore/tensore.ino line 192
0x4020dc54: loop_wrapper at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109
```

The simplest sketch worked, so it seems another race (ARGH!).
The code to go to sleep (that previously worked) was:
```
    // Disable WiFi
    WiFi.mode(WIFI_OFF);
    WiFi.forceSleepBegin();
    delay(1); //Needed, at least in my tests WiFi doesn't power off without this for some reason

    ESP.deepSleep(curPermCfg.sleep*1000000);
```

Just adding another delay() seems to fix it (I made both longer, but I'm not sure it's needed):
```
    // Disable WiFi
    WiFi.mode(WIFI_OFF);
    delay(10); //Needed, at least in my tests WiFi doesn't power off without this for some reason
    WiFi.forceSleepBegin();
    delay(10); //Needed, at least in my tests WiFi doesn't power off without this for some reason

    ESP.deepSleep(curPermCfg.sleep*1000000);
```

NdK73, 2018-01-19:
That's possible, but that didn't happen with "old" environment (2.3.0, IIRC?).
I **could** (barely) understand a HW WDT trigger (well, if I've put the micro to sleep, how is it supposed to feed the wdt?) but the soft WDT? Shouldn't forceSleepBegin() disable it?

earlephilhower, 2019-07-25:
Closing.  We have an active discussion in #6172 related to the problems w/WiFi powerdowns you should follow.

closed 2019-07-25 by earlephilhower