esp8266/Arduino #4191: Make sure GPIOs remain at a given state even during reset of the ESP

titibon, 2018-01-18:
### Basic Infos
I am looking for info on how to make sure GPIOs remain at a given state even during reset of the ESP

#### Hardware
Hardware:  ESP-12E on nodeMCU V1.0
Core Version:    ? 
 
D7 GPIO13 pin is wired to a NPN transistor (2N2222) base via a 1k resistor.
Collector to 3.3V via a white LED
Emitter grounded.
D5 GPIO14 pin is grounded => the LED if off when in the loop() of the sketch.

### Description

Currently, when the ESP exits from the deepsleep mode and restarts, **the LED goes on for a short while**. Once the sketch is in setup(), of course it goes off, as coded.
**I want to avoid this and force the GPIO13 to go low during the reset process.**

I read in the Expressif datasheet the following text:

**For low power operations, the GPIOs can also be set to hold their state. For instance, when the chip is powered down, all output enable signals can be set to hold low.** 

Chapter 4.1 page 13 -ESP8266EX Datasheet v5.7

I searched on ways to exploit this, but couldn't find any info.
Is it possible to control this behaviour with commands implemented into a library ? Which one ?
Any other clue ?

Many thanks.

### Settings in IDE

Module:  			NodeMCU 1.0 ESP12
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

// INPUT
// GPIO14 - D5 on nodeMCU
#define PIN_PIR 14

//OUTPUT
//  GPIO13 - D7 on nodeMCU
#define LED_TOP     13

int Led                 = LED_TOP;
int pirState;

void setup() {
  
  pinMode(Led, OUTPUT);
  digitalWrite(Led, LOW);
  pinMode(PIN_PIR, INPUT);
 
}

void loop() {
  
  pirState = digitalRead(PIN_PIR);
  if (pirState == 1) {
    digitalWrite(Led, HIGH);
  }
  else {
    digitalWrite(Led, LOW);    
  }

  if (millis() > 5000) {
    digitalWrite(Led, LOW);
    ESP.deepSleep(5000000, WAKE_RF_DEFAULT);
    delay(15);
  }
}
```

### Debug Messages

```
messages here
```

5chufti, 2018-01-18:
This is not the standard behaviour of µCs and not a problem (issue) with the Arduino esp8266 core.
As this is no help or supportforum for esp8266 plattform and according to the strict issue policy #3571, please close this issue. You may ask for help at bbs.espressif.com or www.esp8266.com

titibon, 2018-01-18:
ok

closed 2018-01-18 by titibon