esp8266/Arduino #4193: Unstable system with weird reasons

Raphyyy, 2018-01-18:
Hello

I'm writing here because after month of trouble, I can't get trough this very strange bug that makes my entiere program unstable.

I am a software developper in an IoT company and I use ESP8266 as main controller for a big professionnal project (this going to change in the end of the year). I use the framework Arduino to do so.

First of all, keep in mind that this is not a physical issue. The board has been review by two electronical engineer and after many studies, they could assure me that this can't be caused by any bad electronic conception.

Soo, I stop talking, **here how the problem first appear** : 

In my project I use a SIM808 GSM and GPS module to send data via GSM and get position by GPS by AT commands. I use UART to communicate with him via pins 12 & 14 thanks to SoftwareSerial librairy.

The output problem : 
[image]

Sometime I get corrupted data in my communication between the ESP and the SIM808. So to resolve this I try many, many, many things : 
- Increasing the buffer size of the SoftwareSerial object
- Empty the buffer before sending command
- I tested all bauds speed 
- Updated the SoftwareSerial librairy to make the buffer static allocated
- Try to change the GPS and GSM antenna of SIM808
- Tested on many alimentation, checked the current consumption, etc.
- Monitored the Rx/Tx pins of the SIM808 with an extern Arduino : Absolutely no corrupted data. So the problem was from the ESP Rx PIN.

So I had to developp the entiere program with this possibility to have corrupted communication, so catch all potential problem that could cause.

What really grinds my gears is that whenever I changed an absolutely insignificant thing, the problem seems to decrease. So I made conclusions which was always put back on the table... **Because I didn't know the problem was much more fucked up than I though.**

Recently I had a program running without less corrupted datas. I review my code and only add a Serial.println() in a unused function to debug a potential bug. By adding this line, I get a lot more corrupted data between ESP and SIM808. I reboot, same. I try to flash back the program without the Serial.println(), the problem disapear. I re-add this line, re-flash, the problem shows up again.
**This Serial.println() wasn't even called in the boot process**. 

I try with others insignificant function, like delay(0) or init unused var and put them at some random place in my code, even if these was never called/used. By adding/removing them I get more or less corrupted datas in my communication.

So to have a working build, I always have to let my program run on many hours to check if there was no error and add some delay(0) at some random place to get it working. _This is absolutely insane_.

But recently I realise this problem wasn't just affecting the serial communication between the ESP and the SIM808. It affect the entiere program to make it unstable. 

So whenever I get random crash or a problem with a native librairy, I add delay(0) at some random and unused places, and it goes to work.

I insist on the fact that those delay(0) added won't never be called.

Here's a fucked up example I just had yersterday : 
In the boot process, my system reach an HTTP API via WiFi to check if there is no updates available for him.
In my code I have these line (**which haven't any link with this process**) : 
```
#define FW_VERSION "0.4.8c"
#define HW_VERSION "2.4"
#define ID_BUILD 22
```

And here is the Serial output from this process : 
```
Checking for updates...
POST http://x.x.x.x/boot, DATA : 12345678,326565645458787,2121545465565,4,20
Receive : 0,1516187447
```

Now if I edit those line above by just changing 0.4.8c to 0.4.8d and ID_BUILD 22 to ID_BUILD 23 : 
```
#define FW_VERSION "0.4.8d"
#define HW_VERSION "2.4"
#define ID_BUILD 23
```

Here is the Serial output from the process : 
```
Checking for updates...
POST http://x.x.x.x/boot, DATA : 12345678,326565645458787,2121545465565,4,20
ERROR [HTTP] POST Failed, error: 
Checking for updates...
POST http://x.x.x.x/boot, DATA : 12345678,326565645458787,2121545465565,4,20
ERROR [HTTP] POST Failed, error: send header failed
Checking for updates...
POST http://x.x.x.x/boot, DATA : 12345678,326565645458787,2121545465565,4,20
ERROR [HTTP] POST Failed, error: 
ERROR Unable to fetch API, continue...
```
So this step is skipped, and after few minutes, it shows trouble to send data via WiFi (function WiFiClient.write() return 0 without reason). Definitely unstable.

**So in fact, I just changed 2 chars and it makes my whole program to be unstable.**

Of course, I tried many times to undo/redo this edit. I **always get the same behaviour** on the output.

At least, I had at this time a case so I could test those bugs on differents architecture.
But this case is not unique. I had other strange behaviour, exception 0, 28, 29 causing reset triggered with no reason, resolved by adding unused code in my program.

So all of this let me thought that this is a compiler problem.

However, here some infos : 
- The ESP8266 are chinese ESP12E
- This bugs appear on all boards (sometimes different fabricant)
- I tried with 2.1.0, 2.2.0, 2.3.0, 2.4.0 and latest dev version of the arduino ESP8266 framework. 
- I use PlatformIO to compile my code. I try to compile with Arduino IDE, same (GCC compiler is used by these two, so not a surprise)
- I tried to flash my ESP8266 with differents versions of esptool.
- On the compiling output of Arduino, it says that my program use 31% of the flash memory and 58% for static RAM allocations. So this is not a memory issue.
- In my sketch, I stock all volatile string in flash memory, with F() Macro. I also put println(getFreeHeap()) at many place in my code, definitely not a memory issue.
- I tried with 2 others computers, same.

I am writing this issue today because I get enough of this work method.
Just imagine the pain to develop a program that constantly shows up new errors that have no link with what you're doing...

I am sorry, I can't post my entiere program here due to its professionnal privacy. However I can work on settle a public version and send it by private message on demand. But that's not gonna be easy, if changing 2 characters could cause an entiere program to be unstable, I have to develop a version I can share which shows the same behaviour.

I never seen anything that weird and this is driving me crazy. We have many of these boards that's going to be sold and this is causing me a lot of stress...

If you need any other informations, don't hesitate.

Best regards

devyte, 2018-01-18:
This kind of instability is usually caused by oom, or by mem corruption in the user app.
For oom, there are two cases: out of mem, as in no more heap, or fragmentation. There is a debug option in the Arduino IDE to enablen a message print to the serial port. You also have to set the Debug port.
I can't help much more without an MCVE sketch to reproduce the problem. If you want to discuss sensitive code, I offer to pm me on gitter.

Raphyyy, 2018-01-18:
Hi devyte, thanks for your reply.
Like I said in this massive block above : 

> On the compiling output of Arduino, it says that my program use 31% of the flash memory and 58% for static RAM allocations. So this is not a memory issue.
> In my sketch, I stock all volatile string [...]

I also though it was a memory issue. But the fact that changing only 2 chars goes to make the whole program unstable, make me think it's not. Added to the fact that **adding** uncalled function like delay(0) or unused var solve the unstability definitely makes me think it's not...

pieman64, 2018-01-18:
@Raphyyy but the data you provided regarding memory is not static.

Raphyyy, 2018-01-18:
@pieman64 Which data ? Those defined by #define ? How can this not be static ?

devyte, 2018-01-18:
You could have mem fragmentation. Typical when you do many malloc/free or concatenations in series. 
Or you could be nuking the stack if you're using local buffers. Total stack size is only 4K.
Or you could have 1-off mistakes and write past the end of a buffer. Typical when  forgetting the null term.
Please  come up with a sketch, or point me to a private repo and I'll take a look.

Raphyyy, 2018-01-18:
As I said I do not use dynamic allocation (only in one part, I use String to get back POST argument from HTTP Server object, because the method httpServer.arg() return String... And this part is used only in a particular way, not called in the process I post above).
My buffers are static allocated. Even the SoftwareSerial buffer, I made it static instead of dynamic.

For now I can send you via Gitter a zip of the project with sensitive info hidden. But I do not tested the project with this, considering that changing 2 chars make the whole system unstable.

Thank you for your help and your quickness, this is really nice of you

igrr, 2018-01-18:
Does the issue also happen with HardwareSerial?

Raphyyy, 2018-01-18:
I do not have any corrupted data on monitoring the HardwareSerial debug output.
I also tested by making **Serial1** the debug output and **Serial** the UART for ESP <> SIM808, but I had still sometimes corrupted datas... But as I can tell, it's pretty hard to be sure on the reason, since it's appear to be a problem not directly linked to the library or communication.

devyte, 2018-01-19:
On a quick private review of the sketch, I found several things wrong that are likely to account for the instabilities described:
- Embedded html with a long series of String concatenations using oeprator+() (mem fragmentation)
- Use of strcpy & friends instead of strncpy on strings that are potentially unsafe (potential buffer overruns and corruption)
- Concatenation of strings where the destination size is smaller than the sum of the size sources (potential mem corruption)
- At least one frame stack at 2.5KB from the loop inwards, and the frame kept on going after that (potential stack overflow, since the stack is only 4KB total)
- A huge number of const char * strings that are not in the flash, and hence in RAM, including pretty much the entirety of the embedded html (potential oom, and mem fragmentation)

I didn't find anything specific wrong, but then again I didn't look deeper after the above. 
I suggest a serious restructuring of the application, and rewrite of the affected code above. This is particularly true of the large stack usage and embedded html. I strongly suggest replacing the generation of static html in memory with something else, e.g.: serving static html and js from flash strings or files on spiffs, and generating a tiny json dynamically, then get the browser to do the work of inserting the values into the webpage with javascript..

I consider this user error, so closing.

closed 2018-01-19 by devyte

Hobohome, 2018-05-28:
Hi, I have a similar issue and some of the same code issues as mentioned above.
I'm trying to address these code issues (that I'm sure are leading to the stability problems). 
Quick question...
In regard to using += to concatenate strings of HTML, I am using a reserve() statement before creating the html String - should this not avoid memory fragmentation issues?
eg

```String output;
output.reserve(3550);
output = F("<h4>Relay Status</h4>"); 
output += left(F("Last Switched"));
output += right(formatTime(lastSwitch1),"");
...
```

   (the functions "left() and right() simply return strings with HTML wrapped around for formatting.)

Pablo2048, 2018-05-28:
Hi. Probably not - if you use string concatenatin this way there will be created new strings, then added and memory is released. Better way is to use .concat() method after reserving the memory, like output.concat(left(F("Last Switched"));