esp8266/Arduino #4199: How to connect to a WPA2 network?

kidbreaker22, 2018-01-20:
Hi guys!

Basically I need to connect my ESP8266 in my school's WiFi which is a WPA2 network (SSID, username and password). Whenever I tried looking for a solution or an example I always see one with WPA networks only (SSID and password). Can you guys please tell me how to make a code for this? Everything works in my home WiFi, just need it to connect to the School's Wifi for a project.

Thank you.

matasondrak, 2018-01-21:
The 'issues' are not for this kind of questions.

closed 2018-01-22 by kidbreaker22

devyte, 2018-01-22:
As a side note, you seem to be confusing wpa, wpa2, and wpa2 enterprise. It is the latter that requires username and password.
The underlying support is provided in the Espressif sdk, and I've seen mixed reports about it working.
If you do get it working, an example would be appreciated.