esp8266/Arduino #4201: File base class needed for Arduino

PaulStoffregen, 2018-01-20:
Arduino needs a File (and maybe a Dir) base class, similar to Stream, Print, Client & Server, so that libraries which actually *use* files can be given a hardware-neutral File object.  Ideally, a library like a JPEG decoder would be able to use a File from SPIFFS or SD or Bridge or USB Host Shield or any other library which provides access to files.

As mentioned on #2281, here is a new issue to discuss how this might be achieved.  Ideally, Arduino's core libraries for AVR, SAM, SAMD, and ESP8266 & ESP32, and Teensy, ChipKit, Photon & Energia could all use this same File base class, as we all currently do with Print & Stream, so that users of all our platforms can have File providing and File using library "just work".  :-)

tofrnr, 2018-01-20:
I would appreciate more to have a FILE* base which is cstdio compatible (especially for SD) instead and deprecate all Arduino "File" bucklings.

tonton81, 2018-01-20:
+1 for File and Dir class :)

PaulStoffregen, 2018-01-29:
Just wanted to follow up regarding the idea to implement File more efficiently (mentioned on the Arduino dev mail list).  Is this github issue ok to discuss, or should we use the Arduino dev list?

Even if virtual functions and polymorphism are used, we really do need a File (and maybe Dir) base class in Arduino, so libraries using File can work with any other library providing File.

devyte, 2018-01-30:
It is ok to discuss here. Actually, the ball is in my court atm, I have been asked to provide an outline for a templated design. I already have an ongoing discussion with a work collegue, and I have an idea of what it should look like.
About a base class, you're making the same mistake others make, by thinking that a base class would make File usable by others. In the template world, it is the interface that is important (I.e.: how is the class meant to be used), as opposed to who its ancestor is (what the class is).

earlephilhower, 2018-01-30:
Love the idea of defining an interface, but Stream and Print are presently full classes, not templates.  Would File being a template change the way folks would have to use it (vs. what they have to do today w/a Stream where they just subclass and implement the overrides)?

PaulStoffregen, 2018-02-06:
Any progress or news to share on the alternative to polymorphism?

I don't want to be impatient.  Arduino has used the same File class in SD & Bridge for years.  A few weeks is not much time by comparison.  But I am hoping we can move forward this month (February 2018) with something.  Personally, I am ok with polymorphism & virtual functions if something better is not found.

PaulStoffregen, 2018-02-19:
Are you still working on the a non-polymorphism approach?  If not, maybe it's time to start trying to move forward with the traditional Arduino polymorphism way?

devyte, 2018-02-19:
Yes, it hasn't been dropped. I had an extended trip, so there was a pause, but I'll be working on it again this week.
Please don't become impatient. I think this is the first chance for a full redesign, and the only reason the discussion has come up is because of the many different filesystem implementations out there. If this is adopted, it could set a precedent for improved design guidelines for other Arduino objects.
Let's do things right.

PaulStoffregen, 2018-03-16:
Any more progress on this?

While I want to be patient, soon I need to begin work on supporting USB memory sticks....

PaulStoffregen, 2018-03-21:
Two months have passed, or close to 3 if counting from the beginning of the conversation on Arduino's developer mail list on December 28, 2017, though the suggestion to wait for coming up with a template-based approach started on January 19 and this issue was opened on January 20.

Now we're at March 21.  Will you ever show us this template idea?

I want to begin really working on porting libraries in April.  Is that too soon?  Am I being unreasonable?

Misiu, 2018-04-27:
@PaulStoffregen @devyte would this solve problem with name collision described here: https://github.com/esp8266/Arduino/issues/3819

earlephilhower, 2019-01-16:
#5525 implements a compatible File/Filesystem interface for SPIFFS and SD(SDFS) (and LittleFS in #5511).

When that lands, I think this can be closed.

earlephilhower, 2019-04-15:
Closing since we now have SD, SDFS, SPIFFS, (and soon LittleFS) all using a common, compatible File class.

closed 2019-04-15 by earlephilhower