esp8266/Arduino #4210: WIFI.isconnected stil false after router reboot but ping requests reponse etc

d00lar, 2018-01-21:
#### Hardware
Hardware:	sonoff basic
Core Version:      	2.4 

hi
i found out then after reboot my router esp reconnects to network response to pings
even  handle webserwer etc

but i disable my home wifi during nights  and have some function to hanndle night

	if(WiFi.isConnected())
	{
		.......
	}

and at morning after my home wifi is enabled again
isconnected still return false..
checked also 
if (WiFi.status() == WL_CONNECTED)
it also still return false 

is there any other way to handle it propertly ?

on 2.3  (WiFi.status() == WL_CONNECTED) was working fine....

regards

bugadani, 2018-02-14:
I am also looking for a proper solution here. If I forcefully disconnect my device from my router, it reconnects but the status is continuously reported as WL_DISCONNECTED.

As a workaround you can call `WiFi.disconnect(true);` to disable wifi and then reconnect.

RAlexeev, 2018-06-27:
I have the same problem. I expected, that function `WiFi.isConnected()` should return `true` and `WiFi.status()` should return `WL_CONNECTED` code after automatic reconnection to Wi-Fi AP, however for some reason these functions return `false` and `WL_DISCONNECTED` respectively. But in fact, ESP has connected to Wi-Fi AP successfully.

I compile my sketch by Arduino IDE v. 1.8.5:

```
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

const char* ssid = "MYWIFISSID";
const char* password = "MYWIFIPSK";
unsigned long t = millis();

void setup()
{
  Serial.setDebugOutput(true);
  Serial.begin(115200);
  delay(1000);

  WiFi.mode(WIFI_AP_STA);
  WiFi.softAP("my_esp");
  delay(1000);
  
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
}

void loop()
{
  delay(5000);

  DynamicJsonDocument doc;
  JsonObject& obj = doc.to<JsonObject>();

  obj["chip_id"] = String(ESP.getChipId(), HEX);

  JsonObject& wifi = obj.createNestedObject("wifi");
  switch (WiFi.getMode()) {
    case WIFI_OFF:
      wifi["mode"] = "WIFI_OFF";
      break;
    case WIFI_STA:
      wifi["mode"] = "WIFI_STA";
      break;
    case WIFI_AP:
      wifi["mode"] = "SWIFI_AP";
      break;
    case WIFI_AP_STA:
      wifi["mode"] = "WIFI_AP_STA";
      break;
    default:
      wifi["mode"] = "UNKNOWN";
  }
  
  JsonObject& ap = wifi.createNestedObject("ap");
  ap["ssid"] = WiFi.softAPSSID();
  ap["psk"] = WiFi.softAPPSK();
  ap["ip"] = WiFi.softAPIP().toString();
  ap["mac"] = WiFi.softAPmacAddress();
  ap["station_num"] = (uint8_t) WiFi.softAPgetStationNum();
  
  JsonObject& station = wifi.createNestedObject("station");
  station["ssid"] = WiFi.SSID();
  station["psk"] = WiFi.psk();
  station["ip"] = WiFi.localIP().toString();
  station["hostname"] = WiFi.hostname();
  station["mac"] = WiFi.macAddress();
  station["subnet"] = WiFi.subnetMask().toString();
[... 695 lines cut ...]
    },
    "station": {
      "ssid": "MYWIFISSID",
      "psk": "MYWIFIPSK",
      "ip": "0.0.0.0",
      "hostname": "ESP_180B0C",
      "mac": "DC:4F:22:18:0B:0C",
      "subnet": "0.0.0.0",
      "gateway": "0.0.0.0",
      "dns": {
        "0": "192.168.1.1",
        "1": "0.0.0.0"
      }
    }
  },
  "state": {
    "AutoConnect": true,
    "AutoReconnect": true,
    "Persistent": true,
    "isConnected": false,
    "status": "WL_DISCONNECTED"
  }
}
========================================
```

d-a-v, 2018-06-27:
@d00lar @bugadani @RAlexeev  Please try latest master. Yesterday's fixes were about WiFi disconnection and reconnection.

RAlexeev, 2018-06-27:
Library from master branch seems really works properly in this place! Thank you!

RAlexeev, 2018-06-30:
@d-a-v I hurried with conclusions, when said that `WiFi.status()` works properly. Today I've faced this problem again. If I set AutoReconnect to false and when I turned off my Wi-Fi Access Point, `WiFi.isConnected()` return incorrect status `WL_CONNECTED`. If `AutoReconnect` is true (comment out this string `WiFi.setAutoReconnect(false);`), then `WiFi.isConnected()` will return correct status `WL_DISCONNECTED` after turning off Wi-Fi router. But `WiFi.status()` return different statuses with different Wi-Fi routers (see serial monitor outputs below).

I compile my sketch by Arduino IDE v. 1.8.5:

    #include <ESP8266WiFi.h>
    #include <ArduinoJson.h>

    const char* ssid = "MYWIFISSID";
    const char* password = "MYWIFIPSK";

    void printDebug() {
      DynamicJsonDocument doc;
      JsonObject& obj = doc.to<JsonObject>();

      obj["chip_id"] = String(ESP.getChipId(), HEX);

      JsonObject& wifi = obj.createNestedObject("wifi");
      switch (WiFi.getSleepMode()) {
        case WIFI_NONE_SLEEP:
          wifi["SleepMode"] = "WIFI_NONE_SLEEP";
          break;
        case WIFI_LIGHT_SLEEP:
          wifi["SleepMode"] = "WIFI_LIGHT_SLEEP";
          break;
        case WIFI_MODEM_SLEEP:
          wifi["SleepMode"] = "WIFI_MODEM_SLEEP";
          break;
        default:
          wifi["SleepMode"] = "UNKNOWN";
      }
      
      switch (WiFi.getMode()) {
        case WIFI_OFF:
          wifi["mode"] = "WIFI_OFF";
          break;
        case WIFI_STA:
          wifi["mode"] = "WIFI_STA";
          break;
        case WIFI_AP:
          wifi["mode"] = "WIFI_AP";
          break;
        case WIFI_AP_STA:
          wifi["mode"] = "WIFI_AP_STA";
          break;
        default:
          wifi["mode"] = "UNKNOWN";
      }
      
      JsonObject& ap = wifi.createNestedObject("ap");
      ap["ssid"] = WiFi.softAPSSID();
      ap["psk"] = WiFi.softAPPSK();
      ap["ip"] = WiFi.softAPIP().toString();
      ap["mac"] = WiFi.softAPmacAddress();
      ap["station_num"] = (uint8_t) WiFi.softAPgetStationNum();
      
      JsonObject& station = wifi.createNestedObject("station");
      station["ssid"] = WiFi.SSID();
      station["psk"] = WiFi.psk();
      station["ip"] = WiFi.localIP().toString();
      station["hostname"] = WiFi.hostname();
      station["mac"] = WiFi.macAddress();
      station["subnet"] = WiFi.subnetMask().toString();
      station["gateway"] = WiFi.gatewayIP().toString();
      
      JsonObject& dns = station.createNestedObject("dns");
      dns["0"] = WiFi.dnsIP(0).toString();
      dns["1"] = WiFi.dnsIP(1).toString();

      JsonObject& states = obj.createNestedObject("state");
      states["AutoConnect"] = WiFi.getAutoConnect();
[... 675 lines cut ...]
          "station_num": 0
        },
        "station": {
          "ssid": "MYWIFISSID_OTHER",
          "psk": "MYWIFIPSK_OTHER",
          "ip": "0.0.0.0",
          "hostname": "ESP_181458",
          "mac": "DC:4F:22:18:14:58",
          "subnet": "0.0.0.0",
          "gateway": "0.0.0.0",
          "dns": {
            "0": "192.168.0.1",
            "1": "0.0.0.0"
          }
        }
      },
      "state": {
        "AutoConnect": true,
        "AutoReconnect": true,
        "Persistent": true,
        "isConnected": false,
        "status": "WL_DISCONNECTED"
      }
    }
    ========================================

RAlexeev, 2018-06-30:
And second bug, which I encountered, function `WiFi.setAutoConnect(false);` doesn't set `AutoConnect` to `false`, if `persistent` is `false`: `WiFi.persistent(false);`. See example sketch below, compiled by Arduino IDE v. 1.8.5:

    #include <ESP8266WiFi.h>
    #include <ArduinoJson.h>

    const char* ssid = "MYWIFISSID";
    const char* password = "MYWIFIPSK";

    void printDebug() {
      DynamicJsonDocument doc;
      JsonObject& obj = doc.to<JsonObject>();

      obj["chip_id"] = String(ESP.getChipId(), HEX);

      JsonObject& wifi = obj.createNestedObject("wifi");
      switch (WiFi.getSleepMode()) {
        case WIFI_NONE_SLEEP:
          wifi["SleepMode"] = "WIFI_NONE_SLEEP";
          break;
        case WIFI_LIGHT_SLEEP:
          wifi["SleepMode"] = "WIFI_LIGHT_SLEEP";
          break;
        case WIFI_MODEM_SLEEP:
          wifi["SleepMode"] = "WIFI_MODEM_SLEEP";
          break;
        default:
          wifi["SleepMode"] = "UNKNOWN";
      }
      
      switch (WiFi.getMode()) {
        case WIFI_OFF:
          wifi["mode"] = "WIFI_OFF";
          break;
        case WIFI_STA:
          wifi["mode"] = "WIFI_STA";
          break;
        case WIFI_AP:
          wifi["mode"] = "WIFI_AP";
          break;
        case WIFI_AP_STA:
          wifi["mode"] = "WIFI_AP_STA";
          break;
        default:
          wifi["mode"] = "UNKNOWN";
      }
      
      JsonObject& ap = wifi.createNestedObject("ap");
      ap["ssid"] = WiFi.softAPSSID();
      ap["psk"] = WiFi.softAPPSK();
      ap["ip"] = WiFi.softAPIP().toString();
      ap["mac"] = WiFi.softAPmacAddress();
      ap["station_num"] = (uint8_t) WiFi.softAPgetStationNum();
      
      JsonObject& station = wifi.createNestedObject("station");
      station["ssid"] = WiFi.SSID();
      station["psk"] = WiFi.psk();
      station["ip"] = WiFi.localIP().toString();
      station["hostname"] = WiFi.hostname();
      station["mac"] = WiFi.macAddress();
      station["subnet"] = WiFi.subnetMask().toString();
      station["gateway"] = WiFi.gatewayIP().toString();
      
      JsonObject& dns = station.createNestedObject("dns");
      dns["0"] = WiFi.dnsIP(0).toString();
      dns["1"] = WiFi.dnsIP(1).toString();

      JsonObject& states = obj.createNestedObject("state");
      states["AutoConnect"] = WiFi.getAutoConnect();
      states["AutoReconnect"] = WiFi.getAutoReconnect();
      states["Persistent"] = WiFi.getPersistent();
      states["isConnected"] = WiFi.isConnected();

      switch (WiFi.status()) {
        case WL_CONNECTED:
          states["status"] = "WL_CONNECTED";
          break;
        case WL_NO_SHIELD:
          states["status"] = "WL_NO_SHIELD";
          break;
        case WL_IDLE_STATUS:
          states["status"] = "WL_IDLE_STATUS";
          break;
        case WL_NO_SSID_AVAIL:
          states["status"] = "WL_NO_SSID_AVAIL";
          break;
        case WL_SCAN_COMPLETED:
          states["status"] = "WL_SCAN_COMPLETED";
          break;
        case WL_CONNECT_FAILED:
          states["status"] = "WL_CONNECT_FAILED";
          break;
        case WL_CONNECTION_LOST:
          states["status"] = "WL_CONNECTION_LOST";
          break;
        case WL_DISCONNECTED:
          states["status"] = "WL_DISCONNECTED";
          break;
        default:
          states["status"] = "UNKNOWN";
      }

      WiFi.printDiag(Serial);
      serializeJsonPretty(obj, Serial);
    }

    void setup()
    {
      Serial.setDebugOutput(true);
      Serial.begin(115200);
      delay(1000);

      WiFi.setAutoConnect(false);
      WiFi.persistent(false);

      WiFi.mode(WIFI_STA);

      WiFi.begin(ssid, password);
      while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
      }
    }

    void loop()
    {
      delay(10000);

      Serial.println("========================================");
      printDebug();
      Serial.println();
      Serial.println("========================================");
    }

Serial monitor output:

    SDK:2.2.1(cfd48f3)/Core:2.4.1-104-g1eb0645d/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
    bcn 0
    del if1
    usl
    mode : sta(dc:4f:22:18:14:58)
    add if0
    wifi evt: 8
    .wifi evt: 2
    ....scandone
    state: 0 -> 2 (b0)
    .state: 2 -> 3 (0)
    state: 3 -> 5 (10)
    add 0
    aid 5
    cnt 

    connected with MYWIFISSID, channel 6
    dhcp client start...
    wifi evt: 0
    ...ip:192.168.43.228,mask:255.255.255.0,gw:192.168.43.1
    wifi evt: 3
    .pm open,type:2 0
    ========================================
    Mode: STA
    PHY mode: N
    Channel: 6
    AP id: 0
    Status: 5
    Auto connect: 1
    SSID (10): MYWIFISSID
    Passphrase (9): MYWIFIPSK
    BSSID set: 0
    {
      "chip_id": "181458",
      "wifi": {
        "SleepMode": "WIFI_MODEM_SLEEP",
        "mode": "WIFI_STA",
        "ap": {
          "ssid": "ESP_181458",
          "psk": "",
          "ip": "0.0.0.0",
          "mac": "DE:4F:22:18:14:58",
          "station_num": 0
        },
        "station": {
          "ssid": "MYWIFISSID",
          "psk": "MYWIFIPSK",
          "ip": "192.168.43.228",
          "hostname": "ESP_181458",
          "mac": "DC:4F:22:18:14:58",
          "subnet": "255.255.255.0",
          "gateway": "192.168.43.1",
          "dns": {
            "0": "192.168.43.1",
            "1": "0.0.0.0"
          }
        }
      },
      "state": {
        "AutoConnect": true,
        "AutoReconnect": true,
        "Persistent": false,
        "isConnected": true,
        "status": "WL_CONNECTED"
      }
    }
    ========================================

d-a-v, 2018-07-02:
@RAlexeev lwip2 is meant to follow lwip1.4 behavior regarding link layer connection status.
To help narrowing the source of this issue, can you please verify whether same things happen when using lwip1.4 ?

RAlexeev, 2018-07-02:
Using `lwip1.4` doesn't solve this problem. I shortened my sketch for clarity:

    #include <ESP8266WiFi.h>

    void setup()
    {
      Serial.setDebugOutput(true);
      Serial.begin(115200);
      delay(1000);

      WiFi.setAutoConnect(false);

      Serial.print("AutoConnect: ");
      Serial.println(WiFi.getAutoConnect() ? "Yes" : "No");
    }

    void loop() { }

Serial monitor output for lwip2:

    SDK:2.2.1(cfd48f3)/Core:2.4.1-104-g1eb0645d/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
    AutoConnect: Yes
    wifi evt: 7

Serial monitor output for lwip1.4:

    SDK:2.2.1(cfd48f3)/Core:2.4.1-104-g1eb0645d/lwIP:1.4.0rc2/BearSSL:94e9704
    AutoConnect: Yes
    wifi evt: 7

d-a-v, 2018-07-02:
Thanks !
I can't reproduce.
Have you tried the `Tools>Erase Flash>All Flash Contents` option ?

RAlexeev, 2018-07-02:
I remember that some time ago I didnt have these problems. And now it seems to me that it happened after erasing flash: Sketch + WiFi Settings. If I select this errasing option, then function `setAutoConnect` doesn't set `AutoConnect` to `false`. But it set `AutoConnect` to `false`, if I add this string `WiFi.mode(mode);` to my sketch before `setAutoConnect`.

    #include <ESP8266WiFi.h>

    void setup()
    {
      Serial.setDebugOutput(true);
      Serial.begin(115200);
      delay(1000);

      WiFi.mode(WIFI_AP_STA);
      WiFi.setAutoConnect(false);   // successfully sets AutoConnect to false, if there's previous string

      Serial.print("AutoConnect: ");
      Serial.println(WiFi.getAutoConnect() ? "Yes" : "No");
    }

    void loop() { }

If I upload this sketch, erasing only sketch, then `setAutoConnect` works fine without additional string `WiFi.mode(mode);`.

d-a-v, 2018-07-02:
Thanks for the report,
Should it be mentioned in the [documentation](https://github.com/esp8266/Arduino/blob/1eb0645dcb82e03e239e7932ad20802c769c89e0/doc/esp8266wifi/station-class.rst#setautoconnect)  to use `WiFi.setAutoConnect()` (or is it only `WiFi.getAutoConnect()`?) after `WiFi.mode(WIFI_AP_STA)` or `WiFi.mode(WIFI_STA)` ?.

RAlexeev, 2018-07-02:
Thank you!

`WiFi.setAutoConnect(false);` works for me only after `WiFi.mode(WIFI_STA);` or `WiFi.mode(WIFI_AP_STA);`. But `WiFi.mode(WIFI_OFF);` and `WiFi.mode(WIFI_AP);` doesn't affect for `WiFi.setAutoConnect(false);` and AutoConnect is `true` after that. Even more, if I set `WiFi.setAutoConnect(false);` after `WiFi.mode(WIFI_AP_STA);` or `WiFi.mode(WIFI_AP);` and then restart ESP8266, Access Point will be created anyway. And that seems strange, AP's SSID will be empty after `WiFi.softAPdisconnect();`, not even default value after module restart. Is there way to turn off softAP auto creation? Especially empty SSID is annoying.

My sketch:

#include <ESP8266WiFi.h>

    void setup() {
      Serial.setDebugOutput(true);
      Serial.begin(115200);
      delay(1000);

      Serial.print("AutoConnect: ");
      Serial.println(WiFi.getAutoConnect() ? "Yes" : "No");

      WiFi.mode(WIFI_AP);
      WiFi.setAutoConnect(false);

      Serial.print("AutoConnect: ");
      Serial.println(WiFi.getAutoConnect() ? "Yes" : "No");

      Serial.print("SSID: ");
      Serial.println(WiFi.softAPSSID());
      Serial.print("PSK: ");
      Serial.println(WiFi.softAPPSK());
      Serial.print("IP: ");
      Serial.println(WiFi.softAPIP());
      
      delay(1000);
      WiFi.softAPdisconnect();  // it could be WiFi.softAPdisconnect(true);, it doesn't matter
      delay(1000);
      
      Serial.print("SSID: ");
      Serial.println(WiFi.softAPSSID());
      Serial.print("PSK: ");
      Serial.println(WiFi.softAPPSK());
      Serial.print("IP: ");
      Serial.println();
    }

    void loop() { }

First execution after firmware uploading:

    SDK:2.2.1(cfd48f3)/Core:2.4.1-104-g1eb0645d/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
    AutoConnect: Yes
    AutoConnect: Yes
    SSID: ESP_180B0C
    PSK: 
    IP: 192.168.4.1
    bcn 0
    del if1
    usl
    add if1
    dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
    bcn 100
    bcn 0
    del if1
    usl
    mode : null
    wifi evt: 8
    SSID: 
    PSK: 
    IP: 

Second execution (only restart without firmware uploading):

    SDK:2.2.1(cfd48f3)/Core:2.4.1-104-g1eb0645d/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
    AutoConnect: Yes
    mode : softAP(de:4f:22:18:0b:0c)
    add if1
    dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
    bcn 100
    AutoConnect: Yes
    SSID: 
    PSK: 
    IP: 192.168.4.1
    wifi evt: 8
    bcn 0
    del if1
    usl
    add if1
    dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
    bcn 100
    bcn 0
    del if1
    usl
    mode : null
    wifi evt: 8
    SSID: 
    PSK: 
    IP:

RAlexeev, 2018-07-03:
As I understand, `WiFi.softAPdisconnect(true)` switches the soft-AP mode off, but not affects SSID/password stored in flash, because `persistent` is `false`. But after restart ESP I see network with default SSID (in my case it's ESP_181458) instead of my custom name "My_ESP".

```c
#include <ESP8266WiFi.h>

void setup() {
  Serial.setDebugOutput(true);
  Serial.begin(115200);
  delay(30000);

  Serial.print("SSID: ");
  Serial.println(WiFi.softAPSSID());
  Serial.print("PSK: ");
  Serial.println(WiFi.softAPPSK());
  Serial.print("IP: ");
  Serial.println(WiFi.softAPIP());
  
  WiFi.persistent(false);
  WiFi.mode(WIFI_AP);
  WiFi.softAP("My_ESP");

  Serial.print("SSID: ");
  Serial.println(WiFi.softAPSSID());
  Serial.print("PSK: ");
  Serial.println(WiFi.softAPPSK());
  Serial.print("IP: ");
  Serial.println(WiFi.softAPIP());
  
  delay(30000);
  WiFi.softAPdisconnect();
  delay(1000);
  
  Serial.print("SSID: ");
  Serial.println(WiFi.softAPSSID());
  Serial.print("PSK: ");
  Serial.println(WiFi.softAPPSK());
  Serial.print("IP: ");
  Serial.println();
}

void loop() { }
```

Serial monitor output:

```
SDK:2.2.1(cfd48f3)/Core:2.4.1-104-g1eb0645d/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
SSID: ESP_181458
PSK: 
IP: 192.168.4.1
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
SSID: My_ESP
PSK: 
IP: 192.168.4.1
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
wifi evt: 7
wifi evt: 7
SSID: 
PSK: 
IP: 
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
```

Restart ESP

```
SDK:2.2.1(cfd48f3)/Core:2.4.1-104-g1eb0645d/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
SSID: ESP_181458
PSK: 
IP: 192.168.4.1
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
SSID: My_ESP
PSK: 
IP: 192.168.4.1
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
SSID: 
PSK: 
IP: 
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
```

Sometimes I see these three networks at the same time:

```
NAME    CHAN  SSID            BSSID              SECURITY   SIGNAL
AP[1]   1     My_ESP          DE:4F:22:18:14:58  --         100
AP[2]   1     ESP_181458      DE:4F:22:18:14:58  --         100
AP[3]   1     --              DE:4F:22:18:14:58  --         100
```

May be you could give me a clue how to disable default SoftAP at start.

d-a-v, 2021-03-31:
Closing as duplicate #7432, let's follow-up there !

closed 2021-03-31 by d-a-v