esp8266/Arduino #4211: WifiClient read() as stream?

wThomas84, 2018-01-21:
### Basic Infos
Hello, I have a questing regarding the WifiModule with which I want to implement a simple Restful server.

#### Hardware
Hardware:			NodeMcu

### Description

Currently I am able to connect to my Wifi and also send over simple Http POST requests. I can already read the request in a simple manner into a string by pre allocating memory for it.

In order to better handle the request body (json). I want to use the [aJson](https://github.com/interactive-matter/aJson) library which also expects the content as stream. I want to use a stream in order to save memory and not being forced to reserve much memory in advance.
In order to extract the body from the request I use that WifiClient's read function:

```cpp
#include <ESP8266WiFi.h>
#include <aJSON.h>

//... before this function is called the header has already been read. 
char* readRequestBody(WiFiClient client)
{
  int i = 0;
  char * res = (char *) malloc(sizeof(char) * 255);
  while(client.available())
  {
    res[i++] = client.read();
  }
  return res;
}
```

So my question is; Can I somehow obtain a stream that I can hand over to the Json parsing library and thus omit the malloc() call like so:
```
aJsonObject* jsonObject = aJson.parse(client.stream());

```

Thank you and best regards
Thomas

igrr, 2018-01-21:
WiFiClient is derived from Stream, so you should be able to call `parse(&client)` (if `parse` accepts `Stream*`).
Bear in mind that the full contents of JSON might not be available to WiFiClient when you call `parse`. Your aJSON parser needs to be able to handle this somehow. That, however, is a question best asked in the issue tracker of aJSON library.

closed 2018-01-24 by devyte

wThomas84, 2018-01-24:
Thanks for the fast answer. I am going to test it out. Sounds great and haven't seen the inheritance. Best /Thomas