esp8266/Arduino #4212: Network port not showing after running BasicOTA

brianjmurrell, 2018-01-21:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	not sure

### Description

Even after flashing the (modified to connect to my AP instead of running as WIFI_STA) BasicOTA example, I am not seeing my NodeMCU listed as a network port.  I only see a list of _Serial ports_.

The machine running the IDE does see the mDNS advertisement:

```
Service data for service 'esp8266-439455' of type '_arduino._tcp' in domain 'local' on 7.0:
	Host esp8266-439455.local (10.75.22.187), port 8266, TXT data: ['auth_upload=no', 'board=ESP8266_NODEMCU', 'ssh_upload=no', 'tcp_check=no']
```
is from `avahi-discover`.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB/3MB SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			not sure
Flash Frequency:	not sure
Upload Using:		SERIAL
Reset Method:		not sure

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "my_ssid";
const char* password = "...";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
/*  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }
*/
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  
  Serial.println("");
  Serial.println("WiFi connected");
  
  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}
```

### Debug Messages

```
Booting
....
WiFi connected
Ready
IP address: 10.75.22.187
```

MarkyAD, 2018-01-21:
flash via serial
reset esp with hardware reset switch or power cycle
restart ide

MarkyAD, 2018-01-22:
If you're on 2.4.0 core and it still doesn't show up, try switching to lwip 1.4 (in tools menu) and reflash with the steps above. I'm still on 2.4.0-rc2 with working ota on several esp, try that one if nothing else works out.

Ranjith1819, 2019-02-07:
am also facing same problem kindly help me....

expo17, 2019-04-02:
I'm also facing same problem in my project

Ranjith1819, 2019-04-05:
i successfully accomplished the task of ota update

earlephilhower, 2021-02-14:
Closing due to age and MDNS (which publishes the OTA port) being completely rewritten.  If there is a problem with git master, please open a new issue.

closed 2021-02-14 by earlephilhower