esp8266/Arduino #4213: provide a portable way to clear tcp pcb in time-wait

d-a-v, 2018-01-22:
A common API for both lwip-v1 / lwip-v2 is needed to clear PCBs in TIME-WAIT since this workaround is widely used (```tcp_abort(tcp_tw_pcbs)```)

mribble, 2018-01-22:
I've been reading through lots of esp8266 bugs related to this and I haven't found anything that explains exactly what is happening so I went and did some research on this topic.  I'm sure a lot of people who work on esp8266 understand this, but I want to write down a longer description of the problem for people who aren't experts on tcp protocols (like myself).  Experts, if I get anything wrong here please correct me.

This 184 bytes of memory isn't a leak.  It is a tracked tcp connection, and is held around because that is what the TCP spec says you should do.  The reason it is held around is because creating a tcp connection is expensive (requires a 3 way handshake) so it was assumed clients would want to reuse a connection as a performance optimization.  The TCP spec calls this the WAIT_STATE and it's the last state of of a TCP connection where it's just sitting there seeing if the connection will be reused. Here is an article that explains WAIT_STATE in more detail: http://www.serverframework.com/asynchronousevents/2011/01/time-wait-and-its-design-implications-for-protocols-and-scalable-servers.html

So you might be wondering what does tcp have to do with anything?  I'm just seeing this problem when loading webpages on my esp8266.  Here are two common places where I think you might run out of memory:

1. When a lot of different devices connect to the esp8266
2. When sending a lot of new xmlHttpRequest calls from a single device (like you would with an auto refreshing webpage or when using ajax)

Loading webpages and ajax use http and http is layered on top of tcp.  The http protocol have this reuse concept too and they call it "HTTP persistent connection".  https://en.wikipedia.org/wiki/HTTP_persistent_connection

I think it's possible to fix the number 2 issue above.  My problem is that I'm using ajax and constantly creating new xmlHttpRequest objects for each data transfer.  I am planning to try to modify how I use xmlHttpRequests so instead of always creating a new one I will now either reuse one or maybe make a pool of them that I can reuse.  This should both improve performance, reduce power usage, and get rid of the out of memory problem for me.  If you are doing a full html page refresh you won't have an option to reuse the xmlHttpRequest, but if you spend some time on the webpage you could change it to use ajax to do a full refresh of the body of your page and that would allow you to fix the multiple xmlHttpRequest issue.

I don't think number 1 above can be fixed on the client since each connection is coming from a different device and thus they can't reuse xmlHttpRequest connections.  While it might not be common for an esp8266 to host a website that is accessed by lots of devices, I do think it should have a better behavior than consuming all the memory until some other function that requires a few KB of heap space crashes.

I think a final fix for this issue should involve 2 new public APIs:
1) A way to specify how many seconds a tcp connection stays in the WAIT_STATE before closing it
2) A way to specify how many tcp connections can remain active before the least recently used is aborted

If there is objection to having both these new APIs then I think the second one is more important because it lets you control the amount of memory consumed by this tcp wait_state algorithm.  In microcontroller environments like the esp8266 memory usage is often more important than performance.  Some people might want to set it to 0 to keep memory consumption to a minimum.  Most people would just use whatever the default is.  Once I make the modifications above, I might choose some low number that matches well with however many my client's xmlHttpRequest pool uses (perhaps 8).

d-a-v, 2018-01-22:
I should not have written API. I intended to propose an example sketch on how to release those pcb in time-wait state for the two lwip versions. lwip-v1 did not handle them properly and workaround such as the one in your [#1923 comment](https://github.com/esp8266/Arduino/issues/1923#issuecomment-359294328) is widely used. Since lwip2, the include files have changed and there are several questions about how to do this.

The fact are that it should not be necessary (but lwip-v1 timers are buggy) and [this test](https://github.com/esp8266/Arduino/issues/2925#issuecomment-303100724) shows it with lwip2.

In the example I would provide this explanation, maybe a link to this test, and leave the possibility for the user to keep using it with lwip2.

So since you are facing this problem, could you check whether this workaround is still necessary ?

mribble, 2018-01-22:
Two other issues I've noticed with my app with lwip 2:
1) The values in WiFi.getMode() or WiFi.status() changed.  I'm not complaining really because the old values were not what I expected.  Just mentioning this could break some apps (it broke my connection indicator LED, but it's really easy for me to fix in the app).  No fix needed, just something that you should document as something people need to change when porting to lwip 2.
2) OTA updates (specifically the advertising) don't seem to work with lwip 2.  If this isn't a common issue I can try looking into it more.

d-a-v, 2018-01-23:
Maximum time-wait is a defined set to 2mn by default in lwipopts (2*TCP_MSL). Grain is 500ms.
```
#define TCP_MSL 60000UL /* The maximum segment lifetime in milliseconds */
```
Reducing it is not advised for heavily loaded web servers, or when dealing with NAT peers. Time-wait helps a tcp server not confusing two consecutive connections with the same (s-ip,s-port,d-ip,d-port) when the first is closed but duplicate packets lost in internet arrive later during the second. 

The workaround is an artificial way to lower it. I would lean to lower it, but it will never be low enough for some, and it would break TCP as it is defined. So the best move here is to show it in an example.

I was not suggesting you try another sketch than yours.

Please open new issues regarding 1. WiFi.getMode() with an example so we can quickly test / verify. 
Regarding 2. I believe there has been OTA fixes since 2.4.0, I lost track of them and I let you retry with latest git version of the core. Then browse the opened issues or open a new one about OTA.

mribble, 2018-01-23:
If you can show an example to artificially lower that timeout mechanism on the esp8266 that should work for my old use case.  I think such an example would be useful and it's still not clear to me how to do that with lwip v2.  That said in my app I have started reusing http connections in my javascript so I don't actually have the memory issue anymore even with the default configuration.  I think explaining what could be changed to fix this like I did in the first response of this thread would help a lot of people (though some still would need the example from you on how to work around it in the esp8266 code).

For #1, I made a small test app and it didn't reproduce the issue.  I narrowed down what what happening in my bigger app.  For some reason WiFi.status() is returning WL_DISCONNECTED even when it is connected and working.  I'm not sure why this is the case, but I am doing some strange things in my full app.  For now I'll assume it's a bug in my app and will look into it in more detail.  If I find a real issue in your code I'll file a seperate bug as you suggested.

There are a few fixes in master for OTA.  I have verified recent builds work so that seems to be fixed since the 2.4.0 version as you thought it was.  Once there is an ESP8266 2.5.0 release with the OTA fixes I will switch my project to using lwip v2, but until then I'll stick with lwip v1.4 to make it easier for others to contribute without a lot of extra install instructions.

mribble, 2018-01-24:
Thanks you!  That made it look too easy.  I'm sure it will be helpful to a lot of people. 👍

d-a-v, 2018-01-25:
@mribble does lwip2+tcpCleanup fix your problem ?

mribble, 2018-01-25:
As  I mentioned before I have changed my javascript code to work around the issue so I no longer need this code (and in fact don't want to call it because it would hurt performance).

That said I reverted to an old version of my javascript and confirmed with both lwip 1.4 and 2 this code fixes the problem with too many new xlmHttpRequests.

I can't switch to lwip v2 due to the issues with OTA updates in the esp8266 2.4.0 release, but that seems fixed in master and once there is a 2.5.0 release with those fixes I'll switch to using lwip v2.

Thanks for the great support!

PR #4314 (2018-02-07): doc/faq: tcpCleanup
closes #4213
closed 2018-02-07 by devyte