esp8266/Arduino #4214: Can't connect my Light Sensor in my ESP8266!!!

kidbreaker22, 2018-01-22:
So basically I am trying to test my ESP8266 V3 NodeMCU by using a light sensor. It's a simple test where it reads the analog pin 0 and dim the led depending on the brightness level. But then when I try it I am getting random characters. Can someone please look at this and tell me if I did anything wrong? This test works on my Genuino Uno but not on my ESP8266.

```
int sensorValue = 0;
int value = 0;

const int sensorPin = A0;
const int LEDPin = 8;

void setup() {
  // put your setup code here, to run once:

  Serial.begin(9600);

  pinMode(LEDPin, OUTPUT);

}

void loop() {
  // put your main code here, to run repeatedly:

  sensorValue = analogRead(sensorPin);
  delay(500);

  Serial.print("Sensor Value: ");
  Serial.println(sensorValue);

  value = 255 - sensorValue/4;

  Serial.print("Analog Value of Pin D8: ");
  Serial.println(value);

  analogWrite(LEDPin, value);

}
```

[image]

devyte, 2018-01-22:
Have you tried closing and reopening the serial monitor?

kidbreaker22, 2018-01-22:
@devyte closing and reopening the serial monitor gives me either nothing or random characters again like the picture above.

MarkyAD, 2018-01-22:
Try other baud rates? 38400 and 115200 are not unreasonable. cp210x uart is good to near 1Mb/sec and ch34x almost double that.

kidbreaker22, 2018-01-22:
@MarkyAD I just tried all of them but nothing works still. Only at 74880 baud gave me English letters when I press a reset button in ESP8266. 

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

```

MarkyAD, 2018-01-22:
Well that certainly looks promising. You set Serial.begin(38400); in the sketch and re-flashed it?

kidbreaker22, 2018-01-22:
@MarkyAD lol I did not do that earlier please excuse my retardness but I am still getting random characters   for 38400 and 115200

MarkyAD, 2018-01-22:
Try choose board "Core Development Module" it lets you pick "Flash Mode", set to DIO. Set flash size to "4M/1M" or "4M/3M" and reflash sketch.

MarkyAD, 2018-01-22:
I also read some with the same "baud locked to 74880" had to run an esptool -ce (erase_flash) once before they worked. I'm not sure why that would be, Ive got several of the same boards, none of them ever did it.

5chufti, 2018-01-22:
try setting 74880 in your sketch and in the serial monitor.

kidbreaker22, 2018-01-22:
@MarkyAD I don't have a board "Core Development Module" in my Arduino IDE.
[image]

kidbreaker22, 2018-01-22:
@5chufti I just tried 74880 and I am still getting random characters.

lrmoreno007, 2018-01-22:
Everithing was Ok, except this line "const int LEDPin = 8;", you are trying to output in GPIO8 and it is not available in the NodeMCU. I think the one you want to use is D8 (GPIO15), because to use analogWrite you must use a pin with PWM output in NodeMCU (D2, D5, D6 or D8).

Feel the difference between 8 (GPIO8) and D8 (GPIO15).

Select "NodeMCU 1.0..."  board.
Set speed to 9600 in Serial viewer.

```
int sensorValue = 0;
int value = 0;

const int sensorPin = A0;
const int LEDPin = D8;

void setup() {
  // put your setup code here, to run once:

  Serial.begin(9600);

  pinMode(LEDPin, OUTPUT);

}

void loop() {
  // put your main code here, to run repeatedly:

  sensorValue = analogRead(sensorPin);
  delay(500);

  Serial.print("Sensor Value: ");
  Serial.println(sensorValue);

  value = 255 - sensorValue/4;

  Serial.print("Analog Value of Pin D8: ");
  Serial.println(value);

  analogWrite(LEDPin, value);

}
```

[image]

kidbreaker22, 2018-01-22:
@lrmoreno007 Ohhh I see what I did wrong now it works so I either set LEDPin to D8 or 15 then. Thank you for telling me this sorry for me being ignorant about this.

closed 2018-01-22 by kidbreaker22

lrmoreno007, 2018-01-22:
I also recommend using D5 instead of D8, because GPIO15 is a bit special. See note in the image that I attached.

kidbreaker22, 2018-01-22:
Okay I see what you mean thank you your pinout is more advance the one I have.

Samiulpial, 2018-12-07:
I am using this code and getting random value for analog read .I have tested my sensor and its working fine .I have selected NodeMCU board from tool.In client.sendAll(false) ,I have changed it to true or blank but still getting random value .I am using arduino uno.

#include "UbidotsMicroESP8266.h"

#define TOKEN  "*******"  // Put here your Ubidots TOKEN
#define ID_1 "*****" // Put your variable ID here
//#define ID_2 "Your_variable_ID_here" // Put your variable ID here
#define WIFISSID "******" // Put here your Wi-Fi SSID
#define PASSWORD "khaja102" // Put here your Wi-Fi password

Ubidots client(TOKEN);

void setup(){
    Serial.begin(115200);
    client.wifiConnection(WIFISSID, PASSWORD);
    //client.setDebug(true); // Uncomment this line to set DEBUG on
}

void loop(){
    float value1 = analogRead(0);
    //float value2 = analogRead(2)
    client.add(ID_1, value1);
    //client.add(ID_2, value2);
    client.sendAll(false);
    delay(5000);
}

5chufti, 2018-12-14:
this is by no way a problem of esp8266 arduino core but more likely sheer ignorance ...
read [here](https://arduino-esp8266.readthedocs.io/en/latest/reference.html#analog-input) and visit user support forum like https://www.esp8266.com for further support