esp8266/Arduino #4215: ESP_MESH can't print anything by espconn_mesh_print_ver() on ESP8266 Arduino Core 2.4.0.

TridentTD, 2018-01-22:
### Basic Infos

Esp_mesh on ESP8266 Arduino core 2.4.0 release
can't show anything by espconn_mesh_print_ver().

How to show this info ?

#### Hardware
Hardware:			Wemos D1 Mini
Core Version:      	 2.4.0

### Sketch

```cpp
#include <Arduino.h>
#include <mesh.h>

void setup() {
  Serial.begin(9600); Serial.println();
  Serial.println("---------------------");
  espconn_mesh_print_ver();
  Serial.println("---------------------");
}

void loop() {
}
```

tonton81, 2018-01-22:
i think it was already mentioned that new sdk broke mesh, might need to wait for that patch

devyte, 2018-01-22:
@TridentTD The espconn_mesh_print_ver() function is an SDK api. It has nothing to do with the code in this repo.
In addition, Espressif has already pulled libmesh in [their repo](https://github.com/espressif/ESP8266_NONOS_SDK/commit/1f492fbe19a2c2d1761f6aad6bf92af21b6c086b), and it won't be available anymore starting in their upcoming 2.2 release.
Closing.

closed 2018-01-22 by devyte