esp8266/Arduino #4217: ESP-12E Connection Refused

ClockWorkDV, 2018-01-22:
#### Hardware
Hardware:	        ESP-12e
Core Version:      	2.1.0-rc2

### Description

I can not access esp-12e either by ip or by localhost, it's my first doing this kind of thing

### Settings in IDE

Module:  			ESP8266 NodeMCU
Flash Size: 		4M
CPU Frequency:      80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "Batata com Circuitos";
const char* pass = "0124785369aA";

unsigned long MiliAnterior = 0;

int led = LOW;
int led_pin = 16;
int val = 0;

WiFiServer server(80);

void setup() {

Serial.begin(115200);
delay(10);
pinMode(led_pin, OUTPUT);
digitalWrite(led_pin, 0);

Serial.println("");
Serial.println("");
Serial.println("Conectando a ");
Serial.print(ssid);
Serial.println("");

WiFi.begin(ssid, pass);

while(WiFi.status() != WL_CONNECTED){
delay(500);
Serial.print(".");
}

Serial.println("");
Serial.println("Conectado.");
Serial.println("");
Serial.println("Endereço ip:");
Serial.println(WiFi.localIP());

}

void loop() {

if(val){
if(millis() - MiliAnterior >= 1000){
MiliAnterior = millis();
if(led == LOW){

led = HIGH;

}else{

led = LOW;

}
}
}

WiFiClient client = server.available();
if(!client){

return;
}

Serial.println("Novo Cliente");

while(!client.available()){

delay(1);

}

String req = client.readStringUntil('\r');
Serial.println(req);
client.flush();

if(req.indexOf("DESLIGA") != -1 ){val = 0;}
if(req.indexOf("PISCA") != -1 ){val = 1;}

String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n";
s += "<DOCTYPE HTML>\r\n<html><h1>Exemplo_Web Pisca-Desliga</h1>";
s += "<p><a href=\"?com=PISCA";s += "><button><b>PISCA LED</b></button></a></p>";
s += "<p><a href=\"?com=PDESLIGA";s += "><button><b>DESLIGA</b></button></a></p>";
s += "</html>\n";

client.print(s);
delay(1);
Serial.print("Cliente Desconectado");

}

lrmoreno007, 2018-01-24:
You have declared the server but you have not initialized it "server.begin ();" at the end of setup ().

devyte, 2018-01-24:
Closing as user error.

closed 2018-01-24 by devyte