esp8266/Arduino #4224: printf with FlashHelperString

Ariel-International, 2018-01-23:
Hello,
I upgraded to 2.4.0 and I had again to restore my hack to be able to printf with FlashStringHelper string.
I saw the new printf_P feature, but this doesn't solve my issue, because I want to use both F() Strings and normal in debug, without thinking about it.
The new implementation breaks my debug system.

Anyhow: this is the complete hack, which is very simple:

in Print.h

```
/**************************************
 *
 * printf with PROGMEM flashHelperString
 * by ariel@ariel.international
 **************************************/ 
		#include <stdarg.h>
		
		#ifdef F // check to see if F() macro is available
		size_t printf(const __FlashStringHelper *format, ...);
		#endif
```

I put this near the end, just before

```
};
#endif
```

In Print.cpp:

```
size_t Print::printf(const __FlashStringHelper *format, ...)
	{
    va_list arg;
    va_start(arg, format);
    char temp[128];
    char* buffer = temp;
    size_t len = vsnprintf_P(temp, sizeof(temp), (const char *)format, arg);  
    va_end(arg);
    if (len > sizeof(temp) - 1) 
		{
        buffer = new char[len + 1];
        if (!buffer) return 0;
		
        va_start(arg, format);
        vsnprintf_P(buffer, len + 1, (const char *)format, arg);
        va_end(arg);
		}
    len = write((const uint8_t*) buffer, len);
    if (buffer != temp) delete[] buffer;
    return len;
	}
	
```
I usually add this after the original printf(const char *format, ...)  function and it adds the requested feature.

I know this is probably more of a pull request than an issue, but I don't know how to make one, so I figured someone will do that if they find this interesting and useful.

bulentperktas, 2018-02-07:
can you bring complete print.h and print.cpp file with a small example here please?

Here how i can use printf but i can not use it with FlashHelperString.

For example, esp8266 EEPROM dump to serial monitor.

`#include <EEPROM.h>

#define EEPROM_LENGTH 512

extern "C" void system_set_os_print(int);

void setup() {
  Serial.begin(74880);
  
  EEPROM.begin(512);
  
  printEEPROMData();
}

void loop() {}

void printEEPROMData() {

  system_set_os_print(true);  //<----

  Serial.println("Reading EEPROM Dump");

  int rows = EEPROM_LENGTH / 16;
  for (int row = 0; row < rows; row++) {
    Serial.print(" (");
    if (row == 0)
      Serial.print(0, HEX);
    Serial.print(row * 16, HEX);
    Serial.print(") \t ");

    for (int half_col = 0; half_col < 2; half_col++) {
      for (int col = 0; col < 8; col++) {
        int index = (row * 16) + (half_col * 8) + col;
        char b = EEPROM.read(index);
        Serial.printf("%02X ", b); //<----Here printf function which i wish using with FlashHelperString 
                                   //like this Serial.printf(F("%02X "), b);
      }

      if (half_col == 0) {
        Serial.print("- ");
      }
      else {
        Serial.println();
      }
    }
  }
}`

Here is output on Serial Monitor

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
Reading EEPROM Dump
 (00) 	 43 46 47 00 01 14 00 00 - 00 1E 00 00 00 C0 A8 01 
 (10) 	 65 FF FF FF 00 C0 A8 01 - 01 01 01 01 08 08 08 08 
 (20) 	 08 08 04 04 43 46 47 C0 - A8 04 01 FF FF FF 00 C0 
 (30) 	 A8 04 01 00 01 01 FF FF - FF FF FF FF FF FF FF FF 
 (40) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (50) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (60) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (70) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (80) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (90) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (A0) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (B0) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (C0) 	 42 55 4C 45 4E 54 50 45 - 52 4B 54 41 53 28 33 47 
 (D0) 	 57 49 52 45 4C 45 53 53 - 29 00 FF FF FF FF FF FF 
 (E0) 	 51 77 65 31 61 73 79 33 - 62 63 35 43 34 00 FF FF 
 (F0) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (100) 	 30 2E 64 65 2E 70 6F 6F - 6C 2E 6E 74 70 2E 6F 72 
 (110) 	 67 00 FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (120) 	 62 75 6C 65 6E 74 00 63 - 61 72 74 00 FF FF FF FF 
 (130) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (140) 	 30 36 30 31 4C 4F 47 30 - 30 30 31 00 FF FF FF FF 
 (150) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (160) 	 31 32 33 34 35 36 37 38 - 00 FF FF FF FF FF FF FF 
 (170) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (180) 	 64 65 70 6F 00 65 00 30 - 30 30 32 00 FF FF FF FF 
 (190) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (1A0) 	 62 69 6F 6D 65 64 69 68 - 65 61 74 2E 63 6F 6D 00 
 (1B0) 	 FF FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (1C0) 	 50 4F 53 54 20 2F 6B 61 - 79 69 74 2E 70 68 70 3F 
 (1D0) 	 00 FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
 (1E0) 	 50 4F 53 54 20 2F 6B 61 - 79 69 74 2E 70 68 70 3F 
 (1F0) 	 00 FF FF FF FF FF FF FF - FF FF FF FF FF FF FF FF 
`

Thank you,

romansavrulin, 2018-03-09:
seems like we should wait for #4223 and dependants to resolve.