esp8266/Arduino #4225: ESP8266 stay awake

ClockWorkDV, 2018-01-23:
is there any way to keep the ESP-12e always on without sleeping or sleeping for a short period of time?

NdK73, 2018-01-23:
That's the default, unless you call deepsleep() ...
Can you better describe the problem you're having?

ClockWorkDV, 2018-01-23:
I opened a localhost on the ESP-12E nodeMCU to turn on and off lights from my home but once in a while or it stops working or does not wake up from sleep mode, I did the GPIO0-RST pin connection but it did not change anything.

NdK73, 2018-01-23:
Maybe you setup a web server. Are you working from Arduino IDE? If so, it's better to make your sketch available (after removing all passwords). If not, this is the wrong place (and it's probably wrong anyway: this is not a support forum but a bug tracker... but if your ESP enters sleep w/o being requested, it's a bug).
BTW it's GPIO16 (sometimes known as D0) that needs to be connected to RST. Else you won't be able to flash a new image from serial. If you connected GPIO0, some sequences on the serial port will trigger a reset.

ClockWorkDV, 2018-01-23:
This is the code I mounted.

`#include <ESP8266WiFi.h>

const char* ssid = "Batata com Circuitos";
const char* pass = "0124785369aA";

WiFiServer serve(80);

void setup() {

delay(10);
pinMode(5, OUTPUT);
digitalWrite(5, HIGH);

WiFi.begin(ssid, pass);

while(WiFi.status() != WL_CONNECTED){
delay(500);
}

serve.begin();

}

void loop() {

WiFiClient client = serve.available();

if(!client){return;}

while(!client.available()){

delay(1);}

String req = client.readStringUntil('\r');
client.flush();

if(req.indexOf("DESLIGA") != -1 ){digitalWrite(5, 1);}
if(req.indexOf("LIGAR") != -1 ){digitalWrite(5, 0);}
client.flush();

String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n";
s += "<DOCTYPE HTML>\r\n<html><title>Rele</title><body>\n<table align=CENTER><tr align=CENTER><td><h1>Exemplo-Web Rele</h1>\n";
s += "<p>Rele: <a href=\?com=LIGAR><button><b>LIGAR</b></button></a> | \n";
s += "<a href=\?com=DESLIGA><button><b>DESLIGA</b></button></a></p></tr></td></table>\n";
s += "</body></html>\n";

client.print(s);
delay(1);
}`

NdK73, 2018-01-23:
You're probably filling the RAM. That's highly inefficient code that probably leaks memory.
Try adding a ```Serial.println(ESP.getFreeHeap());``` or inserting the same info inside the served page.
And remember that if it's exposed to Internet it's highly unsecure and every script kiddie can DoS it. That's why usually a beefy gateway system (supporting at least HTTPS) is needed for IoT systems.

PS: use triple backtick to start/end a codeblock.

devyte, 2018-01-23:
@ClockWorkDV Not the right place to ask such questions. This is an issue tracker, meant for tracking issues in the core.
Also, when you opened the issue you were presented with an issue template, which you completely ignored.
In addition, github supports markup, including markup for code. I suggest making use of it. The code you posted is hard to read as it is.
Finally, your question is about sleeping, but the code you posted doesn't seem to make use of any sleep functionality.
If you're going to serve html, I suggest making use of the ESP8266webserver or the asyncwebserver instead of trying to do it yourself.
Closing due to lack of info.

closed 2018-01-23 by devyte