esp8266/Arduino #4226: Constant hard crash NodeMCU V1

JakeBeck, 2018-01-23:
[NodeMCU_HardCrash.txt](https://github.com/esp8266/Arduino/files/1657587/NodeMCU_HardCrash.txt)

### Basic Infos  The attached file is a text file with the crash decoded. Checking the line numbers in the crash points to brackets{} in code.  The rest of the line numbers all seem to point to esp core files.
Is this a bug? Where to even begin to figure this out. 

#### Hardware
Hardware:NodeMCU V1			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  	NodeMCU V1		Flash Size: 4MB
CPU Frequency:	80Mhz?
Upload Using:	Serial and OTA

Any help would be greatly appreciated.

devyte, 2018-01-23:
Looks like mem corruption or oom. Without a minimal sketch to reproduce the issue it's unlikely anyone will be able to help.

devyte, 2018-01-23:
Also, which core version? When you posted the issue you were presented with a template. Please edit your post and fill in the requested info.

JakeBeck, 2018-01-23:
Which core? How to get?
The dump file quite a bit of information in it all related to the core files. Checking my code as I said leads to function bracets{}. Which says absolutely nothing about coding errors but does say tons about core issues.
I did fill in the template. How are brackets{} going to tell you anything at all? Oh I see the ? marks have you confused.

JakeBeck, 2018-01-23:
Here is one section that errors by the line numbers in the crash dump.

/// Manage network disconnection
void onSTADisconnected(WiFiEventStationModeDisconnected event_info)
  {
    //Serial.printf("Disconnected from SSID: %s\n", event_info.ssid.c_str());
    //Serial.printf("Reason: %d\n", event_info.reason);
    WriteToInfoLog("onSTADisconnected ", event_info.ssid.c_str(), event_info.reason + "  " + NTPTime);
    NTPStart = true;
    delay(10000);
    //digitalWrite(ONBOARDLED, HIGH); // Turn off LED
    //NTP.stop(); // NTP sync can be disabled to avoid sync errors
  }

As I said the line numbers match the brackets of this function. Just how are those brackets helpful?

JakeBeck, 2018-01-23:
And the other line number refernce the brackets in this section of code.

//////------      infoWrite = Sub(eg. wifi, eg.NTPTime)so on       ------//////
void WriteToInfoLog(String infoWrite, String infoWrite1, String infoTime)
  {
    if (spiffsActive)
      {
        if (SPIFFS.exists(INFOLOG))
          {
            File f = SPIFFS.open(INFOLOG, "a");
            f = SPIFFS.open(INFOLOG, "a");
            if (!f)
              {
                Serial.print("Unable To Open '");
                Serial.print(INFOLOG);
                Serial.println("' for Appending");
                Serial.println();
                server.send (200, "text/html", "The Info Log file is corrupt");
              }
            else
              {
                f.print(infoWrite);
                f.print(infoWrite1);
                f.println(infoTime);
                //f.print(infoWrite + infoWrite1 + infoTime + "\n");
                f.close();
              }
          }
        else
          {
            File f = SPIFFS.open(INFOLOG, "w");
            f.print(infoWrite);
            f.print(infoWrite1);
            f.println(infoTime);
            //f.print(infoWrite + infoWrite1 + infoTime );
            f.close();
          }
      }
  }
Again the line numbers references the function brackets.

devyte, 2018-01-24:
@JakeBeck The line numbers aren't always accurate in the Exception decoder. This is a known issue with the xtensa tools. You have to look at the dumped stack and figure out what the call sequence is in your code to reach the crash point.
I realize from your dump that you're using 2.4.0. The release has some issues, please retest with latest git. Instructions to install are on readthedocs.
Like I said, without a minimal sketch that reproduces the issue it's unlikely we'll get anywhere. Will you be providing it?

tonton81, 2018-01-24:
it should be event.ssid and event.reason, not event_info, last i checked

tonton81, 2018-01-24:
ahh sorry i see you renamed the callback variable, i was going by original

JakeBeck, 2018-01-24:
After backtracking through my work I've narrowed it down to the following code segment.

`int counther = 0;
    File j = SPIFFS.open(FUELHISTORY, "r");
    
    if (!j) 
      {
        Serial.println("file open failed");
      }
    else 
      {
        //Finds out how many records there are in the main data file
        while (j.available())
          {
            counther ++;
            j.readStringUntil('\n');
          }
        
        j.close();
      }
    
    Serial.println(counther);
    
    //if (count > 672)  //One week of records tobe ftped to the PI
    if (counther > 672)
      {
        if (doFTP(FUELHISTORY)) 
          {
            Serial.println(F("FTP Data OK"));     //Upload the data file using the doFTP function
            WriteToInfoLog("FTP Data Transfer OK ", "Creating New File ", NTPTime );
            SPIFFS.remove(FUELHISTORY);
          }
        else                                      //Returns true or false dependin on success
          {
            Serial.println(F("FTP Data FAIL"));
          }
      }`
It took me quit a bit of codeblock comments to arrive at this black of code to finally get it to stop hard crashing. If I put the comment block qoutes around this code no more crashes. If I remove them it starts crashing as soon as the ide closes the boot loader. I will be udateing the IDE a little later on today when I have the chance. So I do see that the debug output line numbers really are useless since this code blocks line numbers are 1187 to 1221. I'm also not sure how the crash dump relates to this code .

devyte, 2018-01-24:
I see:
- opening a file
- reading through its entire contents, upwards of 672 lines, which likely means several KB
- closing a file
- Serial operations
- an FTP operation, which I assume is meant to upload a file
- WriteInfoToLog, which I assume opens a file, appends to it, and closes the file
- file deletion

The entire thing seems very long running. How long does the whole code take front to back, starting at lowest level, e.g.: the loop or ticker callback or whatever?

JakeBeck, 2018-01-24:
Not very long at all. I'm doing the same on a web page and that only takes 2 or 3 seconds over the network. The ftp transfer is faster then loading the webpage. These two files are simple comma separated files. And the core version is 2.4.0

Here's a copy of the webpage for the file that can only reach 672 lines.
`server.sendHeader("Cache-Control", "no-cache, no-store, must-revalidate");
    server.sendHeader("Pragma", "no-cache");
    server.sendHeader("Expires", "-1");
    server.setContentLength(CONTENT_LENGTH_UNKNOWN);
    
    String page = "<!DOCTYPE html><link rel=\"icon\" type=\"image/ico\" href=\"/favicon\"><html>";
    page.concat("<head><meta http-equiv=\"content-type\" content=\"text/html\" charset=\"UTF-8\">");
    page.concat("<title>Fuel Acquisition System</title></head><body text=\"#000000\" ");
    page.concat("vlink=\"#551A8B\" bgcolor=\"#99ffff\" alink=\"#EE0000\" link=\"#0000EE\">");
    page.concat("<div align=\"center\"><font color=\"#000099\" size=\"+2\">");
    page.concat("<font size=\"+3\">Fuel Acquisition</font><br></font><div align=\"left\">");
    page.concat("<br></div><div align=\"left\"><p><p align =\"left\" style=\"color: ");
    page.concat("darkblue\"><font size=\"5\"><u>System Information</u></font></p>");
    page.concat("<div align=\"left\"><ul><ul><ul><li><div align=\"left\"><div align=\"left\"><pre>");
    page.concat("<font color=\"#000099\" size=\"5\">Runtime:");
    page.concat(RunTime);
    page.concat("     Free Disk Space: ");
    page.concat(T_Size);
    page.concat("         Current Time: <iframe src=\"/CurTime\" name=\"/CurTime\" ");
    page.concat("frameborder=\"0\" width=\"133\" height=\"31\" scrolling=\"no\">");
    page.concat("</iframe></font></pre></div></div></li></ul></ul></ul></div>");
    page.concat("<hr color=\"#000099\" size=\"4\" width=\"100%\"><center><table border=\"3\" ");
    page.concat("cellspacing=\"2\" cellpadding=\"2\" bgcolor= \"#ffff99\" width=\"80%\">");
    page.concat("<br><caption><font color=\"#000099\" size=\"+1\"> ");
    page.concat("Fuel Usage History<br></font></caption><tbody>");
    ///=========   Build dynamic grid table to be filled with historic data(FUELHISTORY)  =========///
    page.concat("<tr><th>Date</th><th>Time</th><th>Gallons</th><th>Temperature</th></tr>");
    ///============      The above line ^^^^^^^^^^^^^ puts column headers in place       ======///

    server.send(200, "text/html", "");
    server.sendContent(page);
    page ="";

    if (spiffsActive) 
      {
        if (SPIFFS.exists(FUELHISTORY)) 
          {
            File fuel = SPIFFS.open(FUELHISTORY, "r");
            if (!fuel) 
              {
                Serial.print("Unable To Open '");
                WriteToInfoLog("fuelhistoryPage() "," Unable To Open File ","File is Corrupt or Missing");
              } 
            else
              {
                String s;
                yield();
                while (fuel.position()<fuel.size())
                  {
                    s=fuel.readStringUntil('\n');
                    s.trim();
                    page="<tr><td valign=\"Top\">";
                    page.concat(s.substring(0,11));
                    page.concat("<br></td><td valign=\"Top\">");
                    page.concat(s.substring(11,19));
                    page.concat("<br></td><td valign=\"Top\">");
                    page.concat(s.substring(20,26));
                    page.concat("<br></td><td valign=\"Top\">");
                    page.concat(s.substring(27,33));
                    page.concat("<br></td></tr>");
                    
                    server.sendContent(page);
                    page="";
                  } 
                Serial.println("");
                fuel.close();
                page="";
                page="</tbody></table><br><div align=\"center\">";
                page.concat("<form id=\"/\" action=\"/\" method=\"post\">");
                page.concat("<input type=\"submit\" name = \"\" value=\"Back\"></form><br>");
                page.concat("</div></div></div></body></html>");
                server.sendContent(page);
              }
            
            server.sendContent(""); // this tells web client that transfer is done
            server.client().stop();
            
            return page;
          }
        else
          {
            page="</tbody></table><br><div align=\"center\">";
            page.concat("<form id=\"/\" action=\"/\" method=\"post\">");
            page.concat("<input type=\"submit\" name = \"\" value=\"Back\"></form><br>");
            page.concat("</div></div></div></body></html>");
            server.sendContent(page);
            server.sendContent(""); // this tells web client that transfer is done
            server.client().stop();
          }
      }  
    return page;      `

JakeBeck, 2018-01-25:
After further troubleshooting, I've found it isn't the file loop code causing this hard crash. It is actually the function that the brackets refer to. The entire function. Listed in the code tags;

`void onSTADisconnected(WiFiEventStationModeDisconnected event_info)
  {
    //Serial.printf("Disconnected from SSID: %s\n", event_info.ssid.c_str());
    //Serial.printf("Reason: %d\n", event_info.reason);
    //WriteToInfoLog("onSTADisconnected ", event_info.ssid.c_str(), event_info.reason + "  " + NTPTime);
    NTPStart = true;
    //NTP.stop(); // NTP sync can be disabled to avoid sync errors
  }
`
I have no idea why. I've we worked my entire code base including everything I commented out but this function. It's been running now for over half a day without error. I tried to further narrow it down by uncommenting this block of code and commenting writetoinfolog. This way it was only flipping a boollean value and it still causes a hard crash that is unrecoverable from. You must press the nodeMCU reset button to get it to start working again.

Any suggestions? I'll keep this open for now for a couple of days. If anyone has an idea I'd be eternally grateful, but will be marking this solved next week. Oh and this part of the code that doesn't work comes from this example:
https://github.com/gmag11/NtpClient/blob/master/examples/NTPClientESP8266/NTPClientESP8266.ino

Thanks to all that did reply, Jake

devyte, 2018-01-25:
@JakeBeck 
Crashing within that event handler is most likely not due to the function itself, but due to mem corruption or oom somewhere else. There is an option to enable oom debug in the Arduino IDE.  You also have to set the Debug port to get the msgs (@d-a-v please check me on that).
Before you spend further effort trying to figure out what's wrong with your app, I strongly suggest that you try with current master branch. Installation instructions with git are on readthedocs. Don't forget to uninstall 2.4.0 before installing git, and restarting the Arduino IDE after.

d-a-v, 2018-01-25:
@JakeBeck I'm not sure one can do such complex things as in ```WriteToInfoLog()``` (edited above) in a callback. I'd set a global variable in it then check this variable in the main loop to do the SPIFFS and server stuff.
@devyte yes it's ```Tools>DebugPort>Serial``` and ```Tools>DebugLevel>CORE+...+OOM```

devyte, 2018-01-30:
@JakeBeck here are some things I missed before. From your code above;
```
void onSTADisconnected(WiFiEventStationModeDisconnected event_info)
{
    WriteToInfoLog("onSTADisconnected ", event_info.ssid.c_str(), event_info.reason + "  " + NTPTime);
    NTPStart = true;
    delay(10000);
}
```
=>
`event_info.reason + "  " + NTPTime`
=>
`event_info.reason` is an enum
`"  " `is a const char *
`NTPTime `is an int (probably)

An enum is the same as an int. If you add an int to a const char*, you just advance the pointer (basic C language usage), so it ends up pointing to garbage. On passing a pointer to garbage to the constructor of String... well, you can guess.
You probably mean to do concatenation, so something like this:
`String(event_info.reason) + "  " + NTPTime`

Also, you can't do such a huge delay. Delete that.

Also, I believe that the WiFi event callbacks are SDK tasks, and you can't do blocking operations within tasks. If you need to do blocking operations, set a flag within the callback and check it in the loop. Alternatively, schedule a function to execute as though it were called from the next loop (see Schedule.h, declaration of schedule_function() ). A scheduled function executes only once. There's currently a maximum number of functions that you can schedule to execute in the next loop() run, but it should be enough for your purposes.

Also, I lost count at about 1.5KB of const char strings in your code, which you then concat in a series of operator+(). The const char strings are kept in RAM, which reduces your heap, and the series of concats fragments your heap all over the place, which increases the likelihood of oom, even if your free heap shows as enough. This is an embedded system with very limited resources. Instead of expecting the ESP to do the heavy lifting, I suggest moving the effort of putting the html together to the browser. You do this by creating a static html file to be served from SPIFFS, plus some minimal js code, then have the ESP dynamically create a minimal .json (as opposed to dynamically create the entire html page) that the js code retrieves. The json values are then injected into the html to show the final page.

Also, if you're going to serve html, use the webserver instead of trying to do the parsing and wificlient handling yourself. That's what it's for.

Closing as user error.

closed 2018-01-30 by devyte