esp8266/Arduino #4227: client.connected() function is not working. it keep returning true even if connection is lost!!!!!

bapidebabrata, 2018-01-23:
SDK version: latest master
`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
WiFiClient client;

const char* ssid     = "3C2i4b0k";
const char* password = "3C2i4b0kel6b4rS6eP15d5zujp19CEjhqYLH3MAl0QWmwYzQQyZM07Mq5i60";

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

}
//global variables for checkCommand()
bool authenticated=false;
const char* host ="192.168.137.1";
uint16_t port=31587;
//function definitions for check command.
void checkCommand()
{
  if(!client.connected())
  {
   Serial.println("Commecting to host...");
   if(client.connect(host, port))
   {
    Serial.println("Client connected!");
    //client.write(authStr);
    client.write("AuthStr");
    client.keepAlive(5,1,5);
    Serial.printf("KeepAlive(%d,%d,%d,%d)\n",client.isKeepAliveEnabled(),client.getKeepAliveIdle(),client.getKeepAliveInterval(),client.getKeepAliveCount());
   }
  }
  else
  {
    
  }
}

void loop() {
checkCommand();
}`

client.connected() function is not working. it keep returning true even if connection is lost(Server closed.)!!!!!

bapidebabrata, 2018-01-23:
There is no way to know whether connection is connected or not if there is data on receiver buffer.

bapidebabrata, 2018-01-23:
`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
WiFiClient client;

const char* ssid     = "3C2i4b0k";
const char* password = "3C2i4b0kel6b4rS6eP15d5zujp19CEjhqYLH3MAl0QWmwYzQQyZM07Mq5i60";

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

}
//global variables for checkCommand()
bool authenticated=false;
const char* host ="192.168.137.1";
uint16_t port=31587;
//function definitions for check command.
void checkCommand()
{
  if(!client.connected())
  {
   Serial.println("Commecting to host...");
   if(client.connect(host, port))
   {
    Serial.println("Client connected!");
    //client.write(authStr);
    client.write("AuthStr");
    client.keepAlive(1,0,0);
    Serial.printf("KeepAlive(%d,%d,%d,%d)\n",client.isKeepAliveEnabled(),client.getKeepAliveIdle(),client.getKeepAliveInterval(),client.getKeepAliveCount());
   
   }
  }
  else
  {
    if(client.available())
    {
      char a=client.read();
      Serial.print(a);
    }
  }
}

void loop() {
checkCommand();
}`

The above code works fine. Because there is no data at the time of connection termination.

NayanKaran, 2018-01-23:
client.status() is working fine.

closed 2018-01-23 by NayanKaran

vc-sagar, 2019-01-24:
how to data transmission and receiving at time

kritika38734, 2019-05-22:
I have the same problem, I never started the server on my laptop and ran the client code on Arduino for ESP32. It shows connected