esp8266/Arduino #4229: Acknowledge response to http update

Vinayakj009, 2018-01-24:
Please fill the info fields, it helps to get you faster support ;)

While using the httpupdate library to issue an OTA to a firmware, the system directly updates the new retrieved update.

This is troublesome, if you are designing a product which has OTA updates and the user has to trigger the update.

I feel there should be a callback in the htttp update library, which will allow the old firmware to signal to the user that the OTA is received properly, before updating to the new firmware.

I have tested the same on my local system, and seems to be working fine. But I have no clue how to put it up here.

Any help regarding the same would be great.

devyte, 2018-01-24:
Fork  the repo in this webpage
Git clone your fork, this creates a local repo on your computer
Create a branch in your local repo
Edit whatever files necessary
Git commit to your local repo
Git push to your fork
On this repo's webpage you should see a prompt about your push and a button to create a PR (always review your changes on the webpage before pressing that button)

Vinayakj009, 2018-01-27:
@devyte 👍 
Will get to it on Monday.

devyte, 2020-05-14:
Closing due to age and lack of feedback.
If you think there is still a problem, please retest with latest release or latest git, then open a new issue and follow the template instructions.

closed 2020-05-14 by devyte