esp8266/Arduino #4230: Retrieving softAP SSID

jsphuebner, 2018-01-24:
Hi,

added a function to ESP8266WifiAP.cpp for retrieving the stored softAP SSID

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.print("softAP SSID: ");
  Serial.print(WiFi.softAPSSID());
}

void loop() {

}

```

### Add to ESP8266WifiAP.cpp

```cpp

String ESP8266WiFiAPClass::softAPSSID() {
    struct softap_config config;
    String ssid = "";
    
    if (wifi_softap_get_config(&config))
        ssid = reinterpret_cast<char*>(config.ssid);
    return ssid;
}

```

### Add to ESP8266WifiAP.h

```cpp

String softAPSSID();

devyte, 2018-01-24:
Already implemented and reviewed in #4138 .

yoursunny, 2018-08-27:
#4138 is merged. I guess this issue can close?