esp8266/Arduino #4231: Esp8266 can't connect to the router with the core 2.4.0, but it can connect with  2.3.0

ramonlh, 2018-01-24:
The esp8266 can't connect to the router with the core 2.4.0, but it can with the core 2.3.0.

### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	2.3.0

### Description
With 2.3.0 connection OK

### Settings in IDE

Module:  			NodeMCU 0.9 (ESP-12 Module)
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:			
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

lwIP variant:      v2 Prebuilr (MSS0536)   (for 2.4.0. Core)

### Sketch

/*  This sketch connect to WiFi */
#include <ESP8266WiFi.h>
char ssid[]="MOVISTAR_8453";
char pass[]="********";

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);}

void loop() {
    Serial.println();
    Serial.print("Connecting to:"); Serial.println(ssid);;
    Serial.print("Wait for WiFi...");

    if (WiFi.isConnected())    {WiFi.disconnect();}
    
    WiFi.begin(ssid,pass);
    while(!WiFi.isConnected()) {
      Serial.print(".");
      delay(500);      }

    Serial.println();
    if (WiFi.isConnected())    {
      Serial.println("WiFi connected");
      Serial.print("IP address: ");   Serial.println(WiFi.localIP()); }
    else
      Serial.println("WiFi NOT CONNECTED--------------------------");  }
```

### Debug Messages

**With 2.3.0 Core**
    Connecting to:MOVISTAR_8453
    Wait for WiFi..........
    WiFi connected
    IP address: 192.168.1.64

**With 2.4.0. Core**
Connecting to:MOVISTAR_8453
Wait for WiFi...sta config unchangedscandone
.wifi evt: 2
....scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
....scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
....scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
.....scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
....scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
....scandone

```
messages here
```

lrmoreno007, 2018-01-24:
I can connect well with your code and 2.4.0. Please check that the SSID name and password are correct.
[image]

devyte, 2018-01-25:
I am also using latest git on several Nodemcu and Wemos D1 minis, and I don't have connection problems.
I suggest erasing the entire flash before flashing each time.

BTW, you probably shouldn't be calling
```
if (WiFi.isConnected())    {WiFi.disconnect();}
WiFi.begin(ssid,pass);
```

inside the loop. You're essentially disconnecting and reconnecting as fast as the ESP can execute. I've seen odd things happen with routers, and I seem to recall some router apparently blocking such quick repeated reconnections. Although, that might be a bit sophisticated for a VOMISTAR router :P 
Anyways, instead of checking the WiFi status in a loop, I suggest using the WiFi event system to set/clear a flag, and check that flag in the loop.

Closing due to can't reproduce.

closed 2018-01-25 by devyte

carrch, 2019-04-17:
We are having same issue with connecting to some APs and not others. We have a ESP8266 in a dispenser that connects to the WiFi sends dispenser capacity readings that sleeps. Currently we are polling every minute which means sleep, wake up connect send data back to sleep repeat. 

When it tries to connect to most office APs it never completes the connection.    

void connect()
{
    boolean state = true;
     Serial.println("");
    Serial.println("In Connect ");
    if (WiFi.status() == WL_CONNECTED)
    {
        ControlLed(2, true);
     delay(500);
     ControlLed(2, false);
      Serial.println("Showing  Connected!!!!!!!!!!!! ");
        return;
    }

    // Connect to WiFi network
    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(DeviceConfig.SSIDName);

    WiFi.begin(DeviceConfig.SSIDName, DeviceConfig.SSIDPass);

    while (WiFi.status() != WL_CONNECTED)
    {
        ControlLed(greenLed, state);
        if(state == HIGH)
          state = LOW;
        else
          state = HIGH;
          
        delay(500);
        Serial.print(".");
        CheckWIFIReset();
        if (resetDevice)
        {
          resetCount++;
          Serial.println("Will reset!");
          if (resetCount == 5)
          {
            resetDevice = false;
            ESP.reset();
          }
        }  

        WIFIConnectionTryCnt++;

        if(WIFIConnectionTryCnt > 240)
          ESP.restart();
    }

    while(digitalRead(resetWIFISwitch) == LOW)
      CheckWIFIReset();
      
    ControlLed(greenLed, LOW);
    Serial.println("");
    Serial.println("WiFi connected");
     ControlLed(blueLed, true);
     delay(500);
     ControlLed(blueLed, false);
}

void InitAccessPoint(void)
{  

  char ssid[MAX_DEVICE_ID_SIZE];
   
  Serial.println();
  Serial.println("Configuring access point...");

  //set-up the custom IP address
  
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));   // subnet FF FF FF 00  

  if(strlen(DeviceConfig.ConnectionString) != 0)
  {
      if (GetDeviceId(DeviceConfig.ConnectionString, ssid, MAX_DEVICE_ID_SIZE) > 0)
      {
        strcat(ssid, "_AP");
        Serial.println(ssid); 
        WiFi.softAP(ssid, password);         
      }
  } else {
    /* You can remove the password parameter if you want the AP to be open. */
    WiFi.softAP(APssid, password);
  }
  
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
}

sergeyatarasov, 2019-11-19:
I had the same problem when i could connect to my network at work but not at home, even if my code was from standart library with no changes exept SSID and Password. So i tried to reset wifi settings as mentioned above and it worked! 
I've just set in arduino:  Tools - Erase flash - "Sketch+WiFi settings" instead of "Only sketch"