esp8266/Arduino #4235: Workaround for Alexa Bug

reloxx13, 2018-01-24:
Heya,

this library is used by the Tasmota Firmware to emulate HUE and WEMO Devices to control them with Amazon Alexa.

Currently their seems to be a bug in the Alexa Firmware where the header content is wrong resulting in a wrong parsing and missing args.

Another dev made a workaround for it.

Please check this out and tell us, if their is any chance to include it in your library, since we cannot do anything in the Alexa Firmware.

https://github.com/arendst/Sonoff-Tasmota/issues/1639#issuecomment-360218536
https://github.com/arendst/Sonoff-Tasmota/issues/1434

devyte, 2018-01-24:
The referenced code rings a bell in my mind. There might already be a PR with the proposal, but I don't think Alexa was mentioned.
If there is no PR, please make one for review.

reloxx13, 2018-01-24:
seems u r right, but that pr is a lil bit diffrent.

https://github.com/esp8266/Arduino/pull/4151

i dont know which is better.

devyte, 2018-01-25:
@reloxx13 @Monarch73 please discuss the differences between this PR and #4151 . 
Are they both aimed at Alexa? 
Do they both work?
Are both code changes equivalent, or is one better than the other for some reason?

reloxx13, 2018-01-25:
Its independent from Alexa, but works for her, too.

As Monach said, it fixes the missing args if the server send the "application/x-www-form-urlencoded" header content-type instead of "application/json" and plain text (json) and not as suggestest for form-urlencoded a key/value pair.

i think monarches is better, cuz he made it in the _parseArguments method, while the one i posted is out of that method.

OFreddy, 2018-01-25:
I would prefer my solution as it saves code, because the code for adding the plain content is already existing.

reloxx13, 2018-01-25:
good point 👍

Monarch73, 2018-01-25:
I don't mind how it's done. Either way is fine with me.

OFreddy, 2018-01-25:
I am struggling with every byte to keep ota firmware update via web interface running ;-)
Not very satisfying

Monarch73, 2018-01-25:
I think if you do a direct size-comparison  of the binaries, you wouldn't get any differences. The optimizer in gcc picks up these kind of things pretty well...

how ever, like I said, I don't mind how it's done

OFreddy, 2018-01-25:
In that case I don't mind how it is done either.

themindfactory, 2018-01-27:
Can this work with Google home?

reloxx13, 2018-01-31:
any news about the PR ? :S

devyte, 2018-01-31:
Right now we're in a half-frozen state for 2.4.1, finishing up only critical stability issues and waiting on the imminent sdk 2.2. I'll revisit this after 2.4.1 is out the door.

Winfried-Bantel, 2018-03-05:
I've got the same problem:

I want to control some electrical gimmicks via Hue-V1-Protocol saying "alexa test 50%".
The complete traffic alexa-ESP is working, but on the last function i can't access PUT-Data in JSON-form because alexa sends as Content-type x-www-form-urlencoded and in this case the ESP8266WebServer parses the HTTP-body as Form-Parameters which fails for JSON.

I can get "Content-Length" but no content!
Look here:
http://www.burgestrand.se/hue-api/api/lights/#changing-light-color-and-turning-them-onoff

In my opinion the best soolution would be a body-method.

devyte, 2018-10-17:
Fixed with #5252 .

closed 2018-10-17 by devyte