esp8266/Arduino #4237: WeMos D1 mini Pro not work.

bestpika, 2018-01-25:
### Basic Infos

#### Hardware

Hardware: `WeMos D1 mini Pro`
Core Version: 8a24598d5dbf46effc5ac9189224601950736a70

### Description

Flash bin but device not work.

```txt
esptool.py v2.3-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4018
Detected flash size: 16MB
Hard resetting...
```

### Settings in IDE

Module: `WeMos D1 mini Pro`
Flash Size: `16M (15MSPIFFS)`
CPU Frequency: `80Mhz`
Upload Using: `Serial`
IwIP: I try `v1.4` and `v2 (MSS=536 and MSS=1460)`

I try `erase_flash` but still not work.

### Sketch

```cpp

#include <Arduino.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println(analogRead(A0));
  delay(100);
}

```

### Debug Messages

```txt

No any message.

```

igrr, 2018-01-25:
I see you are using esptool.py. Flashing with that tool is not something we test in this project. Please try to use esptool-ck launched from Arduino IDE.

bestpika, 2018-01-25:
I use Arduino IDE to flash my project.
I `erase_flash` before flash.

bestpika, 2018-01-25:
I can't provide my Sketch content, but I can upload bin.

bestpika, 2018-01-25:
16M build
[qbox.ino.zip](https://github.com/esp8266/Arduino/files/1662383/qbox.ino.zip)

4M build
[qbox.ino.zip](https://github.com/esp8266/Arduino/files/1662391/qbox.ino.zip)

igrr, 2018-01-25:
Without seeing what the sketch does, I can offer very little in terms of advice. However, please try enabling debug output (in IDE tools menu, with level set to CORE) and add Serial.setDebugOutput(true) after your Serial.begin line. This may result in some output which will help you figure out the issue.

bestpika, 2018-01-25:
I set debug CORE.
But after flash, this device can't work.
So I can't see any output.

bestpika, 2018-01-25:
I think this device hang.

bestpika, 2018-01-25:
@igrr I know what's happen.
After `SPIFFS.begin()`, this device hang.

```cpp
#include <FS.h>

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("setup start");
  SPIFFS.begin(); // die at here
  Serial.println("setup ended");
}

void loop() {
  Serial.println(millis());
  delay(100);
}
```

closed 2018-01-25 by bestpika

igrr, 2018-01-25:
Yes, formatting a 15MB SPIFFS filesystem can take very long time...

bestpika, 2018-01-25:
Can spiffs add some debug message?

igrr, 2018-01-25:
I don't see an existing API to have a callback on format process, unfortunately
https://github.com/pellepl/spiffs/blob/1350deb54c097fd06133f3c6d39dd3532a229829/src/spiffs.h#L612-L625