esp8266/Arduino #4238: How does make hellomesh code work ?

WillingLearner, 2018-01-25:
I have two esp8266 12e module and want to create mesh network using these modules. I loaded the same code to the two esp module but they doesn't communicate with each other. I am monitoring whats happening in Serial on the first ESP. Module is reset itself somehow and shows stack info at regular intervals Where am i doing wrong ?

Thanks in advance

skadiexe, 2018-01-25:
[image]

Mine show connections (debug enabled) but regularly reboots too. It is inusable thing, open ESP8266WiFiMesh and see.

Using easymesh/painlessMesh could be very interesting, but now esp8266 lib 2.4.0 does not support easymesh

WillingLearner, 2018-01-25:
I have never tried another libraries but HelloMesh so far. Did you manage to perform mesh network by using esp8266 in any way skadiexe ?

rkoptev, 2018-04-02:
I have similar behavior. I flashed HelloMesh example to 2 ESP8266 NodeMCU boards and they stucks on "Setting up mesh node...". While one of nodes just stuck - second one constantly rebooting.

devyte, 2018-08-01:
Fixed via #4718 .

closed 2018-08-01 by devyte