esp8266/Arduino #4239: WPA2 Enterprise EAP-TLS Error

rovingclimber, 2018-01-25:
#### Hardware
Hardware:			NodeMCU v0.9 (ESP-12)
Core Version:      	2.1.0(deb1901)

### Description

I have failed every attempt to get WPA2 enterprise to work. I've been through every forum entry on this, everyone seems to say it should work but I've been unable to get any example working. I keep getting a mysterious error: "Method private structure allocated failure", which seems to come at the point you'd expect a successful connection. Googling finds a couple of people seem to have hit the same point but no resolution.

I have two IDE PCs, just set one up from scratch with fresh install of Arduino and this lib, still same problem. Interestingly I have also been trying to get this working on an ESP32 using the Arduino lib and get exactly the same error (with quite different code) so something from an underlying lib?

Any help would be gratefully received!

### Settings in IDE

Module:  			NodeMCU 0.9 (ESP-12 Module)
Flash Size: 		4MB (1Mb SPIFFS)
IWIP Variant:            v2.0 Prebuild (MSS=536)
CPU Frequency:		80Mhz

### Sketch

```c
#include "ESP8266WiFi.h"
extern "C" {
  #include "user_interface.h"
  #include "wpa2_enterprise.h"
  #include "c_types.h"
  #include "cert.h"
  #include "cont.h"
  extern cont_t g_cont;
}

static const char* ssid = "myssid";

void wifi_handle_event_cb(System_Event_t *evt)
{
    switch (evt->event) {
        case EVENT_STAMODE_CONNECTED:
            os_printf("[connect to ssid %s, channel %d]\n",
                evt->event_info.connected.ssid,
                evt->event_info.connected.channel);
            break;
        case EVENT_STAMODE_DISCONNECTED:
            os_printf("[disconnect from ssid %s, reason %d]\n",
                evt->event_info.disconnected.ssid,
                evt->event_info.disconnected.reason);
            break;
        case EVENT_STAMODE_AUTHMODE_CHANGE:
            os_printf("[mode: %d -> %d]\n",
                evt->event_info.auth_change.old_mode,
                evt->event_info.auth_change.new_mode);
            break;
        case EVENT_STAMODE_GOT_IP:
            os_printf("[ip:" IPSTR ",mask:" IPSTR ",gw:" IPSTR "]\n",
                IP2STR(&evt->event_info.got_ip.ip),
                IP2STR(&evt->event_info.got_ip.mask),
                IP2STR(&evt->event_info.got_ip.gw));
            break;
        default:
            os_printf("[event: %x]\n", evt->event);
            break;
    }
}

void setup(){
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.printf("SDK version: %s\n", system_get_sdk_version());
  Serial.printf("Free Heap: %4d\n",ESP.getFreeHeap());
  Serial.printf("Free Stack: %4d\n",cont_get_free_stack(&g_cont));

  Serial.printf("Connecting to %s\n", ssid);
   // Setting ESP into STATION mode only (no AP mode or dual mode)
  if (!wifi_set_opmode(STATION_MODE)) {
    Serial.print("Failed setting station mode");
  }

  struct station_config wifi_config;
  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);

  Serial.println("Calling wifi_station_set_config");
  Serial.println(wifi_station_set_config(&wifi_config));
  
  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();

  Serial.println("Calling wifi_station_set_wpa2_enterprise_auth");
  Serial.println(wifi_station_set_wpa2_enterprise_auth(1));

  Serial.println("Calling wifi_station_set_enterprise_cert_key");
  Serial.println(wifi_station_set_enterprise_cert_key((uint8*)user_cert, sizeof(user_cert),
      (uint8*)user_key, sizeof(user_key), (uint8*)user_key_pass, sizeof(user_key_pass)));
  
  Serial.println("Calling wifi_station_set_event_handler_cb");
  wifi_set_event_handler_cb(wifi_handle_event_cb);
  
  delay(20000);
  Serial.println("Calling wifi_station_connect");
  wifi_station_connect();

  Serial.print("Wifi station connect status:");
  Serial.println(wifi_station_get_connect_status());

  // Wait for connection AND IP address from DHCP
  while (WiFi.status() != WL_CONNECTED) {
  Serial.println();
    Serial.println("Not connected");
    WiFi.printDiag(Serial);
    delay(20000);
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop()
{
  delay(2000);
}
```
cert.h:
```c
unsigned char user_cert[] = {
0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x42, 0x45, 0x47, 0x49, 0x4E, 0x20, 0x43, 0x45, 0x52, 
0x54, 0x49, 0x46, 0x49, 0x43, 0x41, 0x54, 0x45, 0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x0D, 
0x0A, 0x4D, 0x49, 0x49, 0x49, 0x6B, 0x44, 0x43, 0x43, 0x42, 0x6E, 0x69, 0x67, 0x41, 
...
0x4F, 0x79, 0x45, 0x39, 0x66, 0x70, 0x62, 0x5A, 0x33, 0x0D, 0x0A, 0x2D, 0x2D, 0x2D, 
0x2D, 0x2D, 0x45, 0x4E, 0x44, 0x20, 0x43, 0x45, 0x52, 0x54, 0x49, 0x46, 0x49, 0x43, 
0x41, 0x54, 0x45, 0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x00
};
unsigned char user_key[] = {
0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x42, 0x45, 0x47, 0x49, 0x4E, 0x20, 0x52, 0x53, 0x41, 
0x20, 0x50, 0x52, 0x49, 0x56, 0x41, 0x54, 0x45, 0x20, 0x4B, 0x45, 0x59, 0x2D, 0x2D, 
0x2D, 0x2D, 0x2D, 0x0A, 0x50, 0x72, 0x6F, 0x63, 0x2D, 0x54, 0x79, 0x70, 0x65, 0x3A, 
...
0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x45, 0x4E, 0x44, 0x20, 0x52, 0x53, 0x41, 0x20, 0x50, 
0x52, 0x49, 0x56, 0x41, 0x54, 0x45, 0x20, 0x4B, 0x45, 0x59, 0x2D, 0x2D, 0x2D, 0x2D, 
0x2D, 0x00	
};
unsigned char ca_pem[] = {
0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x42, 0x45, 0x47, 0x49, 0x4E, 0x20, 0x43, 0x45, 0x52, 
0x54, 0x49, 0x46, 0x49, 0x43, 0x41, 0x54, 0x45, 0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x0D, 
0x0A, 0x4D, 0x49, 0x49, 0x46, 0x73, 0x54, 0x43, 0x43, 0x41, 0x35, 0x6D, 0x67, 0x41, 
...
0x36, 0x35, 0x4A, 0x4E, 0x49, 0x41, 0x62, 0x0D, 0x0A, 0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 
0x45, 0x4E, 0x44, 0x20, 0x43, 0x45, 0x52, 0x54, 0x49, 0x46, 0x49, 0x43, 0x41, 0x54, 
0x45, 0x2D, 0x2D, 0x2D, 0x2D, 0x2D, 0x00
};
unsigned char user_key_pass[] = "mykeypass";
```
### Debug Messages

SDK version: 2.1.0(deb1901)
Free Heap: 37224
Free Stack: 1236
Connecting to myssid
Calling wifi_station_set_config
1
Calling wifi_station_set_wpa2_enterprise_auth
WPA2 ENTERPRISE VERSION: [v2.0] enable
0
Calling wifi_station_set_enterprise_cert_key
0
Calling wifi_station_set_event_handler_cb
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 15
cnt 
Method private structure allocated failure
pm open,type:2 0
Calling wifi_station_connect
Wifi station connect status:1

Not connected
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (8): myssid
Passphrase (0): 
BSSID set: 0
sl
scandone
usl

rovingclimber, 2018-02-07:
No-one? Has anyone managed to get EAP-TLS working and have any suggestion?

daisthorpe1, 2018-04-09:
Hi did you get this solved I am trying to connect to the BTwfi-x

tspspi, 2019-02-26:
I'm currently getting the same error - in my case I've also tried to set the CA certificate (`wifi_station_set_enterprise_ca_cert`) and identity (`wifi_station_set_enterprise_identity`). What I'm seeing in my freeradius logs is that the Node-MCU is never requesting EAP-TLS but only MS-CHAP-V2, PEAP and EAP-TTLS and the request being rejected (which seems not to be recognized by the NodeMCU).

herste, 2019-10-25:
Any news on that? I even tried to compile the wpa2 example of the latest Non-OS SDK (3.1.0-dev(b2602cc)) without any success. The ESP8266 throws an exception, I never see EAP-TLS requests hitting the Radius server.

Help appreciated!