esp8266/Arduino #4240: Translation to different languages

lrmoreno007, 2018-01-25:
### Description

I want to propose the translation of the documentation into different languages ​​and I offer to translate it to Spanish.

I don't know if this is feasible and how it can be done.

At the moment, I have already translated the Readme.md file in my project's fork as an example.
https://github.com/lrmoreno007/Arduino/blob/master/README.md

Regards

devyte, 2018-01-25:
@lrmoreno007 no es mala idea :) I have the following comments:
- if this is done, it should be stressed somewhere that issues and PRs should be opened only in English
- how would we make sure that an update in the docs in one language gets correctly done in all other languages as well?
- there is a lot of documentation that is still pending, including documentation-related issues. If you are willing to contribute with documentation, would you be open to writing some of the missing docs, where possible?

tonton81, 2018-01-26:
why not just google translate the entire website and call it a day? it requires no conversion and no upkeep

lrmoreno007, 2018-01-26:
@devyte Do I answer you:
* It's easy, take a look at my fork again.
* This is the hardest thing. The truth is that I have no idea how this change control can be done, but it occurs to me: make a markup on each PR that affects the documentation to notify the different people in charge of the translation.
* My ability to translate from Spanish to English, or to express myself in English is quite limited. And more when I'm not an Arduino / C programming expert. But if you tell me how to find that missing documentation, I would gladly try it.

Just add that Spanish is the second most spoken native language with 437 million (mandarin Chinese is the first 836 millions), compared to 375 million native English speakers. With this I mean that it is a good opportunity to open the doors to many people.

@tonton81 The English to Spanish translation of Google Translator is not very good in these cases for the following reasons:
- There are many technical words that should not be translated or lose meaning.
- Spanish has many verb tenses and even a verb in English has two or three meanings in Spanish, as for example the verb "to be" in spanish "ser", "estar", haber. (Take a look at this: http://www.worddive.com/grammar/en/spanish-grammar/7-verbs-ser-estar-hay/)
I don't know other languages or as the Spanish to English translation of Google Translator sounds. I have made my answer with him :) .

lrmoreno007, 2018-02-16:
OK, I create a new repository only with the documentation translated to spanish:
 https://github.com/lrmoreno007/ESP8266-Arduino-Spanish

And I create the ReadTheDocs project in spanish with the documentation:
http://esp8266-arduino-spanish.readthedocs.io/es/latest/index.html

It only remains to translate the folders: FAQ and ESP8266WiFi, I'll do it soon.

Tomorrow, I will make a PR with some modifications in: README.md, POLICY.md and index.md, so that it is linked to this documentation and warns that all communications must be done in English.

I only claim that all accepted PRs that modify something in the documentation be marked with the label DOC, for to be able to find it easily, review it and translate it.

tonton81, 2018-02-19:
since you have a PR with comments it would be good to close this stray issue

closed 2018-02-19 by lrmoreno007