esp8266/Arduino #4242: Esp8266 Arduino 2.4.0 is buggy

sezeryalcin, 2018-01-25:
#### Hardware
Hardware:			Wemos D1 mini
Core Version:      	2.4.0

### Description

I have a fully functional code that in summary reads saved Wifi credentials from EEPROM and connect to a wifi.
I have Arduino 1.8.5. This code fully works and ESP is able to connect to wifi using 2.3.0. However I recently upgraded to 2.4.0 and device fails to to connect to wifi. All other functionality seems working fine. Device moves on if it cannot connect to wifi and other stuff works fine. LCD and some sensors are connected to it and all work fine with both 2.3.0 and 2.4.0

Now I downgraded to 2.3.0 and device seems working perfectly as it was. No code changes.

I couldn't find 2.4.0 changes to see if there are some defaults changed which would prevent this. I am connected to a WPA-PSK network using password.

### Settings in IDE

Module:  			Wemos D1 mini
Flash Size: 		4MB
CPU Frequency:		80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		 SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch
Cant provide code ATM.

```

### Debug Messages

```
messages here
```

onkie, 2018-01-25:
ESP8266 PainlessMesh and ArduinoWebSokets problems with 2.4.0 but perfect working with 2.3.0. 
All my ESP8266 modules in 2.4.0  Power OFF/ON, Reset (button/soft) or Restart then a DEAD ESP8266, with 2.3.0 no problem.

devyte, 2018-01-25:
This kind of issue doesn't help. Just saying "version 2.4.0 is buggy, code worked with 2.3.0", or "3rd arty lib xyzn no longer works" doesn't provide enough details to investigate, especially when code is not provided.
Some issues were found in 2.4.0, which have already been fixed in master, so I suggest trying latest git.
Also, ask not what this repo can do for you, but what you can do for this repo. Let's keep in mind that this is not a provided service, but a collaborative effort.

If you have a specific issue reproducible with a minimal sketch, then open an issue and fill in the requested info. If you find a fix, then open a PR referencing the issue, and someone will test/review.

Closing due to lack of info.

closed 2018-01-25 by devyte

sezeryalcin, 2018-01-25:
devyte I guess you are a jr contributor who doesn't want to use any available info. I have a legitimate bug and I report it. That's how people do it. Yes as a 20+ year experienced software developer, I could work more and even probably find and fix this issue but I don't have time. Least you could do is to ask for more information. Maybe I cannot but other people might provide more info.

You close a bug ONLY you are sure there is no bug and believe me there is a bug here. So honestly if you want to improve the code, I would recommend you reopen this issue as I won't bother to submit again.

sezeryalcin, 2018-01-25:
@onkie, if devyte doesn't reopen this, please file a new bug and hope that ivan or another guy triage it.

sezeryalcin, 2018-01-25:
Also @devyte I noticed that 2.4.0 master has fixes in git that you mentioned. I got 2.4.0 within Arduino and expect it to be stable. Maybe it was pushed too early. If you released 2.4.0 as stable, whatever is on git can only be 2.4.1, not 2.4.0.

onkie, 2018-01-25:
**@devyte** you can use the PainlessMesh and ArduinoWebsocket **examples** and don't look strange at all not working perfectly in 2.4.0
@sezeryalcin ;=)

devyte, 2018-01-25:
@sezeryalcin sweety, I'm not the most senior contributor, but I've been around for a while as such, and quite a bit longer even before I was invited to contribute. When I came in, this repo had over 1000 opened issues, and I read through all of them and triaged most. When I finished, the opened issues were cut in half.
In fact, I'm the contributor who both wrote and implemented the current issue policy here. I suggest you read it.
You say you have 20+ years of experience as a software developer, and I have a bit more than that, but in contrast to your 4 opened issues in the last year, and 0 PRs, I have actually contributed bug fixes and enhancements here, and reviewed and improved fixes by others. 
You think you have a legitimate bug, so then please go ahead and post the details. If you can't post the details, then don't waste our time. You are always free to investigate it yourself, within the confines of your privacy. You do have access to this repo's source code, after all.
About the 2.4.0 release, it is called a "release", not a "stable". It marks a point in time where we draw a line and make the current status of bug fixes and enhancements available via a prepackaged solution for easy installation. You can expect whatever you want from that, but the fact is that it is based on a whole lot of effort and blood and sweat from the people here who volunteer their time, and even so it is not perfect. I will stress here the "volunteer" part, and remind you that we are under no obligation to your needs or expectations. Instead of expecting from this repo, I strongly suggest that you find the time to test and investigate, like many others do. At the very least, that will provide additional info for someone else to find and implement a fix, even if you "don't have the time".
We are currently working towards a 2.4.1 release focused on stability, including fixes that are expected to arrive in a new Espressif SDK release in a few days.

devyte, 2018-01-25:
@onkie I understand, but the resources available to develop this repo are limited, and we don't debug 3rd party libs that are not a dependency, unless a developer here is invested in them, and takes on the task himself.
Instead, when faced with an issue, it is expected that an author or user of those libs come up with a reduced sketch to report the issue in this repo. Given enough details, we can figure out if the issue is a duplicate of something already reported, or something new that merits an investigation, or an exposed bug in those libs that was previously hidden, etc.

sezeryalcin, 2018-01-25:
@devyte Solution for this bug could be just marking 2.4.0 as not-stable and that could be it. I filed this bug only because Arduino IDE prompted to upgrade and it did, to an unstable release. I appreciate your work on this and I didn't mean to get any personal. I see problem is not 2.4.0 being unstable but 2.4.0 being released as if it's stable.

everslick, 2018-02-04:
funnily, in my experience the very LATEST git master is usually the most stable, most usable version available. i have not seen regressions often, if ever. I cannot recall a specific one at least. @sezeryalcin it could as well be a bug in your code that is not triggered in the 2.3.0 release, because the memory layout has changed. THAT kind of bugs I have seen hundreds of time. We are talking bare metal development here with no memory protection whatsoever. mem corruption can go unnoticed for a very long time if lucky (or unlucky, depending on how you prefer to look at it).

pieman64, 2018-02-04:
I concur with the comments made by @everslick and for the ESP8266 I **always** recommend users pick up the master and not the releases. I don't recommend the master for other repositories just this one. The 18 month gap between 2.3.0 and 2.4.0 is a prime example where seemingly hundreds of issues were resolved but probably 90% of users didn't benefit from them. This in part is why the backlog of issues got so big i.e. the issues had been resolved but very few users knew about them so they created "new" issues.

I believe 2.4.0 had some problems and they were relatively quickly fixed but again next to nobody will be picking up the master. @devyte I understand the importance of thorough testing before each release but couldn't we adopt a more aggressive approach. Simply issue a new release every week or so and users will quickly report any major problems. Then the release can be rolled back to the best available release.

igrr, 2018-02-04:
Release per week has some limitations (most notably, number of versions in the Arduino IDE drop-down being shown, plus the tame it takes to do the release related procedures), but point taken. We'll try to do release every couple of months or more often if there are major issues (like this time).

sezeryalcin, 2018-02-08:
@everslick I updated to git latest and I don't have any issues. I think that proves my point that 2.4.0 as released is buggy. This code also worked great on 2.3.0 and earlier versions, probably 2.2.0

pieman64, 2018-03-01:
@igrr any idea when all the recent bug fixes are going to be released to ESP8266 users around the world?

igrr, 2018-03-01:
@pieman64 we have one more issue related to WiFi.disconnect which needs to be fixed, once this is done we'll push the release. I'm at embedded world this week, but plan to look into the remaining issue on the weekend.

pieman64, 2018-03-01:
Thanks Ivan (@igrr ) have a good time at the Trade Fair.