esp8266/Arduino #4243: ArduinoIDE OTA mdns refresh button?

Fettkeewl, 2018-01-27:
Hi!

So I'm using this amazing ota functionality as much as possible but one major issue I have is that
it isn't so "prototyping" friendly.

My current steps to OTA
1 Set device in OTA mode
2 Start arduinoIDE
3 OTA discovered, hit upload
4 Wait sooooooo long for compile 
5 Enjoy result. 

After every succesful OTA, if I think of a change to implement, I must do ALL the steps above, this means closing ArduinoIDE and setting myself up to wait for that awfully slow compile.

Possible solution:
Add a "Discover" button under tools that forces a mdns scan so it can pick up my device, so I don't have to restart Arduino for it and do the compile dance of slow death.

Result I'm hoping for, is

"First OTA"
1 Set device in OTA mode
2 Start arduinoIDE
3 OTA device discovered, hit upload
4 Wait sooooooo long for compile 
5 Enjoy result. 

"Subsequent OTA"
1. Make a change
2. Set device in OTA
3. Hit new Magic button to scan mDNS for OTA devices (instead of restarting arduino IDE)
4. OTA device discovered, hit upload
5. Fast compile!

MarkyAD, 2018-01-28:
Once you upload the sketch via OTA the first time, you don't have to reset the device, just recompile and upload. Faster compiles... at least on linux its easy to forget the IDE and just use a text editor and a simple Makefile based around arduino-builder and esptool. If /tmp is on a ramfs, compiles really fly.

Fettkeewl, 2018-01-28:
Yes that is one way for sure but atm I reboot my device after succesful OTA (in code), test it around and find that I sometimes need to alter something. Once the device has been reset and returned to normal functioning, subsequent OTA updates won't work untill I restart Arduino IDE. I do not know why but the device just wont show up in "Port" again if it has been in "normal" mode while IDE was open

pieman64, 2018-01-28:
@Fettkeewl it's a "Windows" issue, disappearing ports don't occur with Ubuntu.

MarkyAD, 2018-01-28:
@Fettkeewl issues with or changes to arduino IDE should go to its repository https://github.com/arduino/Arduino

Fettkeewl, 2018-01-28:
Thanks Costas, having that confirmed shows that the issue isn't here then :)

MarkyAD you can close this issue :)

closed 2018-01-28 by Fettkeewl