esp8266/Arduino #4244: Weird esp crashes with DMD2 library

lonedevwolf, 2018-01-27:
i have an esp 12e connected to the P10 led displays and im driving it with the DMD2 library, i just pull the string from a server and then display it on the display. and everything works, unless you call Wificlient.connect and the connection fails. then the esp crashes 

this is the exception
Exception (0):
epc1=0x402089b0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc50 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdf0:  00000000 40100fab 00000000 3ffedaa0  
3ffffe00:  00000001 00000000 3fffc248 4000050c  
3ffffe10:  4000437d 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 ffffffff 80000000  
3ffffe30:  20000000 3fff1da8 80000000 203fc1e0  
3ffffe40:  80000000 3fffc6fc 3ffeeb28 3fff1dac  
3ffffe50:  00000114 003fc1e0 60000600 00000030  
3ffffe60:  3fffc200 3ffef268 3ffef150 402083d5  
3ffffe70:  3fff13c4 3fff13e4 3fff1404 3fff1424  
3ffffe80:  00000000 006c39e7 00000000 402089d1  
3ffffe90:  00000000 00000000 0000001f 40105139  
3ffffea0:  4000050c 004f9daf 3fffc248 4000050c  
3ffffeb0:  40000f68 00000030 0000001c ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000013 4020f2f4 3ffeeb28 00000000  
3ffffee0:  ffffffff 3ffe9784 3ffeeb28 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffeeb68 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff1bcc 000002f4 003fc000  
3fffff20:  4010569e 3ffeeb50 3ffedbf0 40107418  
3fffff30:  4020edc5 3ffedbf0 3ffeeb50 025fe3b6  
3fffff40:  3fff1bcc 00001000 4020f25a 00000008  
3fffff50:  401060f0 00000000 4020f307 3ffedca4  
3fffff60:  3ffeeb50 0050115b 3ffeeb50 60000600  
3fffff70:  402209e1 3ffedca4 3ffeeb50 025f8d3a  
3fffff80:  40220a26 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffeeb70 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 00015035 3fffdab0 40000f49  

and this is the stripped out version of the sketch

#include <ESP8266WiFi.h>
WiFiClient client;
#include <WiFiManager.h>
WiFiManager wifiManager;

#include <SPI.h>
#include <DMD2.h>
#include <fonts/SystemFont5x7.h>
#include <fonts/Arial14.h>
#include <fonts/Arial_Black_16.h>
#define pin_A D0
#define pin_B D6
#define pin_sclk D8
#define pin_noe D3
#define panel_width 2
#define panel_heigh 1

SPIDMD dmd(panel_width, panel_heigh, pin_noe, pin_A, pin_B, pin_sclk);  // DMD controls the entire display
DMD_TextBox box(dmd);
void setup() {
  Serial.begin(9600); 
  dmd.setBrightness(255);
  dmd.selectFont(Arial_Black_16);
  dmd.begin();
  // put your setup code here, to run once:
  wifiManager.setConfigPortalTimeout(300);
  wifiManager.autoConnect("Makedonsko AutoAP");
  
  

}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(client.connect("192.168.0.104",8000));
  box.print("test");
  delay(1000);
}

you dont even need a led panel to reproduce the error, everything works if there is a server to connect to, if not the esp crashes.

devyte, 2018-01-27:
@lonedevwolf when you opened the issue, you were presented with a template requesting specific information. You pretty much ignored it in its entirety. Please edit your post and add the missing info. 
Also, please use markup for the sketch. It's a matter of just selecting the lines and clicking on the button that says "code".

lonedevwolf, 2018-01-28:
I solved the problem and edited the post, you can mark it as solved. I'm sorry for the template and code as it is my first time using github.

devyte, 2018-01-29:
@lonedevwolf are you using latest git? If not, can you test with that (instructions are readthedocs). Can you provide a minimal sketch that shows that using interrupts crashes wificlient?

lonedevwolf, 2018-02-03:
The code that causes the crash is in my first post, after a lot of fiddling i found that i haven't solved the problem. As it stands i hit i brick wall and any help is appreciated

devyte, 2018-02-03:
Did you test with latest git instead of 2.4 release?

lonedevwolf, 2018-02-05:
i just tried it with 2.4-r2 ,2.4-r1 and 2.4  still doesn't work, i was working on 2.3, on 2.3 it works for a while and then crashed but on 2.4 it crashes immediately :(

devyte, 2018-02-05:
@lonedevwolf again, did you test with LATEST GIT? You keep mentioning release versions that are installed via the IDE board manager. Latest git is NOT installed via the board manager. Lastest git is the current state of the master branch of this repo, and requires manual installation. It has several fixes that were merged after 2.4 release, some of which could be relevant.

closed 2018-02-27 by devyte