esp8266/Arduino #4245: TLS chain verify - long expire time overflows.

piodabro, 2018-01-27:
### Basic Infos
When using chain verification with long lasting CA for example >7500 days, the time decoded in debug is overflown.
I dont have dump but it looks like this:
Not Before:                     Sat Jan 27 20:26:15 2018
Not After:                      Sat Mar 28 20:26:15 1902

Now root CA has to be shorter than 19 years but in future it will be shorter every second :)

Probably this is caused by too small data type that holds the dates...

#### Hardware
Hardware:			Not matters
Core Version:      	2.4.0

piodabro, 2018-02-07:
No one cares, i see.

A little research and we know it is: [image]
And the question is how that problem will be resolved in the future...

devyte, 2018-02-07:
@piodabro you obviously care, so how about you fix it? 
- fork this repo to make your own
- Install git for your fork (uninstall board packaged version)
- create branch in your git
- fix the issue, test it, and commit locally
- push from your local git to your fork
- in the main webpage for this repo you'll see an alert with a green button "Compare and make pull request", click there
- review your changes and click on make pull request

piodabro, 2018-02-08:
Fixing it is not a simple change in code. It need a little discussion on possible fixes.

It is not a problem with SSL lib, but with time_t type, wchich is system wide.

The first one possibility is to extend time_t type to 64-bit value.
But this can cause some lib incompatibility.
Also this needs changes in toolchain libs (type is defined in xtensa-lx106) or overriding it globally for this framework...

d-a-v, 2018-02-08:
Changing definitions is not advised, adding new ones seems better.
This is [worth reading](https://sourceware.org/glibc/wiki/Y2038ProofnessDesign#implementation_design).

We could start adding and using these new types in our core and see how far we can go regarding ssl certs (waiting for our libc to adopt these changes).

earlephilhower, 2018-02-12:
While I agree in theory, @d-a-v , in practice I doubt that there will be any ESP8266s still running in 20 years.  If that's the case, then fixing the cert checker (which uses now()+delta everywhere and potentially has this problem w/super-long certs) would be a much less involved operation affecting nothing else.

That said, I've been looking at lots of certs while doing the BearSSL port and CertStore changes. While I see a few CA certs valid to just under time_t overflow there don't seem to be any that actually cross 2038.  So I'm not sure this is a problem even worth solving.  

As was mentioned, if you're doing self-certs the workaround is "make cert valid until 2038-(current year) " which doesn't seem too onerous.  Electromigration or flash bit rot will kill any ESP8266s in the field long before 2038.

So IMO this isn't worth changing the core for, but if @piodabro submits a patch just fixing AXTLS I absolutely think we should consider it!

igrr, 2018-02-19:
The necessary change to axTLS seems to be fairly minimal (~4 line change). What we need to implement in the core or in newlib is an `mktime64` function, which would return `time64_t`.

devyte, 2018-07-13:
@igrr @earlephilhower considering axtls -> bearssl, what's the path forward here?

earlephilhower, 2018-07-13:
BearSSL should be clean for this.  axTLS, I wouldn't open up if it's going away.

````
....
 632 \ Read a date (UTCTime or GeneralizedTime). The date value is converted
 633 \ to a day count and a second count. The day count starts at 0 for
 634 \ January 1st, 0 AD (that's they year before 1 AD, also known as 1 BC)
 635 \ in a proleptic Gregorian calendar (i.e. Gregorian rules are assumed to
 636 \ extend indefinitely in the past). The second count is between 0 and
 637 \ 86400 (inclusive, in case of a leap second).
 638 : read-date ( lim -- lim days seconds )
....
````