esp8266/Arduino #4246: PinMode as INPUT

Tensa225, 2018-01-27:
When programming a Pin to INPUT mode (on some boards this is required to sink current) the board is not able to achieve the Low Level. It seems to float at 3.3V?

As an OUTPUT it seems to function like an INPUT also and only in the latest core version(s).

The user is asked to watch the requirement on impedance and activate the INPUT state in order to sink current.

Having both characteristics in OUTPUT mode can fry your board if the sink current exceeds the maximum.

OUTPUT should be high impedance, INPUT low impedance. Are they?

tonton81, 2018-01-28:
on the ESP, LOW is HIGH and HIGH is LOW
if you drive a pin LOW you effectively set it as 3.3V
digitalWrite(pin, HIGH) effectively outputs 0V
it's a chinese design i guess :)

tonton81, 2018-01-28:
problem is, if it's changed in the core, everyone's code/projects will be effected. not good...

igrr, 2018-01-28:
> if you drive a pin LOW you effectively set it as 3.3V

That's not the case, as far as I can tell. How did you arrive to this conclusion?

WereCatf, 2018-01-28:
@tonton81 If that was true, none of the thousands and thousands of sensors people use every day with the ESP would work. That is to say, not even remotely true.

tonton81, 2018-01-30:
sorry im confused, i had the adafruit huzzah breakout and when i set a gpio high the output was low, i havnt played with the gpios lately though im not too fond of their shared finicky “states” :)

WereCatf, 2018-01-30:
@tonton81 How did you determine that the output was low? Did you check it with multimeter? Looking at the pictures at https://www.adafruit.com/product/2471 there is no inverting logic anywhere on the board, and the ESP itself doesn't use inverted logic like that.

tonton81, 2018-01-30:
it was wired to another mcu’s input i was using esp as output teensy as slave, perhaps i did something wrong, anyways was a few months ago :P

devyte, 2018-02-04:
I'm don't know the characteristics of the ESP by memory, but the following is my understanding:

@Tensa225 said:

> OUTPUT should be high impedance, INPUT low impedance. Are they?

I think you have it backwards, or you are confusing the meaning of high impedance. OUTPUT internally connects either VCC or GND to the pin, and is therefore low impedance. INPUT just reads what is connected to the pin, and needs to allow the entire range from GND to VCC, so it should be high impedance.

When you set a pin to INPUT, and you don't have anything connected to it, the reading could be anything (it floats). Not all pins behave the same though, some have pullups/pulldowns. If a pin doesn't have a pullup/pulldown, then you should probably add one yourself, depending on your application.

Having said the above, this is not the right place to discuss this. This is an issue tracker for the core, and the GPIO behavior doesn't currently have issues related to your description.
Closing.

closed 2018-02-04 by devyte

Tensa225, 2018-02-07:
No one should normally define the IO to INPUT in order to sink current.

OUTPUT generates a voltage level at a current that is drawn by the circuit. Sometimes with the added internal PULLUP resistor if needed and available in the microprocessor.

INPUT does not read anything it only enables the sinking of current. OUTPUT should be able to prevent sinking current in situations where current is too large to sink for instance.

A "analog" IO can read the voltage using a internal ADC.

There`s a difference in behaviour for the ESP Core versions regarding the PinMode operation.

vdeconinck, 2018-02-07:
@Tensa225 I may be mistaken, but I really your message above does not make much sense. 

Sorry if I'm stating the obvious, but OUTPUT vs INPUT has nothing to do with the direction of the current (sourcing or sinking), it has to do with which component drives the voltage level.
- If you want the ESP to drive the voltage level, use pinMode(pin, OUTPUT). If you then call digitalWrite(pin, HIGH), it will source current if the other end is trying to pull that voltage low (e.g. resistor to GND), or it will sink current if the other end is trying to pull that voltage up (e.g. resistor to VCC). 
- If you want the attached component to drive the voltage level, and the ESP to get that information, use pinMode(pin, INPUT), then let the other end impose its voltage and call digitalRead(pin)

OUTPUT is called "low impedance" , because there is almost no resistance between the pin and either VCC (if digitalWrite(pin, HIGH)) or GND (if digitalWrite(pin, LOW)).
INPUT is called "high impedance", because the ESP has no influence on the signal (like if it was disconnected or had a very high resistance). If nothing is connected to an input pin, the value returned by digitalRead() is generally undefined.

Now additionally to that, in input mode, a setting can be used to connect a resistance inside the chip from the pin to VCC (INPUT_PULLUP available for GPIO0-15) or GND (INPUT_PULLDOWN_16, only available for GPIO16, hence the name). This of course lowers the impedance, but the resistance is high enough so that the pin voltage can still be set by the component at the other end (which really has a LOW impedance).

Finally, many ESP8266 modules include external pull-up or pull down resistors that make sure the chip boots correctly (e.g. from Flash), and you cannot do anything to disable them apart from unsoldering them of course.

Analog pins (the ESP only has one), as you say, are INPUT pins that have the particularity to be read not only as a HIGH or LOW level by digitalRead(), but also as the actual value of the applied voltage using analogRead().

If you have noticed a change between versions (please specify which ones exactly) of the cores, please post a drawing of your schematic and a minimal sketch demonstrating the issue, as well as observed results with each tested version so we can reproduce the issue.

WereCatf, 2018-02-08:
At this point I do have to ask, if this is really the right place to try and teach someone electronics basics? 
@Tensa225 has completely misunderstood how GPIO works in a microcontroller, which is not an issue with the code, and keeping this thread alive here doesn't make much sense.

vdeconinck, 2018-02-08:
@WereCatf At this point I do have to admit, you are right :-). Sorry couldn't resist to explain...

vader9988, 2018-06-26:
You should take a look at this:

https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup/