esp8266/Arduino #4254: NodeMCU (ESP8266-12E) AP mode not working

bitboy85, 2018-01-29:
Hi,
i'm using a nodemcu 1.0 (ESP8266-12E module) and try to use the example Code to create an Access point.
Arduino IDE Version is 1.8.1 and esp8266 library has Version 2.3.0.

The wifi network is created but i'm unable to connect from my Laptop (T410, Linumxmint 17.3).
If found this https://github.com/esp8266/Arduino/issues/1094
and added the code lines suggested (wifi.disconnect() and clearflash), it seems that a connection from my android device is now more stable but it is still not possible to connect from my laptop.

devyte, 2018-01-29:
@bitboy85 when you opened this issue, you were presented with an issue template. You pretty much ignored it in its entirety. Without at least a MCVE sketch, it's unlikely your issue will be looked at.
Release 2.3.0 is very old. Latest release is 2.4.0, so opening issues for 2.3.0 doesn't make sense. However, I strongly suggest trying with latest git instead of 2.4.0 due to known issues.
Also, make sure to erase the entire flash before loading a new sketch.
Also, the ESP has only one radio. If you're using wifi in mode SoftAP, then the AP side will be dragged along channel hopping when the Station side scans or connects, and the device you're trying to connect from will see tha SoftAP disappear and may drop the attempt. The SoftAP channel should be set to the same channel as the AP that the station side tries to connect to.

I can't reproduce this from my own app, I see no problems with connecting either from my smartphone or laptop.
Closing due to lack of info and can't reproduce.

closed 2018-01-29 by devyte

irwanayeah, 2018-09-06:
I have a lot of nodemcu and only 1 pc has a case like the title above. so it seems like a hardware problem.

bitboy85, 2018-09-06:
I removed mdns from the sample code and it seems to work much better now.