esp8266/Arduino #4257: v2.4.0 - wifi.status() and wifi.localIP not updated after a reconnect.

Sparatutto, 2018-01-29:
Hello.
My HW and setup is the following:

[image]

Core Version:   2.4.0

### Description

When the connection with the AP (it is a Vodafone station if is useful to know) is lost, the module automatically try to reconnect.
Enabling the debug it seems to reconnect but the IP and wifi.status are not properly updated.

I did read many different issues and it seems to be solved with lwIP variant v2 536 and 1460.
But in my case it is not working.
Instead with lwIP v1.4 it works.

Am I doing something wrong?

Thanks,

Here below the test sketch for the issue.

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <time.h>                       // time() ctime()
#include <sys/time.h>                   // struct timeval

const char* ssid = "Vodafonxxxxxxxx";
const char* password = "mypwd";

time_t now;
 struct tm * timeinfo;

void setup() {
	byte mac[6];
    // put your setup code here, to run once:
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    delay(100);
	configTime(3600, 1 * 3600, "pool.ntp.org", "time.nist.gov");
    Serial.println("Start! ....");
    WiFi.macAddress(mac);
    Serial.printf("\r\nMAC ADDR:%02X:%02X:%02X:%02X:%02X:%02X \r\n\r\n",mac[5],mac[4],mac[3],mac[2],mac[1],mac[0]);
	WiFi.mode(WIFI_STA);
	WiFi.hostname("TestA");
	// WiFi.setAutoReconnect(true);
	// wifi_station_set_hostname(sysSetting.modLabel);
	WiFi.begin(ssid, password);

	for(int tent=0; tent<30; tent++) 
	{
		if(WiFi.status() != WL_CONNECTED)
		{
			delay(1000);
			Serial.print(".");
		}
		else
		{
			break;
		}
	}
	if(WiFi.status() == WL_CONNECTED)
	{
		Serial.println("\r\nWiFi connected");
		//Serial.println(WiFi.localIP());
		//WiFi.mode(WIFI_STA);
		WiFi.setAutoReconnect(true);
	}
	else
	{
		Serial.println("\r\nFailed to connect WiFi"); 
	}  
}

void loop() {
  // put your main code here, to run repeatedly:
    now = time(nullptr);
	timeinfo = localtime (&now);
	// any 5 seconds
	if(((timeinfo->tm_sec)%5) == 0)
	{
		// human readable
		// Serial.print(" ctime:");
		// Serial.print(ctime(&now));
	  
		if(WiFi.status() != WL_CONNECTED)
		{
		  Serial.printf("@time:%02d:%02d - Wifi conn status:%02d - ERROR\r\n",timeinfo->tm_hour, timeinfo->tm_min, WiFi.status());
		  Serial.printf("Is connected:%01d - IP: %d.%d.%d.%d \r\n",(int)WiFi.isConnected(), WiFi.localIP()[0],WiFi.localIP()[1],WiFi.localIP()[2],WiFi.localIP()[3]);
		  // ESP.restart();
		  // WiFi.disconnect();
		  // if(!connectToWiFi())
			  // ESP.restart();
		}
		else
		{
			Serial.printf("@time:%02d:%02d:%02d Is connected:%01d - IP: %d.%d.%d.%d \r\n",timeinfo->tm_hour, timeinfo->tm_min, timeinfo->tm_sec, (int)WiFi.isConnected(), WiFi.localIP()[0],WiFi.localIP()[1],WiFi.localIP()[2],WiFi.localIP()[3]);
		}
	}
	delay(1000);
}
```

### Debug Messages

```
@time:16:22:40 Is connected:1 - IP: 192.168.1.4 
@time:16:22:45 Is connected:1 - IP: 192.168.1.4 
@time:16:22:50 Is connected:1 - IP: 192.168.1.4 
@time:16:22:55 Is connected:1 - IP: 192.168.1.4 
@time:16:23:00 Is connected:1 - IP: 192.168.1.4 
@time:16:23:05 Is connected:1 - IP: 192.168.1.4 
@time:16:23:10 Is connected:1 - IP: 192.168.1.4 
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
@time:16:23 - Wifi conn status:06 - ERROR
Is connected:0 - IP: 0.0.0.0 
scandone
no Vodafonexxxxxx found, reconnect after 1s
reconnect
scandone
no Vodafonexxxxxx found, reconnect after 1s
reconnect
@time:16:23 - Wifi conn status:06 - ERROR
Is connected:0 - IP: 0.0.0.0 
scandone
no Vodafonexxxxxx found, reconnect after 1s
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with Vodafonexxxxxx, channel 1
dhcp client start...
@time:16:23 - Wifi conn status:06 - ERROR
Is connected:0 - IP: 0.0.0.0 
@time:16:23 - Wifi conn status:06 - ERROR
Is connected:0 - IP: 0.0.0.0 
pm open,type:2 0
@time:16:23 - Wifi conn status:06 - ERROR
Is connected:0 - IP: 0.0.0.0 

```

devyte, 2018-01-29:
@Sparatutto is this valid with latest git?

huckes, 2018-01-29:
Hi,
although I am nearly a newbie too, let me drop into this thread.
I encountered the same issue. After reconnection WiFi.isConnected() remains returning false and does not switch back to true as expected. Also the WiFi.status remains in 06.
My environment is Win10, Arduino IDE 1.8.5 with the board extension 
"http://arduino.esp8266.com/stable/package_esp8266com_index.json"
I found the ESP8266WiFi library below the "AppData-Path in
"AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries"
So I assume that is the official core version 2.4.0
Is the "git version" the version to which this issue-list is attached? I would try to install this one for test purpose  but I cannot find a download button. 
Best regards,
Hans

Sparatutto, 2018-01-29:
@devyte yes it is already installed with board Manager:
[image]

I will try to find a way to use the git procedure (to install Python 2.7 could be a problem).

Does it mean that the version available using git is more updated than the version linked by board manager (URL)?

devyte, 2018-01-29:
The git version is always more updated than the board manager version. The downside is that it could potentially get broken every so often, since the most recent commit is development by definition. We try to avoid that, but it can happen.

Sparatutto, 2018-01-31:
@devyte i'm trying to get the git working and verify if the issue is still present but it is not easy.
I don't have all the admin rights in my system and it is very complicated.
Anyway the situation now is : nothing is working (Arduino IDE included :-(  ) .

with github Desktop I have cloned the git:

[image]

But I cannot run python from a cmd window. I have a python shell (3.6.3) but running get.py in the shall it takes hours and the result doesn't work.

So no way for me to test it.
If someone else can test, or you confirm it has been fixed I can wait the next release usable with board management procedure.

tonton81, 2018-01-31:
it has to be 2.7 not 3.x, git version is always good fixes, think of it as nightlies :)

Sparatutto, 2018-01-31:
Ok after the approval of the admin I got python 2.7 and git 2.16 ....
It is slower than before probably (with lwIP v1.4 it was less than 1 min) but after a while (more than 1 min wih this last v2.0) the status and ID are re-assigned.

So the issue can be closed for me.

One question before to close: have I to clone and run python time by time to have always the last update correct?

Thanks!

d-a-v, 2018-01-31:
You can try with lwip v2 and mss=1460 for better speed.
Menu will soon change and the erroneous unstable mention be removed.

devyte, 2018-01-31:
No, not necessary. The python version can probably stay as is, at least for the foreseeable future. 
However, if you want to update the core files to rebuild with latest git, you'll need to git pull before building.
You don't need to do it every time, though. Update only when needed, or once in a while (maybe monthly, up to you).
Closing.

closed 2018-01-31 by devyte

tonton81, 2018-01-31:
you could create a batch file to do it, then run it as a task when your computer is idle every couple weeks

drschlaumeier, 2018-02-18:
I found same problem today and can confirm that 2.4.0 has a bug.
IP & status are not updated correctly after reconnect. My router shows that its connected and has valid IP. I also can call the websever.
The newest git from today has not bug/problems and ip & status are updated correctly. 

Rgds

devyte, 2018-02-18:
@drschlaumeier thanks, it's always welcome when users confirm against git.

drschlaumeier, 2018-02-18:
Thanks. How far away is git from new release? weeks? month?

tonton81, 2018-02-18:
you dont like using git? i dont prefer /outdated/ buggy releases, nightlies are better :)

drschlaumeier, 2018-02-19:
hmm, I thought that stable releases are extensively tested and mostly bug free.
Git is development version and normally not stable.
So whats the process for Arduino?

tonton81, 2018-02-19:
the development branch is all about bug fixes and new features, and sometimes, new features can bring more bugs, most of the time tho they fix the problems that others see, that wasnt previously caught or figured out yet

drschlaumeier, 2018-02-20:
Hmm, I will try git for my smart home systems but I'm still sceptical. 
Anyhow, what are the plans for next release since official arduino ide updates releases only?
Its more difficult to keep git up2date and follow all the features and bugs and bug fixes.