esp8266/Arduino #4259: I think you are looking in a bad direction

omiagut, 2018-01-30:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			Dell & ubuntu 16.04
Core Version:      	several

### Description

Problem description

### Settings in IDE both Arduino and ESPlorer

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

omiagut, 2018-01-30:
If I am interest in this discussion, it's also because the problem is mine.
Sorry, but I'm a dummy, and a beginner for ESP8266, but let me tell you a little story.
My ESP's worked all perfectly. ESP-01 connected with a FDTI232 and ESP12E connected with ..341. and working with Arduino 1.8.5 IDE, Ubuntu 16.04.
Suddenly  they didn't work anymore. I just noted that the ESP12E wich usually wanted to be named 192.168.1.80 wanted to be named 192.168.1.81. Well. Never mind.
At list I thought it was an Arduino problem and tried to connect with esptool and ESPlorer. esptool seems to flash perfectly, "screen" work once or twice, but ESPlorer : connecting with MCU ... , nothing more, the reset button stays disabled.  Well never mind. It's an Ubuntu problem.
I have installed Ubuntu three times, done an dmesg -s 1024 to see if everything was OK, it seems so, more or less. 
Well, now you now everything about a dummy's three weeks period.
At this moment, I thought it's not an USB problem, my keyborad and mouse are perfectly connected. What about my network ? I found a command arp -a on ubuntu's forum, and obtained 
box (192.168.1.10 à 00:25:15:44:68:04 [ether] sur enp2s0
? (192.168.1.81) à <incomplet> sur enp2s0
HUAWEI_P8_2017 (192.168.1.80) à <incomplet> sur enp2s0
? (192.168.1.25) à <incomplet> sur enp2s0
enp2s02 is my SFR box (french phone, tv and TCP/IP network)
The .25 is an ESP-01 I tried to connect as a server.
My wife has connected a new Huawey phone, ithis explains the 80 -> 81 of my esp12E.  She has also changed the contract with SFR, including my PC (192.168.1.43) 
Sorry, once again, I am a dummy and may be wrong. But now I try to understand how to connect to my SFR private network, or how to create my own private network.

devyte, 2018-01-30:
@omiagut this is not the right place to ask for help with your setup or network or ubuntu problems. This is an issue tracker, meant to track issues in the core libs, and not a discussion forum. You say you're a beginner, so I suggest requesting help from peers in a forum like esp8266.com or stackoverflow.
Closing as off topic, see issue POLICY doc for details.

closed 2018-01-30 by devyte